espressif/esp-idf #1742: spiram heap_caps_check_integrity(MALLOC_CAP_8BIT, true) assertion

chmorgan, 2018-03-19:
Using an ALB32-WROVER module with 4MB of external spiram, 8MB of spiflash.

```
I (5569) esp_image: segment 6: paddr=0x0014d4d0 vaddr=0x400c0000 size=0x00000 (     0) load
I (5583) boot: Loaded app from partition at offset 0x30000
I (5583) boot: Disabling RNG early entropy source...
I (5586) spiram: SPI RAM mode: flash 40m sram 40m
I (5589) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (5596) cpu_start: Pro cpu up.
I (5600) cpu_start: Starting app cpu, entry point is 0x40081338
```

spiram is configured to only make spiram available via heap_caps_malloc() and I'm not allocating any.

```
assertion "next > (intptr_t)block" failed: file "/Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x400e5b23 on core 0
0x400e5b23: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40093793:0x3ffd86a0 0x40093a34:0x3ffd86d0 0x400e5b23:0x3ffd8700 0x40092da3:0x3ffd8730 0x400d767b:0x3ffd8770 0x400d7b2c:0x3ffd87a0 0x400d246a:0x3ffd87d0
0x40093793: invoke_abort at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./panic.c:652

0x40093a34: abort at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./panic.c:652

0x400e5b23: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40092da3: get_next_block at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c:670
 (inlined by) multi_heap_check at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c:626

0x400d767b: heap_caps_check_integrity at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./heap_caps.c:400

0x400d7b2c: app_main at /Users/cmorgan/projects/loadcontrol/app/main/.//app.c:204

0x400d246a: main_task at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./cpu_start.c:467

Rebooting...
```

This issue goes away if I don't build in support for spiram. Using latest esp-idf master (7e268a).

Is this module potentially defective? There are no errors during startup even though 'Run memory test on SPI RAM initialization' is enabled.

Also, I have tried allocating spiram with heap_caps_malloc() and even small amounts, say 30k, fail to allocate.

igrr, 2018-03-19:
Could you please attach your sdkconfig file?

chmorgan, 2018-03-19:
Hi @igrr, of course, attached to this response.

I've also verified that the esp32 is reporting rev 1 via esp_chip_info():

```
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("%d cores, chip revision %d\n", chip_info.cores, chip_info.revision);
```

```
...
2 cores, chip revision 1
...
```

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/1823491/sdkconfig.txt)

igrr, 2018-03-19:
With the following code in app_main:
```c++
    for (int i = 0; i < 1000; ++i) {
        printf("%d\n", i);
        free(heap_caps_malloc(4096, MALLOC_CAP_SPIRAM));
        heap_caps_check_integrity(MALLOC_CAP_SPIRAM, true);
    }
```
application runs correctly with your sdkconfig, only two options changed: 1) changed partition table to the default one, and 2) increased interrupt watchdog timeout to 1 second.
I don't have an ALB32-WROVER module to test, this is all on ESP32-WROVER module mounted on ESP-WROVER-KIT devboard.

chmorgan, 2018-03-20:
Hi @igrr 

I also extended the watchdog timeout to 50 seconds.

Does this indicate that there is an issue with the spiram? I would have expected the memory checks mentioned in the spiram kconfig section to have caught physical issues with the chip.

There was an issue with the heap capabilities checking that I worked on with @ESP_Angus on the forum, could this be related to that vs. an actual issue with the memory?

```
I (6535) cpu_start: Pro cpu start user code
I (6540) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (191) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
0
1
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40092f44  PS      : 0x00060234  A0      : 0x80093304  A1      : 0x3ffd7670  
0x40092f44: verify_fill_pattern at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap_poisoning.c:326

A2      : 0x3fb3e100  A3      : 0x000c1ef4  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00000000  A7      : 0xfefefefe  A8      : 0x80093148  A9      : 0x00000001  
A10     : 0x3f800000  A11     : 0x5421e307  A12     : 0x3f800018  A13     : 0xfefefefe  
A14     : 0x3f800019  A15     : 0x3f801029  SAR     : 0x00000014  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40097734  LEND    : 0x4009773f  LCOUNT  : 0xffffffff  
0x40097734: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4009773f: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x40092f44:0x3ffd7670 0x40093301:0x3ffd76b0 0x40092d75:0x3ffd76e0 0x400d75df:0x3ffd7720 0x400d7a09:0x3ffd7750 0x400d2482:0x3ffd7780
0x40092f44: verify_fill_pattern at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap_poisoning.c:326

0x40093301: multi_heap_internal_check_block_poisoning at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap_poisoning.c:334

0x40092d75: multi_heap_check at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c:670

0x400d75df: heap_caps_check_integrity at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./heap_caps.c:400

0x400d7a09: app_main at /Users/cmorgan/projects/loadcontrol/app/main/.//app.c:139 (discriminator 3)

0x400d2482: main_task at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./cpu_start.c:467

Core 1 register dump:
PC      : 0x400d6d6c  PS      : 0x00060834  A0      : 0x8008fcb2  A1      : 0x3ffd8360  
0x400d6d6c: esp_vApplicationIdleHook at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./freertos_hooks.c:85

A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x3ffd7ea8  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffb38cc  A9      : 0x3ffb3890  
A10     : 0x00000000  A11     : 0x5421e307  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060820  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40097734  LEND    : 0x4009773f  LCOUNT  : 0xffffffff  
0x40097734: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:142

0x4009773f: memset at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/memset.S:152

Backtrace: 0x400d6d6c:0x3ffd8360 0x4008fcaf:0x3ffd8390
0x400d6d6c: esp_vApplicationIdleHook at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./freertos_hooks.c:85

0x4008fcaf: prvIdleTask at /Users/cmorgan/projects/loadcontrol/esp-idf/components/freertos/./tasks.c:4571

Rebooting...
ets Jun  8 2016 00:22:57
```

projectgus, 2018-03-20:
> I also extended the watchdog timeout to 50 seconds.
> 
> Does this indicate that there is an issue with the spiram? I would have expected the memory checks mentioned in the spiram kconfig section to have caught physical issues with the chip.

Hi @chmorgan,

The default interrupt WDT period (300ms) does time out when verifying SPIRAM if Comprehensive checking is on. This is because we have to disable interrupts while checking a heap, and reading every byte from SPIRAM takes some time. We can increase the minimum value in Kconfig for the case when SPIRAM & comprehensive heap checking is on, so this doesn't happen.

Do I understand correctly that the last trace you posted above is with INT WDT period set to 50 seconds? Does the crash happen after a 50 second delay, or sooner? If so, please try dropping the timeout to 1 second or so and see if it still occurs - it may be there's an overflow in the wdt timeout calculation for very long periods (which we'll fix if so.)

Angus

chmorgan, 2018-03-20:
@projectgus I changed the task watchdog to 50 seconds, not the interrupt watchdog timeout.

Extending the wtd to 1000ms lets the code continue without issue (stopped it after a few dozen runs), so that does look like the issue that was occurring?:

```
I (0) cpu_start: Starting scheduler on APP CPU.
0
allocated
1
allocated
2
allocated
3
allocated
4
allocated
5
allocated
6
allocated
7
allocated
8
allocated
```

What is the expected fix for that? To increase the value by default when people enable spiram? Would save the next guy a number of hours of debugging.

So I modified the code to allocate more memory and you can see that it can allocate a bunch:

```
376832 allocated
93
380928 allocated
94
385024 allocated
95
389120 allocated
96
393216 allocated
97
397312 allocated
98
401408 allocated
99
405504 allocated
```

But I'm still seeing the task level check:

```
    while(1)
    {
        if (!heap_caps_check_integrity(MALLOC_CAP_8BIT, true)) { <----- line 214
#if defined(CONFIG_HEAP_TRACING)
            heap_trace_dump();
#endif
            abort();
        }
...
```

failing like this:
```
assertion "next > (intptr_t)block" failed: file "/Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x400e5a7b on core 0
0x400e5a7b: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

Backtrace: 0x40093787:0x3ffd7650 0x40093a28:0x3ffd7680 0x400e5a7b:0x3ffd76b0 0x40092d97:0x3ffd76e0 0x400d7573:0x3ffd7720 0x400d7a6c:0x3ffd7750 0x400d2416:0x3ffd7780
0x40093787: invoke_abort at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./panic.c:652

0x40093a28: abort at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./panic.c:652

0x400e5a7b: __assert_func at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x40092d97: get_next_block at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c:670
 (inlined by) multi_heap_check at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./multi_heap.c:626

0x400d7573: heap_caps_check_integrity at /Users/cmorgan/projects/loadcontrol/esp-idf/components/heap/./heap_caps.c:400

0x400d7a6c: app_main at /Users/cmorgan/projects/loadcontrol/app/main/.//app.c:214

0x400d2416: main_task at /Users/cmorgan/projects/loadcontrol/esp-idf/components/esp32/./cpu_start.c:467
```

It doesn't make sense that the heap caps would fail in that manner given that I've set the memory allocation to require the explicit heap_caps_allocate() call.

chmorgan, 2018-03-20:
@projectgus I think my second set of comments may not have been clear. When I enable spiram I'm seeing that heap_caps_check_integrity(MALLOC_CAP_8BIT, true) fails. Without spiram enabled there aren't any heap integrity issues.

Note that I'm also setting the allocation option that requires heap_caps_malloc(), "Make ram allocatable using heap_caps_alloc...", and not making any calls to heap_caps_alloc() anywhere except in the test code that @igrr provided (with some minor modifications).

Even if I compile out the allocation code that @igrr provided and I modified I see the same heap_caps_check_integrity(MALLOC_CAP_8BIT, true) failure simply by enabling spiram (without using it). This integrity check was working fine since the fix that you made some time ago from the cJSON realloc() issue. The only change is enabling spiram support. I can disable spiram support and I won't see that heap_caps_check_integrity() failure.

It sounds like there may be something else going on.

chmorgan, 2018-03-20:
@projectgus I'll keep bisecting here. I can't reproduce in the hello world test app so I think it may be something in my code breaking things.

I wish it were easier to catch these fault conditions. If there were some way to perform heap checking on the exit of each function called that imo would be ideal. Even if it killed performance it would be super useful if it would let someone catch an issue without so much effort.

chmorgan, 2018-03-20:
@projectgus and I think I've found it. I have to check a few things but will post a follow up.

Thank you again for the help with the wdt setting.

chmorgan, 2018-03-21:
@projectgus alright, so I've pretty much confirmed that the issue was that I was configuring GPIO16 and GPIO17, the lines also used for the spiram interface, for pwm usage. There isn't an external device on those pins but clearly driving them from the mcpwm module was messing things up.

Would it make sense for the GPIO or pin configuration code to check if GPIO16 and GPIO17 were being configured when CONFIG_SPIRAM_ENABLED (or whatever the flag was) was set? I know the gpio code already detects internal configuration issue, such as trying to configure one of the input only gpio pins for output. Thoughts?

panfeng-espressif, 2018-04-02:
Hello chmorgan,
You would like us to confirm whether it would make sense if we check GPIO16 & GPIO17 at pin configuration code? Is that right?

chmorgan, 2018-04-02:
Correct. The idea is to avoid users enabling those pins for good when using
spiram, a condition that is known to cause issues.

FayeY, 2018-11-30:
Sorry for the late reply. After evaluation, we may not add this feature to check GPIOs, since it will affect the efficiency of GPIO APIs. Sorry for the inconvenience.

closed 2018-11-30 by FayeY