espressif/esp-idf #1743: When using WPS, the SSID is not saved.

perm27, 2018-03-19:
In EPS-WROOM-32 when using WPS, the SSID is not saved.
 Immediately after the WPS command is executed, SSID is set and it works well.
 However, it seems that you can not connect automatically with AP when restarting or turning off / on the power again.

 Is there a way to save the SSID etc acquired by WPS to connect even after reboot?

yake827, 2018-03-22:
Hi @perm27 ,It is very easy, here is my code.

`    case SYSTEM_EVENT_STA_WPS_ER_SUCCESS:       /**< ESP32 station wps succeeds in enrollee mode */
        esp_wifi_wps_disable();
        esp_wifi_get_config(WIFI_IF_STA, &conf);     // Get config info , include ssid and pwd
        esp_wifi_set_storage(WIFI_STORAGE_FLASH);    // set save to flash
        esp_wifi_set_config(WIFI_IF_STA,&conf);         // set config , it will save to flash
        esp_wifi_set_storage(WIFI_STORAGE_RAM);     // change to save to ram
        esp_wifi_connect();         // connect
        at_printf("wps success,connecting ap ...\r\n");
        break;
`

perm27, 2018-03-23:
Hi@yake827, Thank you code.

It is now possible to connect to the AP paired with WPS even after rebooting.
In the code as it is, after WPS connection, IP address could not be acquired with bssid:00:00:00:00:00:00.
Therefore, I added the next line.
    esp_wifi_get_config (WIFI_IF_AP, & conf);

Thank you.

closed 2018-03-23 by perm27