espressif/esp-idf #1744: [Issue]: TLS handshake returns 0x50 on 80MHz CPU clock

farizadp, 2018-03-19:
Hello,

We've got an issue here with mbedtls client handshake in combination with 80MHz CPU clock.
On the ESP32 (client side) SSL handshake returns 0x50, and it turns out that it's the mbedtls_net_send() function which returns MBEDTLS_ERR_NET_CONN_RESET.
Looking at mbedtls_net_send() function, error variable contains ECONNRESET which I initially suspected that it's because the ESP32 which is too slow responding to the server.

However I performed comparison of each SSL states duration, and they seem to be performing at the same duration, which in this case denies my suspicion that ESP32 is responding too slow.

state	enter(160MHz)	dur(ms)	enter(80MHz)	dur(ms)
0	16702	34	        16694	16
1	16736	132	        16710	66
2	16868	221	        16776	111
3	17089	4166	16887	4068
4	21255	50	        20955	25
5	21305	210	        20980	105
6	21515	208	        21085	104
7	21723	134	        21189	80
8	21857	403	        21269	346
9	22260	2159	21615	
10	24419	134		
11	24553	176		
12	24729	209		
13	24938	252		
14	25190	45		
15	25235			

So, I am not sure where else to look. FYI, The 80MHz is our board limit due to it's current budget, and going to 160MHz is really on the edge of our current budget.
Is there anything related with 80MHz instability in relationship to Hardware acceleration for example? or any other issue which might be related to this?

Switching to 160MHz temporarily on the fly during the handshake, and switch back to 80MHz does solve the handshake issue. But I consider this as workaround, and as I mentioned we are getting to the edge of our current budget.

Anyone can help?

Regards,
Dion

farizadp, 2018-04-13:
Hi,

Is there any issue related to this?

Dion

jitin17, 2018-08-23:
@farizadp Hey, while recently working on `https_request` example, located at `examples/protocol/https_request`, I was able to perform TLS handshake at 80 MHz CPU clock. So can you check if you still face the same issue?

farizadp, 2018-09-06:
@jitin17: Just checked it with the latest esp-idf. SSL handshake on 80MHz CPU frequency works fine. This issue can be closed.

closed 2018-09-06 by mahavirj