espressif/esp-idf #1748: NVS partition usage

hvishu, 2018-03-20:
When "simply factoryapp, no OTA" option is choosed from make menuconfig for app like blink, I see the below during board bootup.  

I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000

I am required to use custom partition table,  not sure about the nvs size required for wifi.  In my app, I required to aws, and in one of the trial, if I use the nvs size less than 0x3000, esp_wifi_init() gets failed.

I required to use nvs for my application along with nvs used WIFI.  If there is any documentation or information on NVS requirement for WIFI and for what purpose this is used, it would be helpful to create the partitions.

igrr, 2018-03-20:
If you don't use NVS APIs in your application, then the default size of 4 sectors (16kB) is the most suitable for WiFi. This is not the smallest size required (it can work with 12kB), but one extra sector gives a bit of headroom in case WiFi driver adds some feature in the future version.
(We'll also add this information to the docs.)

hvishu, 2018-03-20:
OK.

closed 2018-03-20 by hvishu