espressif/esp-idf #1750: aws_iot: failed! mbedtls_ssl_handshake returned -0x4310

satishIoT, 2018-03-20:
Hi 
I keep facing this error now and then when ever i work with shadow.
1) aws_iot: failed! mbedtls_ssl_handshake returned -0x4310
    E (18143) APP_DEBUG: aws_iot_shadow_connect returned error -4, aborting...
2) Shadow

ameyinamdar, 2018-03-21:
Satish, the error -0x4310 refers to MBEDTLS_ERR_RSA_PRIVATE_FAILED (0x4300) + MBEDTLS_ERR_MPI_ALLOC_FAILED (0x0010). It looks like your system is running out of memory.

satishIoT, 2018-03-24:
Any more input im using ERS32 WROOM module and i guess we have enough memory to deal with all tasks. Any changes in menuconfig to be taken care in mbedtls ?

mahavirj, 2018-03-24:
@satishIoT 
Did you check how much dynamic memory is available (please note that WROOM module does not have external SPIRAM) before starting TLS connection (or at various stages in AWS IoT shadow task)? You can use `esp_get_free_heap_size` and `esp_get_minimum_free_heap_size` APIs for this.

negativekelvin, 2018-03-24:
https://github.com/espressif/esp-idf/issues/1695

satishIoT, 2018-03-27:
Thanks @negativekelvin your suggestion was spot on. 
But i have another issue ; im using the shadow example and as understood if n/w disconnects then the autoreconnect of shadow should have happened and should have looped in 
while(NETWORK_ATTEMPTING_RECONNECT == rc || NETWORK_RECONNECTED == rc || SUCCESS == rc) {
        rc = aws_iot_shadow_yield(&mqttClient, 200);
        if(NETWORK_ATTEMPTING_RECONNECT == rc || shadowUpdateInProgress) {
            rc = aws_iot_shadow_yield(&mqttClient, 1000);
            // If the client is attempting to reconnect, or already waiting on a shadow update,
            // we will skip the rest of the loop.
            continue;
        }
But instead it goes all down through the loop and give rc = -28. 
Pl suggest how to initiate the reconnect when ever n/w breakdown happens and comes back. 

thanks 
satish

satishIoT, 2018-03-27:
[image]

pl suggest on above errors.

satishIoT, 2018-03-30:
Dear Experts , Awaiting for  suggestions on how to deal with above errors , to make my connection robust .

thanks in advance !!

ameyinamdar, 2018-04-02:
@satishIoT "mbedtls_net_connect returned -0x52" indicates hostname resolution failure (internet connectivity issue?)

satishIoT, 2018-04-04:
@ameyinamdar - Im using hotspot to verify/test how the connectivity works say if i try posting mqtt messages say at every minute then i dont see such issue but if i dont enable my hotspot for 45 min or more ( kind of connection break )  i see this error along with -14 and -13 error also. i want to have a robust method to connect my device when ever connectivity returns back. what should i need to take care in such cases.

Alvin1Zhang, 2019-03-15:
@satishIoT Hi, sorry that get back to you so late. Does your issue still exist? Thanks.

mahavirj, 2019-04-18:
@satishIoT Closing this issue (not reproducible at our end), in case it still exists please feel free to reopen at https://github.com/espressif/esp-aws-iot (we moved `aws-iot` to independent repository)

closed 2019-04-18 by mahavirj