espressif/esp-idf #1751: promiscuous : wifi_pkt_rx_ctrl_t field "noise_floor" always "0"

jonathanmuller, 2018-03-20:
Hello,

For some weeks now, I'm stuck with this "noise_floor" field being always 0, without having found any way to get it's value. 

## **Goal**
Measure RF noise around the ESP32 to deduce its surroundings

## **Problem**

I'm trying to access the "**noise_floor**" field of packets received by my callback function in promiscuous mode. Despite receiving multiple packets with nearly all the other fields varying, the "**noise_floor**" is stuck to 0, same for some others :

List of non working field : 
**noise_floor**, **rx_state**, cwb, stbc, fec_coding, sgi (for cwb & sgi I may only not have such network around me)

List of working field : 
rssi, rate, sig_mode, mcs, smoothing, not_sounding, ampdu_cnt, channel, timestamp, sig_len

## **Expected** 

I expect "**noise_floor**" to be the noise of the surrounding RF and to vary if the device is in open field area or in a shielded box. As there is not a lot (not at all ?) documentation on this field I don't really know what to expect.

## **How to reproduce**

My callback : esp_wifi_set_promiscuous_rx_cb(&promiscuous_mode);
(whole code at bottom)
```
void promiscuous_mode(void *buff, wifi_promiscuous_pkt_type_t type){   
  /*
  printf("test %d ", loop);
  */
  switch(type) {
    case WIFI_PKT_MGMT: 
      printf(" MGMT : ");
      break;
    case WIFI_PKT_DATA: 
      printf(" DATA : ");
      break;
    default:  
      case WIFI_PKT_MISC: printf(" MISC : ");
  } 
  
  const wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buff;
  const wifi_ieee80211_packet_t *ipkt = (wifi_ieee80211_packet_t *)ppkt->payload;
  const wifi_ieee80211_mac_hdr_t *hdr = &ipkt->hdr;  

  printf("rssi %d ", ppkt->rx_ctrl.rssi); // OK
  printf("rate %u ", ppkt->rx_ctrl.rate); // OK
  printf("sig_mode %u ", ppkt->rx_ctrl.sig_mode); // OK
  printf("mcs %u ", ppkt->rx_ctrl.mcs); // OK
  printf("cwb %u ", ppkt->rx_ctrl.cwb); // XX
  printf("smoothing %u ", ppkt->rx_ctrl.smoothing); // OK
  printf("not_sounding %u ", ppkt->rx_ctrl.not_sounding); // OK
  printf("stbc %u ", ppkt->rx_ctrl.stbc); // XX
  printf("fec_coding %u ", ppkt->rx_ctrl.fec_coding); // XX
  printf("sgi %u ", ppkt->rx_ctrl.sgi); // XX
  printf("noise_floor %u ", ppkt->rx_ctrl.noise_floor); // XX
  printf("ampdu_cnt %u ", ppkt->rx_ctrl.ampdu_cnt); // OK
  printf("channel %u ", ppkt->rx_ctrl.channel); // OK
  printf("timestamp %u ", ppkt->rx_ctrl.timestamp); // OK
  printf("sig_len %u ", ppkt->rx_ctrl.sig_len); // OK
  printf("rx_state %u ", ppkt->rx_ctrl.rx_state); // XX
  printf("\n");
}
```
With typical outcome : 
```
 DATA : rssi -70 rate 11 sig_mode 1 mcs 3 cwb 0 smoothing 1 not_sounding 1 stbc 1 fec_coding 0 sgi 0 noise_floor 0 ampdu_cnt 1 channel 1 timestamp 1536593933 sig_len 84 rx_state 0 
 MGMT : rssi -92 rate 1 sig_mode 0 mcs 0 cwb 0 smoothing 0 not_sounding 0 stbc 0 fec_coding 0 sgi 0 noise_floor 0 ampdu_cnt 0 channel 1 timestamp 1536841824 sig_len 187 rx_state 0 
 DATA : rssi -68 rate 9 sig_mode 0 mcs 0 cwb 0 smoothing 0 not_sounding 0 stbc 0 fec_coding 0 sgi 0 

noise_floor 0 ampdu_cnt 0 channel 1 timestamp 1536477117 sig_len 329 rx_state 0
``` 
Outcome doesn't vary (except from RSSI) when placing the ESP32 in a shielded box

## **Guess**

It seems that the ESP32 is using the RF noise coming from wifi/bt as entropy for the RNG in a register which is -if I'm not mistaken- directly filled by the hardware. I probably am doing something wrong with the way I'm catching the packets, bus as few documentation is present I can't really see where is comes from. This would explain why some other field ()

Have any of you ever seen the floor_level working ? It may me a faulty unit

It may also have to do with the "**rx_state**" fixed to 0 

/
/

Thank you and have a good day

Bests,
Jonathan

Whole code bellow :

```
#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_wifi_internal.h"
#include "lwip/err.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"

//  http://blog.podkalicki.com/esp32-wifi-sniffer/

static wifi_country_t wifi_country = {.cc="CN", .schan=1, .nchan=13, .policy=WIFI_COUNTRY_POLICY_AUTO};

typedef struct {
  unsigned frame_ctrl:16;
  unsigned duration_id:16;
  uint8_t addr1[6]; /* receiver address */
  uint8_t addr2[6]; /* sender address */
  uint8_t addr3[6]; /* filtering address */
  unsigned sequence_ctrl:16;
  uint8_t addr4[6]; /* optional */
} wifi_ieee80211_mac_hdr_t;

typedef struct {
  wifi_ieee80211_mac_hdr_t hdr;
  uint8_t payload[0]; /* network data ended with 4 bytes csum (CRC32) */
} wifi_ieee80211_packet_t;

static void wifi_sniffer_init(void);

esp_err_t event_handler(void *ctx, system_event_t *event)
{
    return ESP_OK;
}

void promiscuous_mode(void *buff, wifi_promiscuous_pkt_type_t type){   
  static int loop=0;
  static long int last_time=0;
  static int frame_count=0;

  /*
  printf("test %d ", loop);
  */
  switch(type) {
    case WIFI_PKT_MGMT: 
      printf(" MGMT : ");
      break;
    case WIFI_PKT_DATA: 
      printf(" DATA : ");
      break;
    default:  
      case WIFI_PKT_MISC: printf(" MISC : ");
      printf("Reussito !!\n");
      int i=0;
      while(1){
        i+=i;
        }
  } 
  
  const wifi_promiscuous_pkt_t *ppkt = (wifi_promiscuous_pkt_t *)buff;
  const wifi_ieee80211_packet_t *ipkt = (wifi_ieee80211_packet_t *)ppkt->payload;
  const wifi_ieee80211_mac_hdr_t *hdr = &ipkt->hdr;  

  /*
  //printf("%d ", ppkt->rx_ctrl.rssi);
  if (esp_log_timestamp()-last_time>100){
    last_time = esp_log_timestamp();
    printf("Nb frame : %d\n",frame_count);
    frame_count=0;
  }
  //printf("%d",esp_log_timestamp()-last_time);
  frame_count ++;
  printf("RSSI : %d \n", ppkt->rx_ctrl.rssi);
  */

  
  printf("rssi %d ", ppkt->rx_ctrl.rssi); // OK
  printf("rate %u ", ppkt->rx_ctrl.rate); // OK
  printf("sig_mode %u ", ppkt->rx_ctrl.sig_mode); // OK
  printf("mcs %u ", ppkt->rx_ctrl.mcs); // OK
  printf("cwb %u ", ppkt->rx_ctrl.cwb); // XX
  printf("smoothing %u ", ppkt->rx_ctrl.smoothing); // OK
  printf("not_sounding %u ", ppkt->rx_ctrl.not_sounding); // OK
  printf("stbc %u ", ppkt->rx_ctrl.stbc); // XX
  printf("fec_coding %u ", ppkt->rx_ctrl.fec_coding); // XX
  printf("sgi %u ", ppkt->rx_ctrl.sgi); // XX
  printf("noise_floor %u ", ppkt->rx_ctrl.noise_floor); // XX
  printf("ampdu_cnt %u ", ppkt->rx_ctrl.ampdu_cnt); // OK
  printf("channel %u ", ppkt->rx_ctrl.channel); // OK
  printf("timestamp %u ", ppkt->rx_ctrl.timestamp); // OK
  printf("sig_len %u ", ppkt->rx_ctrl.sig_len); // OK
  printf("rx_state %u ", ppkt->rx_ctrl.rx_state); // XX
  printf("\n");
}

void app_main(void)
{
  nvs_flash_init();
  tcpip_adapter_init();
  ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
  wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
  ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
  ESP_ERROR_CHECK( esp_wifi_set_country(&wifi_country) ); /* set country for channel range [1, 13] */
  ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
  ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_MAX) );
  ESP_ERROR_CHECK( esp_wifi_start() );
  esp_wifi_set_promiscuous(true);
  
  esp_wifi_set_promiscuous_rx_cb(&promiscuous_mode);

}

```

jack0c, 2018-03-21:
Thanks for your feedback, I will check if we can do this. Can you use `RSSI` instead?

jonathanmuller, 2018-03-21:
Hi,

Thank you for your quick answer.

It would be possible to deduce the noise by comparing a reference signal (no noise) with the received one (noisy). Unfortunately, I don't have a reference signal to compare with as it will be in an "unknown"/new environment.

liuzfesp, 2018-04-03:
@jonathanmuller currently the noise_floor is not set to wifi_pkt_rx_ctrl_t->noise_floor and it is always 0. I think we can add it.

jonathanmuller, 2018-04-04:
@liuzfesp Do you have an idea on how long it would take, like an order of magnitude ? (weeks, months, ...)

liuzfesp, 2018-05-28:
Hi @jonathanmuller the noise floor is merged into IDF master. The commit ID is: 041faad1c93523fecdaf05e63c782e70eb2283a7

jonathanmuller, 2018-05-28:
@liuzfesp Thank you (and the team) for the fast correction !

closed 2018-05-28 by jonathanmuller