espressif/esp-idf #1752: ble_ibeacon sender don't work with iphones

ciccio88fcrlab, 2018-03-20:
Hi all,
I use ble_ibeacon to send advertising on my app, it works well with Android devices but with iPhones it doen't  work. Use to test iBeacon nRF Connect, in Android I see all the parameters, with IoS instead iBeacon is seen as a generic and unrecognized device. Can someone help me?

chegewara, 2018-03-20:
Hi, using iBeacons that are not produced or licensed by Apple is very hard with iPhones. Ive seen a lot issues about it, since i dont have iPhone (im lucky guy) i cant make tests, but ive read it may be related to advertising time parameters. Also Apple specs about iBeacons is very strict and you have follow all parameters to make it works, but not all can be found on internet.

https://github.com/nkolban/esp32-snippets/issues/348#issuecomment-365500832

ciccio88fcrlab, 2018-03-20:
Thanks for the reply, I tried various parameters including official ones, but without positive results.

Weijian-Espressif, 2018-03-22:
@ciccio88fcrlab, I test our ibeacon demo , it works well. you can test it with wechat APP "Yao Yi Yao Zhou Bian". or you can modify "#define ESP_UUID    {0xE2, 0xC5, 0x6D, 0xB5, 0xDF, 0xFB, 0x48, 0xD2, 0xB0, 0x60,0xD0, 0xF5, 0xA7, 0x10, 0x96, 0xE0}",  download "Dartle iBeacon Locator" APP from APP store and try.

ciccio88fcrlab, 2018-03-22:
Thank you so much for your answer. "Dartle iBeacon Locator" sees my iBeacons, but I still can not make them work with my App, the same App in the Android version works well, the same code (I use Phonegap) does not work on the iPhone, you have any idea on what can be the problem?

closed 2018-03-24 by ciccio88fcrlab