espressif/esp-idf #1753: Master @ cd54a35cae290df7def0207db7031644d9729ce4 and release/v3.0 build failure

vonnieda, 2018-03-20:
Master @ cd54a35cae290df7def0207db7031644d9729ce4 and release/v3.0 does not currently build:

```
/Users/jason/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c: In function 'btm_ble_process_adv_pkt':
/Users/jason/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c:3029:27: error: 'tBTM_SEC_BLE {aka struct <anonymous>}' has no member named 'current_addr_type'
             p_dev_rec->ble.current_addr_type = temp_addr_type;
                           ^
/Users/jason/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c:3030:34: error: 'tBTM_SEC_BLE {aka struct <anonymous>}' has no member named 'current_addr'
             memcpy(p_dev_rec->ble.current_addr, temp_bda, BD_ADDR_LEN);
                                  ^
/Users/jason/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_ble_gap.c:3031:27: error: 'tBTM_SEC_BLE {aka struct <anonymous>}' has no member named 'current_addr_valid'
             p_dev_rec->ble.current_addr_valid = true;
                           ^
CC build/bt/bluedroid/stack/btm/btm_pm.o
AR build/wpa_supplicant/libwpa_supplicant.a
make[1]: *** [bluedroid/stack/btm/btm_ble_gap.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/Users/jason/esp/esp-idf/components/bt/bluedroid/stack/btm/btm_pm.c:953:20: warning: 'mode_to_string' defined but not used [-Wunused-function]
 static const char *mode_to_string(tBTM_PM_MODE mode)
                    ^
make: *** [component-bt-build] Error 2
```

sdkconfig attached, in case it's needed:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/1830390/sdkconfig.txt)

davctv, 2018-03-21:
Hi,
I have fixed putting this code

#if (defined BLE_PRIVACY_SPT && BLE_PRIVACY_SPT == TRUE)
        //save current adv addr information if p_dev_rec!= NULL
        tBTM_SEC_DEV_REC *p_dev_rec = btm_find_dev (bda);
        if(p_dev_rec) {
            p_dev_rec->ble.current_addr_type = temp_addr_type;
            memcpy(p_dev_rec->ble.current_addr, temp_bda, BD_ADDR_LEN);
            p_dev_rec->ble.current_addr_valid = true;
        }
#endif

between line 3026 and 3033 of btm_ble_gap.c file.

In previuos version it was there, now master version doesn't have anymore.. that elements of the stucture are enable only if SMP is enabled.

hopkinskong, 2018-03-23:
Confirming this bug still exists as of current master.

vonnieda, 2018-03-23:
Bug still exists, and I just want to point out that this means anyone who is following the Getting Started guide will not be able to build a project.

hopkinskong, 2018-03-23:
@jack0c See if the pull request is OK?

jack0c, 2018-03-25:
@hopkinskong Sorry, In fact, already have a fix on both master and release/v3.0.

closed 2018-03-29 by FayeY