espressif/esp-idf #1754: esp_wifi_set_ps

Reiner1210, 2018-03-20:
I have a funny thing here

I use esp_wifi_get_ps(WIFI_PS_MODEM).
It works fine when the device is connected via a normal USB cable to the computer.
When I use a power only cable I does not work (go not to sleep). There is also a blue mini LED on the board blinking the whole time

Board is this one:
https://www.amazon.de/gp/product/B074RGW2VQ/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

Any idea whats going on ?
I have already tried to disable log outputs no success.

jack0c, 2018-03-21:
Any other difference?  Did you connect to an AP when you `use a power only cable`.

Reiner1210, 2018-03-21:
Yes the same in both cases.

I assume that the UART - USB chip intrrrupts the esp because he can not 
establish connection

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

TimXia, 2018-03-23:
@Reiner1210 Did your device connect to computor or outlet with power only cable?

Reiner1210, 2018-03-23:
I tried it with a power only cable on computer and 5V USB chrager. When I 
do the same with a full USB cabke it works fine. I assume that the open 
data wires from USB disturbe and it stucks in bootloader

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

closed 2018-03-23 by Reiner1210