espressif/esp-idf #1755: How about the packets capturing ability of esp32 promiscuous mode?

jixunx, 2018-03-21:
I did a experiment.Send packets from esp32,using esp_wifi_80211_tx.Another esp32 capture data in promiscuous mode.Data size per packet is 961 bytes. Counting how many packets been captured by esp32's promiscuous mode per 120 packets.It is showed that, esp32's promiscuous mode lost a lot of packets.Is it because of esp32's low capturing ability?Or another reason ?

[issue report.zip](https://github.com/espressif/esp-idf/files/1831746/issue.report.zip)

The attachment is the code,sdkconfig file,log file of sender project and promiscuous mode project.
Please review.
Best Regards!

jack0c, 2018-03-21:
1. ESP32 do not do not have hardware limit.
1. Current SDK do not capture CTRL packets, so you will lost about 50% packets. This feature will be merged into idf3.1.
1. Please do not print any thing when you receive a packet, it will stuck the packet in the receive buffer.

jixunx, 2018-03-21:
1. ESP32 do not do not have hardware limit.
2. Current SDK do not capture CTRL packets, so you will lost about 50% packets. This feature will be merged into idf3.1.
---------------------------------Is CTRL packets means WIFI_PKT_MGMT?Normally, grab the "WIFI_PKT_DATA" type package. Will there be a 50% packet loss rate?
3. Please do not print any thing when you receive a packet, it will block the packet in the receive buffer.
--------------------------------I did not print between packet and packet,and I printed when transimission is not busy.

jack0c, 2018-03-21:
ieee80211 packets include Data, Mgmt, Ctrl packets. Ctrl packets mainly include ACK,RTS,CTS.

TimXia, 2018-03-23:
@jixunx Did you use a third part tool to capture packet and compare the result with that of esp32? Pay attention that the result to be compared should only include the data packet that is sent from another esp32 and exclude other packet.

jixunx, 2018-03-23:
Yes,I did compare.I modify my code.According to the result of compare,the capture ability of esp32 is similar with third part tool.

closed 2018-03-29 by FayeY

LEOBELONI, 2020-04-16:
Does anyone solve this problem ? I have same scenario and % of lost packets in promiscuous mode.

esp-HarveyRong, 2020-04-16:
Hi @Lbeloni,
Can you elaborate on the problem you are experiencing? 
Do you use third-party tools to capture packets and compare the results with esp32?
BTW, captured packet is only a data packet sent from another esp32, there is no other data packet

Lbeloni, 2020-04-17:
Yes! Im comparing with Wireshark. 
Just sniffing probe requests packets and the % of lost packets  in ESP32 is almost 20%.
I used to turned off bluetooth with btStop() function and tag callback function with IRAM_ATTR but it doesn't works.

AmandaYaoYao, 2020-04-17:
>     2\. urrent SDK do not capture CTRL packets, so you will lost about 50% packets. This feature will be merged into idf3.1.
Hi,
*  you mentioned it cannot capture control packet, plz make sure that your filter has been set to "control" and the control subtype has been set. As I know the default  control subtype is somthing like NULL it means no control subtype is opened. BTW, "WIFI_PKT_MGMT" is something else, not the same thing.
* please do not do the print in you rx callback. 

* another advice, I have check your code, please try your rx callback without data filter like something

```
for(int i=0;i<pre_pack_num;i++)
				{
					printf("%d,",pack_seq[i]);
				}
```
Because I do not you do this has really  capture packets from your sender.j ust filter your recv mac is OK.

Lbeloni, 2020-04-17:
P

> * you mentioned it cannot capture control packet, plz make sure that your filter has been set to "control" and the control subtype has been set. As I know the default  control subtype is somthing [...]

I need a hand to understand that point, please ... 
Follow my filter in init function:

  wifi_promiscuous_filter_t filter;
  filter.filter_mask = WIFI_PROMIS_FILTER_MASK_MGMT;
  ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&filter));

liuzfesp, 2020-04-30:
Hi @Lbeloni , 

It will be helpful if you can provide following info:

- The whole code related to sniffer, including the wifi_promiscuous_cb_t handler
- What's the WiFi mode of the ESP32 that runs sniffer, does it connect to AP?
- Where are "probe requests packets" from? Are they sent by another ESP32 device via `esp_wifi_80211_tx()`, or you want to sniffer all probe request packets in the air?
- Does the ESP32 device that runs sniffer have the same channel number as the device that sends the "probe request packets"

archef2000, 2025-01-13:
When you have bt/ble enabled it steals the airtime from wifi.