espressif/esp-idf #1757: Proposal: ESP_LOGX

Reiner1210, 2018-03-21:
It would be nice if the macros of ESP_LOGX would be in curll brackets

at the moment I write

<code>
if ((err = esp_wifi_set_config(WIFI_IF_STA, &config)) != ESP_OK)
      {ESP_LOGE(tag, "esp_wifi_set_config (STA) failed with '%s (%d)'", esp_err_to_name(err), err);}
</code>

yf13, 2018-03-23:
It seems to me current one is better since it enforces clear coding style.

Reiner1210, 2018-03-24:
I think a nacro should behave like a function. If you do not know that it 
is a macro you can use it the wrong way.

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

piquan, 2018-03-26:
I personally agree with @Reiner1210; if a macro expands to multiple statements, it's usually best to make it a block. Otherwise, it can create silent bugs if you misuse the macro. If your project wants to require braces for `if` etc. all the time (for instance, for MISRA compliance requirements), there are many tools to enforce that without risking bugs.

The usual style for a function-like macro is to enclose it in `do { ... } while (0)`, to make sure the semicolon at the end gets removed as well; see https://gcc.gnu.org/onlinedocs/cpp/Swallowing-the-Semicolon.html .

igrr, 2018-03-26:
ESP_LOGx seems to expand to in `if (condition) { statements; }` block, so you should be able to use it in place of a single statement, similar to a do ... while construct. Are you seeing a compilation error or a warning when you don't have curly braces around ESP_LOGx?

FayeY, 2018-03-29:
Hi, did the above suggestion help you resolve the issue?

FayeY, 2018-04-04:
Closing due to lack of response, please feel free to reopen if this is still an issue.

Reiner1210, 2018-04-04:
I tried it - at the moment no errors. Have you changed it the last weeks. my first try some weeks ago faild

igrr, 2018-04-05:
If you look at ESP_LOGx macros, one thing they do is concatenating some strings together, which will not be possible if using an inline function.

Please feel free to reopen if you can reproduce this again.

ammaree, 2018-04-05:
@igrr @Reiner1210 
I previously raised the issue around ESP_LOGx in a thread [https://github.com/espressif/esp-idf/issues/464](url) focussing on printf and syslog, but the following points are important.

1: Logging to UART for debugging is fine, but ultimately logging to a syslog server is critical in a production environment where the 99.99% of the time no console is attached.

2: The ESC sequences appended to the messages are nice on a console but become a problem in a syslog environment where, depending on the logging system, the string often screw up the filter logic.

3: We believe the best solution is where the components forming the logging message, being priority , tag (message ID), format string and parameters, are passed onto a function. The function can then decide between console and/or syslog (port, UDP/TCP etc), colours etc.

We have achieved most of the above by
a) replacing the newlib printf modules with our own; and
b) replacing the ESP32 log module with our own module,
but cannot get rid of the trailing ESC sequences and various other items.

Current ESP_LOGx functionality is inflexible and more so since it is tightly integrated into the whole IDF and closed Wifi library. Makes getting clean IDF and Wifi library messages into a syslog system difficult.

Reiner1210, 2018-04-05:
@ammaree @igrr 

Hello,
I agree that log messages should not include ESC sequences. To make it nice and colorful the monitor application can do it recognizing the message type.

Reiner1210, 2018-04-19:
This does not work !!

		if (adapter == TCPIP_ADAPTER_IF_STA)
			ESP_LOGI(tag, "Station adapter info:");
		else
			{ESP_LOGI(tag, "Access point adapter info:");}

projectgus, 2018-04-19:
Hi @Reiner1210 ,

Sorry this was closed by mistake re: correct macro statement expansion. I'm going to reopen it

projectgus, 2018-04-20:
Hi @ammaree ,

Regarding your requirements of logging to an external server and disabling console ESC sequences:

* You can disable ANSI escape sequences in logs by un-setting this configuration item: http://esp-idf.readthedocs.io/en/latest/api-reference/kconfig.html#log-colors - Although I realise that this will not disable escape sequence formatting from the precompiled WiFi/etc libraries. I'll open another issue to track this.
* Redirecting log output from UART0 to a custom destination is supported via http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html#_CPPv219esp_log_set_vprintf14vprintf_like_t - calls from WiFi libraries, etc, should also end up in this hook function. Although it may not meet all the requirements you mentioned, there is some ability for customer filtering here because the first format arguments to the vprintf()-like function will be predictable - tag string, level as a character, timestamp.

> We have achieved most of the above by
> a) replacing the newlib printf modules with our own; and
> b) replacing the ESP32 log module with our own module,
> but cannot get rid of the trailing ESC sequences and various other items.

If there are specific issues with flexibility that you're seeing here then please open issues to report them. We want overriding the log component to work for more customised logging than is provided by the hooks mentioned above. It shouldn't be necessary to replace printf() for this to work, and I'd like to understand why this is so that we can fix the issue.

projectgus, 2018-04-20:
> Although I realise that this will not disable escape sequence formatting from the precompiled WiFi/etc libraries. I'll open another issue to track this.

Sorry, I should have checked before writing this. The WiFi libraries are compiled with CONFIG_LOG_COLORS disabled already, I guess for this exact reason. If you're seeing escape sequences in logs that aren't disabled by turning off CONFIG_LOG_COLORS, please open a new issue with details and we'll look into it.

igrr, 2018-04-20:
@projectgus regarding printf replacement, context is in this issue: https://github.com/espressif/esp-idf/issues/464.

projectgus, 2018-04-20:
Thanks @igrr, I'd missed this discussion and the link to the related issue entirely. Makes sense. I have one other suggestion which I'll post on the related thread.

closed 2018-04-26 by igrr

ankostis, 2022-06-13:
> In order to make this lot work we have replaced the esp-idf log component 

@ammaree do you plan to publish this code somehwere, eg some merge-request?