espressif/esp-idf #1759: SSL_pending always return 0

jas39, 2018-03-22:
I have a working SSL connection setup based on the Open_SSL server demo and works for smaller packages of data. I'm trying to have an non blocking read of all available data with the code below but SSL_pending is always 0. I think this is a bug.

`
while (SSL_pending(ssl)>0) {
  ret = SSL_read(ssl, recv_buf+endPek, OPENSSL_EXAMPLE_RECV_BUF_LEN - 1- endPek);
  if (ret>0) {
            endPek+=ret;
   } else break;
     ESP_LOGD(TAG, "SSL_read %d, endPek%d",ret,endPek);
     ESP_LOGD(TAG, "SSL pending: %d",SSL_pending(ssl));
}
`

negativekelvin, 2018-03-22:
https://stackoverflow.com/questions/28509593/ssl-pending-read-and-write-with-non-blocking-sockets

jas39, 2018-03-24:
The SSL_read in the above xample does NOT return all available data in the first call even though there's room in rec_buf

donghengqaz, 2019-08-01:
It maybe that there is really no applicant data in SSL RX  fragment buffer, so the function "SSL_pending" return 0, but TCP has cached some data, so when you call SSL_read which function will read the TCP data by socket and transform it to be SSL applicant data.

Alvin1Zhang, 2019-09-29:
@jas39 Thanks for the reporting and sorry for the slow turnaround. Would you please help if any updates of your issue? Has it been resolved? Thanks.

jas39, 2019-09-29:
I have worked around this some way, no longer sure how...  Feel free to close the issue if you think it's not an issue

Alvin1Zhang, 2019-10-12:
@jas39 Thanks for reporting, feel free to reopen if the issue still exists. Thanks.

closed 2019-10-12 by Alvin1Zhang