espressif/esp-idf #1762: Problem reading PID key from bonded device list in call to esp_ble_get_bond_device_list()

slivings99, 2018-03-22:
I am using the esp_gap_ble_api.h structures and esp_ble_get_bond_device_list and trying to read the static_addr from the pid key. Here is an example of my code:
```C
    int dev_num = esp_ble_get_bond_device_num();

    esp_ble_bond_dev_t *dev_list = (esp_ble_bond_dev_t *)malloc(sizeof(esp_ble_bond_dev_t) * dev_num);
    esp_ble_get_bond_device_list(&dev_num, dev_list);
    for (int i = 0; i < dev_num; i++) {
        esp_ble_pid_keys_t *deviceKey = &(dev_list[i].bond_key.pid_key);
        ESP_LOGI(TAG, "pid key addr type = %d", deviceKey->addr_type);
        ESP_LOGI(TAG, "pid static_addr:");
        ESP_LOGI(TAG, ESP_BD_ADDR_STR, ESP_BD_ADDR_HEX(deviceKey->static_addr));
        if (deviceKey->addr_type == BLE_ADDR_TYPE_RANDOM) {
            ESP_LOGI(TAG, "Adding private static address to white list.");
            esp_err_t result = esp_ble_gap_update_whitelist(true, deviceKey->static_addr);
            if (result != ESP_OK) {
                ESP_LOGI(TAG, "Error updating white list - %d.", result);
            }
        } else {
            ESP_LOGI(TAG, "Adding public address to white list.");
            esp_err_t result = esp_ble_gap_update_whitelist(true, dev_list[i].bd_addr);
            if (result != ESP_OK) {
                ESP_LOGI(TAG, "Error updating white list - %d.", result);
            }
        }
    }
    free(dev_list);
```
I am getting errant results as follows:
```
I (30619) gatts_controller: pid key addr type = 1830487040
I (30620) gatts_controller: pid static_addr:
I (30620) gatts_controller: 7e:2f:41:ce:ce:ce
I (30620) gatts_controller: Adding public address to white list.
I (30621) gatts_controller: pid key addr type = 741305344
I (30621) gatts_controller: pid static_addr:
I (30621) gatts_controller: e4:d1:27:ce:ce:ce
I (30622) gatts_controller: Adding public address to white list.
```
I believe this is related to a difference between the esp definitions and bluedroid definitions of this structure in esp_gap_ble_api.h
```C
typedef struct
{
    esp_bt_octet16_t          irk;           /*!< The irk value */
    esp_ble_addr_type_t       addr_type;     /*!< The address type */
    esp_bd_addr_t             static_addr;   /*!< The static address */
} __attribute__((packed)) esp_ble_pid_keys_t;                        /*!< The pid key type */
```
and in esp_bt_defs.h
```C
typedef enum {
    BLE_ADDR_TYPE_PUBLIC        = 0x00,
    BLE_ADDR_TYPE_RANDOM        = 0x01,
    BLE_ADDR_TYPE_RPA_PUBLIC    = 0x02,
    BLE_ADDR_TYPE_RPA_RANDOM    = 0x03,
} esp_ble_addr_type_t;
```
Because `esp_ble_addr_type_t` is defined as an enum typedef, it is getting compiled as a 32 bit int and causing the errant reads that I am getting. In bluedroid, this is defined as follows in btm_api.h:
```C
typedef struct {
    BT_OCTET16          irk;
    tBLE_ADDR_TYPE      addr_type;
    BD_ADDR             static_addr;
} tBTM_LE_PID_KEYS;
```
and tBLE_ADDR_TYPE in bt_types.h as
```C
#define BLE_ADDR_PUBLIC         0x00
#define BLE_ADDR_RANDOM         0x01
#define BLE_ADDR_PUBLIC_ID      0x02
#define BLE_ADDR_RANDOM_ID      0x03
typedef UINT8 tBLE_ADDR_TYPE;
```
So instead of an enum it is specifically defined as a uint8 and the enumerated values set as specific numbers. I believe there is therefore a corruption in the exchange happening in the call to btc_storage_get_bonded_ble_devices_list.

Weijian-Espressif, 2018-03-30:
@slivings99 thanks for your reminder, I will fix it the week. I will notify you when done, thanks.

Weijian-Espressif, 2018-04-02:
@slivings99  I have fixed the bug, please update the IDF and try, thanks.

closed 2018-04-04 by FayeY