espressif/esp-idf #1763: Using BLE SMP with white list

slivings99, 2018-03-22:
I am trying to use the white list feature and either I am doing something wrong or I misunderstand the purpose of the advertising filter. I would like to limit scans and connections to previously bonded devices. So I am adding the bda of each bonded device to the white list using the following code:
```C
int dev_num = esp_ble_get_bond_device_num();

esp_ble_bond_dev_t *dev_list = (esp_ble_bond_dev_t *)malloc(sizeof(esp_ble_bond_dev_t) * dev_num);
esp_ble_get_bond_device_list(&dev_num, dev_list);
for (int i = 0; i < dev_num; i++) {
      esp_err_t result = esp_ble_gap_update_whitelist(true, dev_list[i].bd_addr);
      if (result != ESP_OK) {
          ESP_LOGI(TAG, "Error updating white list - %d.", result);
      }
}
free(dev_list);
```
Then I set the advertising filter as follows:
```C
static esp_ble_adv_params_t master_adv_params = {
    .adv_int_min        = 0x0020,
    .adv_int_max        = 0x0040,
    .adv_type           = ADV_TYPE_IND,
    .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
    .channel_map        = ADV_CHNL_ALL,
    .adv_filter_policy    = ADV_FILTER_ALLOW_SCAN_WLST_CON_WLST;,
};

esp_ble_gap_start_advertising(&master_adv_params);
```
The behavior I am seeing is not what I would expect. If I use a legacy device that does not use privacy mode, the filtering seems to work and I can reconnect with a previously bonded device. However, if I am using a more recent version of Android with privacy mode enabled, the master device cannot see my device in scanning using nRF connect nor can it reconnect.

If, however, I set the filter to ADV_FILTER_ALLOW_SCAN_WLST_CON_ANY, I can connect with the Android with privacy mode.

What I can see is that the Android is using a consistent BDA even when in privacy mode because it has previously bonded with my device. Nevertheless, it does not see my device and cannot connect. Am I using the wrong bda for setting up the whitelist? I have tried to use a static address using the pid_key, but I ran up against a different problem expressed here https://www.esp32.com/viewtopic.php?f=14&t=4790.

Anyone else have success using white listing and bonded connections?

Weijian-Espressif, 2018-04-02:
Sorry, random address has this bug, we are repairing.

jkhax0r, 2018-04-02:
Thank you.  I am also interested in this so please keep us updated on when it might be pushed.

Weijian-Espressif, 2019-03-14:
Sorry, the whitelist only support public address and static random address without encryption.

copercini, 2019-06-06:
I'm also hitting this bug =/

My current workaround is to check if the device is in bonded list before connect, but would be better if the results of the scan were already filtered by the whitelist