espressif/esp-idf #1765: release/v3.0: components/esp32/pm_esp32.c:64:31: error 'default_freq' is used unitialized

eagi223, 2018-03-22:
I ran a git pull this morning and now I'm getting the following error message when running make:
```
...
CC build/driver/sdspi_crc.o
CC build/driver/pcnt.o
CC build/driver/timer.o
CC build/driver/gpio.o
CC build/driver/sdspi_host.o
CC build/driver/ledc.o
CC build/driver/periph_ctrl.o
CC build/driver/i2s.o
CC build/driver/sdspi_transaction.o
CC build/driver/uart.o
CC build/driver/sigmadelta.o
CC build/driver/spi_master.o
CC build/driver/spi_common.o
AR build/driver/libdriver.a
CC build/esp32/int_wdt.o
CC build/esp32/brownout.o
CC build/esp32/system_api.o
CC build/esp32/restore.o
CC build/esp32/dport_access.o
CC build/esp32/esp_timer.o
CC build/esp32/hw_random.o
CC build/esp32/pm_esp32.o
C:/esp-idf/components/esp32/pm_esp32.c: In function 'esp_pm_impl_init':
C:/esp-idf/components/esp32/pm_esp32.c:464:31: error: 'default_freq' is used uninitialized in this function [-Werror=uninitialized]
         s_cpu_freq_by_mode[i] = default_freq;
                               ^
cc1.exe: some warnings being treated as errors
make[1]: *** [/c/esp-idf/make/component_wrapper.mk:274: pm_esp32.o] Error 1
make: *** [/c/esp-idf/make/project.mk:450: component-esp32-build] Error 2

```

igrr, 2018-03-27:
@eagi223 could you please show the part of `make V=1` output related to compilation of `pm_esp32.c`? I'm interested in the compiler flags.

Just checked both debug and release builds, and didn't find the error you have mentioned.
One possible reason is that `assert` does not get expanded correctly. To check whether this is the case, you can copy&paste the compilation command for `pm_esp32.c` (it will be shown in `make V=1` output), replace `-c -o pm_esp32.o` flags with `-E -o pm_esp32.c`, and run it. This will put the preprocessed `pm_esp32.c` file into the current directory. You can then attach this preprocessed file to the issue. Thanks!

FayeY, 2018-04-04:
@eagi223 Any update about this issue? Is it solved now? 
If not, could you provide the information above, so we can help debug it?
Thanks.

eagi223, 2018-04-04:
Ah, sorry! I didn’t see that first reply. I’ll check on this tomorrow morning and post my results.

eagi223, 2018-04-11:
It is actually still happening with assertions turned on, so I'll pull the latest update, then give you the debug output if it is still occuring

eagi223, 2018-04-12:
This is the result of make V=1 with respect to pm_esp32 and the preprocessed.c file (zipped because oddly enough Github apparently doesn't support .c attachments.

I'm still having the issue occasionally,  but it's not happening at the moment, so I don't know if any of this is helpful or not.

[pm_esp32_preprocessed.zip](https://github.com/espressif/esp-idf/files/1903504/pm_esp32_preprocessed.zip)

```
xtensa-esp32-elf-gcc -std=gnu99 -Os -ggdb -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-old-style-declaration -DESP_PLATFORM -D IDF_VER=\"v3.0-rc1-243-gf8515688-dirty\" -MMD -MP   -DWITH_POSIX -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H -I /c/esp-idf/components/esp32/include -I /c/esp-idf/components/app_trace/include -I /c/esp-idf/components/app_update/include -I /c/esp-idf/components/bootloader_support/include -I /c/esp-idf/components/bt/include -I /c/esp-idf/components/bt/bluedroid/bta/include -I /c/esp-idf/components/bt/bluedroid/bta/sys/include -I /c/esp-idf/components/bt/bluedroid/btcore/include -I /c/esp-idf/components/bt/bluedroid/device/include -I /c/esp-idf/components/bt/bluedroid/gki/include -I /c/esp-idf/components/bt/bluedroid/hci/include -I /c/esp-idf/components/bt/bluedroid/osi/include -I /c/esp-idf/components/bt/bluedroid/utils/include -I /c/esp-idf/components/bt/bluedroid/external/sbc/decoder/include -I /c/esp-idf/components/bt/bluedroid/btc/core/include -I /c/esp-idf/components/bt/bluedroid/btc/profile/esp/blufi/include -I /c/esp-idf/components/bt/bluedroid/btc/profile/esp/include -I /c/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/include -I /c/esp-idf/components/bt/bluedroid/btc/profile/std/gap/include -I /c/esp-idf/components/bt/bluedroid/btc/profile/std/a2dp/include -I /c/esp-idf/components/bt/bluedroid/btc/profile/std/include -I /c/esp-idf/components/bt/bluedroid/btc/include -I /c/esp-idf/components/bt/bluedroid/btif/include -I /c/esp-idf/components/bt/bluedroid/stack/btm/include -I /c/esp-idf/components/bt/bluedroid/stack/btu/include -I /c/esp-idf/components/bt/bluedroid/stack/gap/include -I /c/esp-idf/components/bt/bluedroid/stack/gatt/include -I /c/esp-idf/components/bt/bluedroid/stack/hcic/include -I /c/esp-idf/components/bt/bluedroid/stack/l2cap/include -I /c/esp-idf/components/bt/bluedroid/stack/sdp/include -I /c/esp-idf/components/bt/bluedroid/stack/smp/include -I /c/esp-idf/components/bt/bluedroid/stack/avct/include -I /c/esp-idf/components/bt/bluedroid/stack/avrc/include -I /c/esp-idf/components/bt/bluedroid/stack/avdt/include -I /c/esp-idf/components/bt/bluedroid/stack/a2dp/include -I /c/esp-idf/components/bt/bluedroid/stack/include -I /c/esp-idf/components/bt/bluedroid/utils/include -I /c/esp-idf/components/bt/bluedroid/api/include -I /c/esp-idf/components/bt/bluedroid/include -I /c/esp-idf/components/coap/port/include -I /c/esp-idf/components/coap/port/include/coap -I /c/esp-idf/components/coap/libcoap/include -I /c/esp-idf/components/coap/libcoap/include/coap -I /c/esp-idf/components/console/. -I /c/esp-idf/components/cxx/include -I /c/esp-idf/components/driver/include -I /c/esp-idf/components/esp_adc_cal/include -I /c/esp-idf/components/ethernet/include -I /c/esp-idf/components/expat/port/include -I /c/esp-idf/components/expat/include/expat -I /c/esp-idf/components/fatfs/src -I /c/esp-idf/components/freertos/include -I /c/esp-idf/components/heap/include -I /c/esp-idf/components/idf_test/include -I /c/esp-idf/components/jsmn/include/ -I /c/esp-idf/components/json/cJSON -I /c/esp-idf/components/libsodium/libsodium/src/libsodium/include -I /c/esp-idf/components/libsodium/port_include -I /c/esp-idf/components/log/include -I /c/esp-idf/components/lwip/include/lwip -I /c/esp-idf/components/lwip/include/lwip/port -I /c/esp-idf/components/lwip/include/lwip/posix -I /c/esp-idf/components/lwip/apps/ping -I /c/rph_esp32_engine_alert_lock/main/include -I /c/esp-idf/components/mbedtls/port/include -I /c/esp-idf/components/mbedtls/include -I /c/esp-idf/components/mdns/include -I /c/esp-idf/components/micro-ecc/micro-ecc -I /c/esp-idf/components/newlib/platform_include -I /c/esp-idf/components/newlib/include -I /c/esp-idf/components/nghttp/port/include -I /c/esp-idf/components/nghttp/nghttp2/lib/includes -I /c/esp-idf/components/nvs_flash/include -I /c/esp-idf/components/openssl/include -I /c/esp-idf/components/pthread/include -I /c/esp-idf/components/sdmmc/include -I /c/esp-idf/components/soc/esp32/include -I /c/esp-idf/components/soc/include -I /c/esp-idf/components/spi_flash/include -I /c/esp-idf/components/spiffs/include -I /c/esp-idf/components/tcpip_adapter/include -I /c/esp-idf/components/ulp/include -I /c/esp-idf/components/vfs/include -I /c/esp-idf/components/wear_levelling/include -I /c/esp-idf/components/wpa_supplicant/include -I /c/esp-idf/components/wpa_supplicant/port/include -I /c/esp-idf/components/wpa_supplicant/../esp32/include -I /c/esp-idf/components/xtensa-debug-module/include -I /c/rph_esp32_engine_alert_lock/build/include  -I. -c /c/esp-idf/components/esp32/./pm_esp32.c -o pm_esp32.o
echo "/c/esp-idf/components/esp32/./pm_esp32.c:" >> pm_esp32.d
```

igrr, 2018-04-13:
The preprocessed file you attached matches earlier version of pm_esp32.c (with no `for` loop in `esp_pm_impl_init`). I'm not sure what does "occasionally" mean in your case (i.e. you are occasionally using a different version of IDF?) but if you see the issue happening again (i.e. compilation error), please repeat the step to generate preprocessed file and attach it again. Thank you.

closed 2018-09-11 by eagi223