espressif/esp-idf #1766: Standalone

Reiner1210, 2018-03-22:
Is it possible that dev boards do not start when not connected to USB port ?
I see that when it is full connected with USB it works and when only the power cables are present the device seems not to boot

Reiner1210, 2018-03-22:
I think it was my multimeter ....if I connect the power wires directly it works ....

negativekelvin, 2018-03-23:
What are you connecting to?  Maybe some "smart chargers" which use the data lines to detect supported charging current can mess up reset circuit?

projectgus, 2018-03-23:
Hi @Reiner1210 ,

What development board do you have?

There have been reports of some third party boards where the serial control lines are wired up such that a serial program has to be open for it to run. I haven't heard of this just for USB. Maybe check the board is receiving adequate voltage from the USB source as well?

Reiner1210, 2018-03-23:
It is this
https://www.amazon.de/gp/aw/d/B074RGW2VQ/ref=yo_ii_img?ie=UTF8&psc=1

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

Reiner1210, 2018-03-23:
The problem is that it works withe the data lines connected.

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

Reiner1210, 2018-03-23:
I have some news from it:

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:7820
load:0x40078000,len:0
load:0x40078000,len:14116
entry 0x400793e4
I (28) boot: ESP-IDF  2nd stage bootloader
I (28) boot: compile time 00:02:37
I (28) boot: Enabling RNG early entropy source...
D (33) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
D (39) boot: magic e9
D (42) boot: segments 04
D (44) boot: spi_mode 02
D (47) boot: spi_speed 00
D (50) boot: spi_size 02
I (52) boot: SPI Speed      : 40MHz
I (56) boot: SPI Mode       : DIO
I (60) boot: SPI Flash Size : 4MB
D (64) bootloader_flash: mmu set paddr=00000000 count=1
D (70) boot: mapped partition table 0x8000 at 0x3f408000
D (75) flash_parts: partition table verified, 4 entries
I (80) boot: Partition Table:
I (84) boot: ## Label            Usage          Type ST Offset   Length
D (91) boot: load partition table entry 0x3f408000
D (96) boot: type=1 subtype=2
I (99) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (107) boot: load partition table entry 0x3f408020
D (112) boot: type=1 subtype=1
I (115) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (122) boot: load partition table entry 0x3f408040
D (127) boot: type=0 subtype=0
I (130) boot:  2 factory          factory app      00 00 00010000 00100000
I (138) boot: End of partition table
D (142) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (148) esp_image: reading image header @ 0x10000
D (153) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (160) esp_image: image header: 0xe9 0x07 0x02 0x04 40080eac
V (166) esp_image: loading segment header 0 at offset 0x10018
V (172) esp_image: segment data length 0x10f9c data starts 0x10020
V (178) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3f400020
I (186) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x10f9c ( 69532) map
D (195) bootloader_flash: mmu set paddr=00010000 count=2
V (224) esp_image: loading segment header 1 at offset 0x20fbc
D (224) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (226) esp_image: segment data length 0x2ca4 data starts 0x20fc4
V (232) esp_image: segment 1 map_segment 0 segment_data_offs 0x20fc4 load_addr 0x3ffb0000
I (240) esp_image: segment 1: paddr=0x00020fc4 vaddr=0x3ffb0000 size=0x02ca4 ( 11428) load
D (249) bootloader_flash: mmu set paddr=00020000 count=1
V (259) esp_image: loading segment header 2 at offset 0x23c68
D (260) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (267) esp_image: segment data length 0x400 data starts 0x23c70
V (273) esp_image: segment 2 map_segment 0 segment_data_offs 0x23c70 load_addr 0x40080000
0x40080000: _iram_start at /home/reiner/data/programm/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (281) esp_image: segment 2: paddr=0x00023c70 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/reiner/data/programm/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

D (290) bootloader_flash: mmu set paddr=00020000 count=1
V (296) esp_image: loading segment header 3 at offset 0x24070
D (302) bootloader_flash: mmu set block paddr=0x00020000 (was 0xffffffff)
V (308) esp_image: segment data length 0xbf98 data starts 0x24078
V (315) esp_image: segment 3 map_segment 0 segment_data_offs 0x24078 load_addr 0x40080400
I (323) esp_image: segment 3: paddr=0x00024078 vaddr=0x40080400 size=0x0bf98 ( 49048) load
D (332) bootloader_flash: mmu set paddr=00020000 count=2
V (357) esp_image: loading segment header 4 at offset 0x30010
[... 143 lines cut ...]
D (500) nvs: nvs_get ap.phybw 1
D (500) nvs: nvs_get ap.sndchan 1
D (510) nvs: nvs_set_blob sta.mac 6
D (510) nvs: nvs_set_blob ap.mac 6
I (520) wifi: Init dynamic tx buffer num: 32
I (520) wifi: Init data frame dynamic rx buffer num: 32
I (520) wifi: Init management frame dynamic rx buffer num: 32
I (530) wifi: wifi driver task: 3ffb956c, prio:23, stack:4096
I (530) wifi: Init static rx buffer num: 10
I (540) wifi: Init dynamic rx buffer num: 32
I (540) wifi: wifi power manager task: 0x3ffc4c04 prio: 21 stack: 2560
D (550) RTC_MODULE: Wi-Fi takes adc2 lock.
D (550) phy_init: loading PHY init data from application binary
D (560) nvs: nvs_open_from_partition phy 0
D (560) nvs: nvs_get cal_version 4
V (560) phy_init: phy_get_rf_cal_version: 383

D (570) nvs: nvs_get_str_or_blob cal_mac
D (570) nvs: nvs_get_str_or_blob cal_data
D (590) nvs: nvs_close 4
V (590) phy_init: register_chipv7_phy, init_data=0x3f4106ac, cal_data=0x3ffc4dc0, mode=0

Brownout detector was triggered

ets Jun  8 2016 00:22:57

Reiner1210, 2018-03-23:
The following line helps

WRITE_PERI_REG(RTC_CNTL_BROWN_OUT_REG, 0); //disable brownout detector

negativekelvin, 2018-03-23:
Brownout means not strong enough power supply.  You didn't say what you use as power supply and how you connect it to board.

Reiner1210, 2018-03-23:
I have tried USB power supplies ~1-2 A, Computer based USB ports. It depends on the cable.

Reiner1210, 2018-04-04:
I can switch of brownout in make menuconfig. All works fine.
I think it is the wifi_enable which lowers the voltage - perhaps the brownout should be less sensible ?

projectgus, 2018-12-04:
Hi @Reiner1210 ,

Sorry for the long delay before someone got back to you on this.

The default [brownout settings](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/kconfig.html#config-brownout-det) have been changed in recent IDF versions, the detector is now set to the lowest configurable level (2.43V) by default.

If you're still getting brownouts at this level, something is definitely wrong with the power supplied to the ESP32 - it could be the development board doesn't have sufficient capacitance in the 3.3V power rail for the chip, the USB cable could be faulty and isn't delivering current reliably, or the USB port may have some issue supplying current.

The chip may work correctly with the brownout detector off, but there may also be subtle bugs because the chip is now running out of specification with glitches of too low supply voltage.

Closing the issue because this is not something we can fix in ESP-IDF.

closed 2018-12-04 by projectgus