espressif/esp-idf #1771: Guru Meditation Error of type StoreProhibited occurred on core  1

Natalie88, 2018-03-24:
Hi I am trying to change the value of this register RTC_CNTL_WIFI_FORCE_PD to force the WiFi to stay down even after a reboot happens. I am using this function WRITE_PERI_REG(0x00020000, 1); but I am getting the following error
`Guru Meditation Error of type StoreProhibited occurred on core  1. Exception 
was unhandled.
Register dump:
PC      : 0x400d1245  PS      : 0x00060c30  A0      : 0x801174fe  A1      : 
0x3ffd53e0  
A2      : 0x00020000  A3      : 0x00000001  A4      : 0x00000025  A5      : 
0x3ffd13b8  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d123f  A9      : 
0x3ffd53c0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3f400f33  A13     : 
0x00000000  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 
0x0000001d  
EXCVADDR: 0x00020000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 
0xffffffff  

Backtrace: 0x400d1245:0x3ffd53e0 0x401174fb:0x3ffd5400`

negativekelvin, 2018-03-24:
RTC_CNTL_WIFI_FORCE_PD is not a register.

`SET_PERI_REG_MASK(RTC_CNTL_DIG_PWC_REG, RTC_CNTL_WIFI_FORCE_PD);`

Natalie88, 2018-03-25:
@negativekelvin Thank you very much for your help, the ESP32 in my case ( it sense then goes into deep sleep then wakeup sense again ....) until it reaches the required numbers of sample it reports using the WiFi to a local server. 
 by default every time it wakes up from the deep Sleep it turns the Radio on again because it will be switched off during the deep Sleep, However I don't need the Radio for the sensing task "it increases the energy consumption" what I want to accomplish is to stop the ESP from turning the Radio on when its going to boot from the deep Sleep if the next task is sensing and turn it on when only the next task is reporting 
and since this Flag RTC_CNTL_WIFI_FORCE_PD  is an RTC flag so it should survive the deepSleep so if I switch it on the WiFi should never turned on after a deepSleep until I change it state. 
This what I thought, and please forgive me I am a beginner.

igrr, 2018-03-25:
ESP32 doesn't switch on WiFi radio unless you call esp_wifi_start. So you don't need to do anything special with the RTC registers; just make sure esp_wifi_start is not called unless you need WiFi.

Natalie88, 2018-03-25:
@igrr to me you are like Christiano Rolando to football players, Thank you for your contribution.
Is there any way to reduce the power needed to boot the ESP32

igrr, 2018-03-25:
To reduce the total energy consumed by the chip, you should reduce one or more of: voltage, current, run time. Reducing the voltage gives you only a very limited reduction of power. Reducing the  current can be done by setting lower CPU frequency and/or enabling dynamic frequency scaling. Run time can be reduced by closing unnecessary log output, enabling QIO mode and increasing frequency of the flash memory to 80MHz. All these configuration changes can be done if you use Arduino-esp32 as an ESP-IDF component, which is described in Arduino-esp32 project readme.

Natalie88, 2018-03-25:
@igrr I tried the following code but I am getting ESP_ERR_INVALID_ARG

`esp_pm_config_esp32_t pm_config = {
   .max_cpu_freq =RTC_CPU_FREQ_160M,
   .min_cpu_freq = RTC_CPU_FREQ_XTAL,
};

void setup() {
  esp_err_t ret;
  if((ret = esp_pm_configure(&pm_config)) != ESP_OK) {
      printf("pm config error %s\n", \
              ret == ESP_ERR_INVALID_ARG ? \
              "ESP_ERR_INVALID_ARG":"ESP_ERR_NOT_SUPPORTED");
  }
}`

negativekelvin, 2018-03-26:
https://stackoverflow.com/questions/18731707/why-does-c11-not-support-designated-initializer-lists-as-c99

negativekelvin, 2018-03-26:
In c++ you have to initialize all the members of pm_config

Natalie88, 2018-03-26:
@negativekelvin I have init all of them now and still getting the same error 
esp_pm_config_esp32_t pm_config = {
   .max_cpu_freq =RTC_CPU_FREQ_160M ,
   .min_cpu_freq = RTC_CPU_FREQ_XTAL ,
   .light_sleep_enable = false,
};

Natalie88, 2018-03-26:
I have tried also to change the Freq from boards.txt but according to my measurements I still have the same power consumption after making it 
esp32.build.f_cpu=80000000L

closed 2018-04-03 by Natalie88