espressif/esp-idf #1773: BT mic/speaker

mingshima, 2018-03-26:
Hi, all. I am trying to do a project of BT mic/speaker, I have tried spp examples, the spp initator collects the voice by using adc1_get_voltage(), and then transmit the data to the speaker, but it seems to work poorly. Each time, the mic send an array and the size is 10 or the other, and then the speaker receives the array and output each data, but the voice is not smooth, the sampling speed is about 1KHz, I think it's enough. Can anyone help me to find the reason or give some suggestion? Many thanks.

the mic
```
if (param->write.cong == 0) {
			for (int i = 0; i < SPP_DATA_LEN; ++i) {
			int val = adc1_get_voltage(ADC1_CHANNEL_5);
			int output = val * 255 / 4095;
			spp_data[i] = output & 0xff;
			//printf("%5d",spp_data[i]);
			}
            esp_spp_write(param->write.handle, SPP_DATA_LEN, spp_data);
        }
```
the speaker
```
for(volatile int i = 0; i < param->data_ind.len; i++)
		{
			dac_out_voltage(DAC_CHANNEL_1, param->data_ind.data[i]);
		}

```
the log of initiator
```
I (5648) SPP_INITIATOR_DEMO: ESP_SPP_WRITE_EVT len=10 cong=0
I (5648) : 95 95 95 95 94 95 95 95 95 95
I (5658) SPP_INITIATOR_DEMO: ESP_SPP_WRITE_EVT len=10 cong=0
I (5658) : 95 95 95 96 95 95 96 95 95 94
I (5668) SPP_INITIATOR_DEMO: ESP_SPP_WRITE_EVT len=10 cong=0
I (5668) : 95 96 95 95 95 96 95 94 96 95
I (5678) SPP_INITIATOR_DEMO: ESP_SPP_WRITE_EVT len=10 cong=0
I (5678) : 95 96 95 95 90 96 96 95 90 95
I (5688) SPP_INITIATOR_DEMO: ESP_SPP_WRITE_EVT len=10 cong=0
I (5688) : 96 95 95 95 99 95 95 96 95 92
```

the log of acceptor
```
I (326208) SPP_ACCEPTOR_DEMO: ESP_SPP_DATA_IND_EVT len=10 handle=129
I (326208) : 97 97 97 97 97 97 97 97 97 97
I (326218) SPP_ACCEPTOR_DEMO: ESP_SPP_DATA_IND_EVT len=10 handle=129
I (326228) : 8f 97 97 97 97 97 97 97 97 96
I (326228) SPP_ACCEPTOR_DEMO: ESP_SPP_DATA_IND_EVT len=10 handle=129
I (326238) : 97 96 96 97 96 97 97 97 96 96
I (326238) SPP_ACCEPTOR_DEMO: ESP_SPP_DATA_IND_EVT len=10 handle=129
I (326248) : 9c 96 97 94 97 97 96 97 97 96
I (326248) SPP_ACCEPTOR_DEMO: ESP_SPP_DATA_IND_EVT len=10 handle=129
I (326258) : 96 95 97 96 95 96 96 96 9b 96
I (326268) SPP_ACCEPTOR_DEMO: ESP_SPP_DATA_IND_EVT len=10 handle=129
I (326268) : 96 96 97 96 96 96 96 96 96 96
```

Weijian-Espressif, 2018-03-28:
@blueMoodBHD help to view the problem.

blueMoodBHD, 2018-03-29:
@mingshima Hello, SPP realizes passthrough of data and without real-time performance, it is not suitable for streaming media transmission. 
You can try to use a2dp_sink and a2dp_source , or add caching and packet loss mechanisms yourself.

mingshima, 2018-03-29:
@blueMoodBHD  Thanks for reply, I tried a2dp example and they work well, but now question comes. In a2dp_source we use the function below to generate noise, if I want to collect the signal from a microphone,  what is the format of the data, as the signal of microphone is voltage value. If I use adc1_get_voltage() to get the value, how should I transform it into the value I need. I simply use 
`int val = adc1_get_voltage(ADC1_CHANNEL_5);` but can't get the result I want.

```
static int32_t bt_app_a2d_data_cb(uint8_t *data, int32_t len)
{
    if (len < 0 || data == NULL) {
        return 0;
    }

    // generate random sequence
    int val = rand() % (1 << 16);
    for (int i = 0; i < (len >> 1); i++) {
        data[(i << 1)] = val & 0xff;
        data[(i << 1) + 1] = (val >> 8) & 0xff;
    }

    return len;
}
```

blueMoodBHD, 2018-03-30:
You cloud try as follows:
```
static int32_t bt_app_a2d_data_cb(uint8_t *data, int32_t len)
{
    if (len < 0 || data == NULL) {
        return 0;
    }
    int val, output;
    for (int i = 0; i < (len); i++) {
        val = adc1_get_voltage(ADC1_CHANNEL_5);
        output = val * 255 / 4095; //Some of your processing
        data[i] = output & 0xff;
    return len;
}
```
But the sampling speed in a2dp is  44.1KHz, yours is about 1KHz, may causing some errors.

By the way, if you want to use SPP, Increasing the value of `SPP_DATA_LEN` may be effective

FayeY, 2018-04-04:
Hi @mingshima , did the above suggestion help you resolve the issue?

mingshima, 2018-04-04:
Hi, @FayeY , thanks for your suggestion, I have tried to increase the value of  SPP_DATA_LEN, but didn't get a better result. And in your code, as in the sink bits_per_sample = 16, so I think the code should be as below
```
static int32_t bt_app_a2d_data_cb(uint8_t *data, int32_t len)
{
    if (len < 0 || data == NULL) {
        return 0;
    }

    // generate random sequence
    adc1_config_width(ADC_WIDTH_12Bit);
	adc1_config_channel_atten(ADC1_CHANNEL_5,ADC_ATTEN_0db);
	int val = adc1_get_voltage(ADC1_CHANNEL_5);	
	val = val * 255 / 4095;
	val = val % (1 << 16);
    for (int i = 0; i < (len >> 1); i++) {
		data[(i << 1)] = val & 0xff;
        data[(i << 1) + 1] = (val >> 8) & 0xff;
    }

    return len;
}
```
when I tried these code, there are a lot of noise. Unfortunately, my microphone was broken I bought a new one and have to wait it to come. By the way, I have to study I2S to know how it works.

blueMoodBHD, 2018-05-04:
Hi, @mingshima Does it work well, now?

closed 2018-06-13 by FayeY