espressif/esp-idf #1774: Smart config example fails to connect to wlan

eflukx, 2018-03-26:
The `smart_config` example (using esp-idf commit 8478823) runs and install fine. The correct SSID and PASSWORD is received, but after that connecting to wlan seems to fail with output below.
I tried it on two ESP32 modules, both have same result. (The `simple_wifi` example works fine on both modules.)

```
I (305) cpu_start: Pro cpu up.
I (308) cpu_start: Starting app cpu, entry point is 0x40080f74
0x40080f74: call_start_cpu1 at /home/eflukx/code/esp32/esp-idf/components/esp32/./cpu_start.c:225

I (0) cpu_start: App cpu up.
I (319) heap_init: Initializing. RAM available for dynamic allocation:
I (326) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (332) heap_init: At 3FFB9090 len 00026F70 (155 KiB): DRAM
I (338) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (344) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (351) heap_init: At 4008E858 len 000117A8 (69 KiB): IRAM
I (357) cpu_start: Pro cpu start user code
I (151) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (194) wifi: wifi firmware version: c202b34
I (194) wifi: config NVS flash: enabled
I (194) wifi: config nano formating: disabled
I (194) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (204) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (224) wifi: Init dynamic tx buffer num: 32
I (224) wifi: Init data frame dynamic rx buffer num: 32
I (224) wifi: Init management frame dynamic rx buffer num: 32
I (224) wifi: Init static tx buffer num: 16
I (234) wifi: wifi driver task: 3ffc7018, prio:23, stack:4096
I (234) wifi: Init static rx buffer num: 10
I (244) wifi: Init dynamic rx buffer num: 32
I (244) wifi: wifi power manager task: 0x3ffcbc58 prio: 21 stack: 2560
I (314) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (314) wifi: mode : sta (24:0a:c4:8d:e6:58)
I (364) smartconfig: SC version: V2.7.1
I (5184) wifi: ic_enable_sniffer
I (5184) sc: SC_STATUS_FINDING_CHANNEL
I (6294) smartconfig: TYPE: ESPTOUCH
I (6294) smartconfig: T|AP MAC: d8:50:e6:d8:ea:e8
I (6294) sc: SC_STATUS_GETTING_SSID_PSWD
I (7944) smartconfig: T|pswd: m************t
I (7944) smartconfig: T|ssid: i********Q
I (7944) smartconfig: T|bssid: d8:50:e6:d8:ea:e8
I (7944) wifi: ic_disable_sniffer
I (7944) sc: SC_STATUS_LINK
I (7954) sc: SSID:iottaHQ
I (7954) sc: PASSWORD:maakmijnieuit
W (7974) wifi: lmac stop, discard packet
I (8094) wifi: n:4 0, o:1 0, ap:255 255, sta:4 0, prof:1
I (8744) wifi: state: init -> auth (b0)
W (8754) wifi: lmac stop, discard packet
I (9754) wifi: state: auth -> init (2)
I (9754) wifi: n:4 0, o:4 0, ap:255 255, sta:4 0, prof:1
W (9754) wifi: lmac stop, discard packet
I (9874) wifi: n:4 0, o:4 0, ap:255 255, sta:4 0, prof:1
I (9874) wifi: state: init -> auth (b0)
W (9874) wifi: lmac stop, discard packet
I (10874) wifi: state: auth -> init (2)
I (10874) wifi: n:4 0, o:4 0, ap:255 255, sta:4 0, prof:1
W (10874) wifi: lmac stop, discard packet
I (10994) wifi: n:4 0, o:4 0, ap:255 255, sta:4 0, prof:1
I (10994) wifi: state: init -> auth (b0)
W (10994) wifi: lmac stop, discard packet

**snip** (--etc etc etc--)
```

TimXia, 2018-03-27:
@eflukx This is a bug. We have found the root cause and will fix it this week.

eflukx, 2018-03-27:
@TimXia Thanks for the feedback!
Please post the commit nr that includes the fix, interested to see what was wrong 😃

perll, 2018-03-30:
@eflukx
Sr, is that bug fixed?
Regards.

TimXia, 2018-04-01:
@eflukx @perll The bugfix is merging. Due to the gitlab server is maintaining this weekend, it will be merged next monday.

perll, 2018-04-01:
Guru Meditation Error: Core  1 panic'ed (StoreProhibited)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x4008f22e  PS      : 0x00050033  A0      : 0x4008f137  A1      : 0x3ffd4860  
0x4008f22e: vPortYieldFromInt at /home/raimundo/esp/esp-idf/components/freertos/./portasm.S:595

0x4008f137: _frxt_int_exit at /home/raimundo/esp/esp-idf/components/freertos/./portasm.S:206

A2      : 0x0d4ab040  A3      : 0x00000000  A4      : 0x00000212  A5      : 0x3ffc0be0  
A6      : 0x00000001  A7      : 0x80000001  A8      : 0x40082509  A9      : 0x00000001  
0x40082509: _xt_lowint1 at /home/raimundo/esp/esp-idf/components/freertos/./xtensa_vectors.S:1098

A10     : 0x00060021  A11     : 0x00000000  A12     : 0x8008d47a  A13     : 0x3ffc0bd0  
A14     : 0x3ffc0c18  A15     : 0x00000001  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0d4ab040  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008f22e:0x3ffd4860 0x4008f134:0x3ffd4870
0x4008f22e: vPortYieldFromInt at /home/raimundo/esp/esp-idf/components/freertos/./portasm.S:595

0x4008f134: _frxt_int_exit at /home/raimundo/esp/esp-idf/components/freertos/./portasm.S:205

TimXia, 2018-04-03:
@eflukx The bug has been fixed in the latest IDF master branch. Please check it.

TimXia, 2018-04-03:
@perll We do not recommend that HTTP_REQUEST and SMART_CONFIG work together. Better to use HTTP_REQUEST and esp_wifi_scan() to scan for beacon and send them by web.

eflukx, 2018-04-03:
@perll Thanks, I will check it out. Closing the issue.

closed 2018-04-03 by eflukx

perll, 2018-04-03:
Preliminary. until yesterday, after a few minutes running, the log already showed memory errors. Today, after upgrading with the master, 30 minutes later and no memory errors. Continuing with the tests!
@TimXia , TKU a ton!!!!

perll, 2018-04-05:
After get the new master (3.0), the heap memory issues was solved in my code but, now, with HTTP_REQUEST example, I'm getting:
I (2140007) event: station ip lost. After few minutes running. I've ran in four esp roow32, 5.1V 850mA; 5.1V 700mA; USB; 5V 1000mA. All whit the same resuult error:
I (2140007) event: station ip lost.
If I get a downgrade, I will solve that issue but get back memory leak. Please, could somebody help me?

perll, 2018-04-05:
I (2074527) event: sta ip: 192.168.0.30, mask: 255.255.255.0, gw: 192.168.0.1
I (2074527) WIFI...: Connected to AP
I (2074527) WIFI...: DNS lookup succeeded. IP=50.30.38.132
I (2074527) WIFI...: ... allocated socket
I (2080437) wifi: bcn_timout,ap_probe_send_start
I (2082937) wifi: ap_probe_send over, resett wifi status to disassoc
I (2082937) wifi: state: run -> init (1)
I (2082937) wifi: pm stop, total sleep time: 0/14780518

I (2082937) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
**I (2140007) event: station ip lost**

TimXia, 2018-04-05:
@perll Did you use sniffer(promiscuous) mode? If so, Please change to home channel which is the channel of the AP that station is connected to for somewhile between sniffer two different channels.

eflukx, 2018-04-05:
@perll Please open another issue for another issue. The current discussion seems to be off topic IMO.

perll, 2018-04-05:
I (2074527) event: sta ip: 192.168.0.30, mask: 255.255.255.0, gw: 192.168.0.1
I (2074527) WIFI...: Connected to AP
I (2074527) WIFI...: DNS lookup succeeded. IP=50.30.38.132
I (2074527) WIFI...: ... allocated socket
I (2080437) wifi: bcn_timout,ap_probe_send_start
I (2082937) wifi: ap_probe_send over, resett wifi status to disassoc
I (2082937) wifi: state: run -> init (1)
I (2082937) wifi: pm stop, total sleep time: 0/14780518

I (2082937) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (2140007) event: station ip lost

perll, 2018-04-05:
**''event: station ip lost''**
Could somebody help me with this issue? I'm not in PROMISCUOS MODE. it is FALSE.
I'v been working on a code that I join HTTP_REQUEST and GATT_CLIENTE.
My goal is scan for beacons and send them throught a WIFI conn to my web server. It is already working but just  for 10' 15'. Regards.

perll, 2018-04-06:
Sir,

"(...)HTTP_REQUEST and esp_wifi_scan() to scan for beacon(...)"
I did not undertand what you meant. Scan for beacons near by me using
HTTP_REQUEST ?
Could you point me a example?

Regards

TimXia, 2018-04-06:
@perll You said `I've beem work on a app that is a mix from GATT_CLIENTE, HTT_REQUEST and wifi SMART_CONFIG.`. So, I was thinking that you use wifi SMART_CONFIG to scan for beacon. What did you use to scan for beacon? BTW, the issue seems to be related with WiFi and BT/BLE coexistence which  works better on the latest master branch of IDF. It is recommended to update IDF to the latest version. Please have a try.

perll, 2018-04-06:
Sir, I've updated my workspace with the last master. All failures have been
resolved. The question now is:

E (35781) WIFI ...: ... socket connect failed errno = 113
Address 184
Eu (2140007) event: ip station lost

I'm not working on Promiscuous mode.

I have used the sample codes:
1 - HTTP_REQUEST - for request
2 - GATT_CLIENT - for BLE scanning

I deleted the SMART_CONFIG code, as you told me.

Now, after 15 '20' after the start, I get this message and the ESP is
stopped.
Do you have any tips to make it work together: HTTP_REQUEST and GATT_CLIENT.

Regards

perll, 2018-04-06:
Just to add
In the last 2 days, I was running an HTTP_REQUEST and, in the GET method, I
added 500 bytes of data. It's working fine. without GATT_CLIENT.

TimXia, 2018-04-07:
@perll The issue is caused by the bug of WiFi and BLE coexistence. I suggest to update with the latest IDF which improves the stability of WiFi and BLE coexistence.

perll, 2018-04-07:
Ok. Updated with master <https://github.com/espressif/esp-idf> Updated 2
days ago by Spritetm <https://github.com/Spritetm>
Running. I will keeping you updated about the test results.

By the way, do you know what mean:

I (419833) WIFI...: ... done reading from socket. Last read return=0
errno=128

I've been getting this after each request  response

Regards

perll, 2018-04-07:
Sir,

good news (almost there!)

Tests results from the last 9h:

*Scenario*:
4 esp wroom32 running,  at same time,  for 9 hours.
Only 1 processor stopped.
***

*It was connected to a 5.1V 850mA source.*

Each processor, sending 12 (one beacon mac address) bytes by HTTP_REQUEST,
plus HTTP header.

*** Reason :

HTTP/1.1 200 OK

ate: Sat, 07 Apr 2018 12:I (99163i4:34 )  MATTC_DEMORT:
@initialise_wifi has been detected.sk

Backtrace: 0x40090050:0x3ffd4710 0x40090227:0x3ffd4730
0x40090240:0x3ffd4750 0x4008d93b:0x3ffd4770 0x4008f184:0x3ffd4790
0x4008f13a:0x2820456d
0x40090050: invoke_abort at
/home/raimundo/esp/esp-idf/components/esp32/./panic.c:648

0x40090227: abort at
/home/raimundo/esp/esp-idf/components/esp32/./panic.c:648

0x40090240: *vApplicationStackOverflowHook* at
/home/raimundo/esp/esp-idf/components/esp32/./panic.c:648

0x4008d93b: *vTaskSwitchContext* at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:3529

0x4008f184: _frxt_dispatch at
/home/raimundo/esp/esp-idf/components/freertos/./portasm.S:406

0x4008f13a: _frxt_int_exit at
/home/raimundo/esp/esp-idf/components/freertos/./portasm.S:206

*Rebooting*...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5664
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13912
entry 0x40078fd8
I (30) boot: ESP-IDF v3.1-dev-601-gda27816-dirty 2nd stage bootloader
I (30) boot: compile time 00:29:53
I (32) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00180000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x35dec
[... 138 lines cut ...]

0x400d5438: __assert_f*unc at /Users/ivan/e/*
newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:58

0x4008dc97: vPortCPUAcquireMutexIntsDisabledInternal at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:3529
 (inlined by) vPortCPUAcquireMutexIntsDisabled at
/home/raimundo/esp/esp-idf/components/freertos/./portmux_impl.h:98
 (inlined by) vTaskEnterCritical at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:4216

0x4008cfe1: xQueueGenericSendFromISR at
/home/raimundo/esp/esp-idf/components/freertos/./queue.c:2037

0x40088e8a: pp_post at ??:?

0x40088639: lmacPostTxComplete at ??:?

0x4008b639: wDev_ProcessFiq at ??:?

0x40082561: _xt_lowint1 at
/home/raimundo/esp/esp-idf/components/freertos/./xtensa_vectors.S:1105

Rebooting...
ets Jun  8 2016 00:22:57

perll, 2018-04-07:
New issue:

GATT_CLIENTE with HTTP_REQUEST. 12 bytes bytes of request.
I think Heap, memory leak... are come back

I (241143) WIFI...: ... socket send success
I (241143) WIFI...: ... set socket receiving timeout success
HTTP/1.1 200 OK
Date: Sat, 07 Apr 2018 21:59:14 GMT
Server: Apache
abort() was called at PC 0x40082d4b on core 0
0x40082d4b: lock_init_generic at
/home/raimundo/esp/esp-idf/components/newlib/./locks.c:81

Backtrace: 0x40090050:0x3ffd56b0 0x40090227:0x3ffd56d0
0x40082d4b:0x3ffd56f0 0x40082d6c:0x3ffd5710 0x40082eb1:0x3ffd5740
0x400d971f:0x3ffd5760 0x400d54a9:0x3ffd5a20 0x400d5438:0x3ffd5a70
0x4008dc97:0x3ffd5aa0 0x4008f96a:0x3ffd5ac0 0x400827c4:0x3ffd5ae0
0x400827f5:0x3ffd5b00 0x40082ca1:0x3ffd5b20 0x4000beaf:0x3ffd5b40
0x4008ccfe:0x3ffd5b60 0x4008cf30:0x3ffd5b80 0x40082d46:0x3ffd5ba0
0x40082d6c:0x3ffd5bc0 0x40082eb1:0x3ffd5bf0 0x400d971f:0x3ffd5c10
0x400d54a9:0x3ffd5ed0 0x400d5438:0x3ffd5f20 0x4008dc97:0x3ffd5f50
0x4008f96a:0x3ffd5f70 0x400827c4:0x3ffd5f90 0x400827f5:0x3ffd5fb0
0x40082ca1:0x3ffd5fd0 0x4000beaf:0x3ffd5ff0 0x4008ccfe:0x3ffd6010
0x4008cf30:0x3ffd6030 0x40082d46:0x3ffd6050 0x40082d6c:0x3ffd6070
0x40082eb1:0x3ffd60a0 0x400d971f:0x3ffd60c0 0x400d54a9:0x3ffd6380
0x400d5438:0x3ffd63d0 0x4008dc97:0x3ffd6400 0x4008f96a:0x3ffd6420
0x400827c4:0x3ffd6440 0x400827f5:0x3ffd6460 0x40082ca1:0x3ffd6480
0x4000beaf:0x3ffd64a0 0x4008ccfe:0x3ffd64c0 0x4008cf30:0x3ffd64e0
0x40082d46:0x3ffd6500 0x40082d6c:0x3ffd6520 0x40082eb1:0x3ffd6550
0x400d971f:0x3ffd6570 0x400d54a9:0x3ffd6830 0x400d5438:0x3ffd6880
0x4008dc97:0x3ffd68b0
0x40090050: invoke_abort at
/home/raimundo/esp/esp-idf/components/esp32/./panic.c:648

0x40090227: abort at
/home/raimundo/esp/esp-idf/components/esp32/./panic.c:648

0x40082d4b: lock_init_generic at
/home/raimundo/esp/esp-idf/components/newlib/./locks.c:81

0x40082d6c: lock_acquire_generic at
/home/raimundo/esp/esp-idf/components/newlib/./locks.c:132

0x40082eb1: _lock_acquire_recursive at
/home/raimundo/esp/esp-idf/components/newlib/./locks.c:169

0x400d971f: _vfiprintf_r at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:859
(discriminator 9)

0x400d54a9: fiprintf at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:49

0x400d5438: __assert_func at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:58

0x4008dc97: vPortCPUAcquireMutexIntsDisabledInternal at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:3529
 (inlined by) vPortCPUAcquireMutexIntsDisabled at
/home/raimundo/esp/esp-idf/components/freertos/./portmux_impl.h:98
 (inlined by) vTaskEnterCritical at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:4216

0x4008f96a: multi_heap_internal_lock at
/home/raimundo/esp/esp-idf/components/heap/./multi_heap.c:352
 (inlined by) multi_heap_malloc_impl at
/home/raimundo/esp/esp-idf/components/heap/./multi_heap.c:388

0x400827c4: heap_caps_malloc at
[... 109 lines cut ...]
0x40082d6c: lock_acquire_generic at
/home/raimundo/esp/esp-idf/components/newlib/./locks.c:132

0x40082eb1: _lock_acquire_recursive at
/home/raimundo/esp/esp-idf/components/newlib/./locks.c:169

0x400d971f: _vfiprintf_r at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:859
(discriminator 9)

0x400d54a9: fiprintf at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:49

0x400d5438: __assert_func at
/Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:58

0x4008dc97: vPortCPUAcquireMutexIntsDisabledInternal at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:3529
 (inlined by) vPortCPUAcquireMutexIntsDisabled at
/home/raimundo/esp/esp-idf/components/freertos/./portmux_impl.h:98
 (inlined by) vTaskEnterCritical at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:4216

Rebooting...
ets Jun  8 2016 00:22:57

TimXia, 2018-04-08:
@perll `errno = 128` means TCP connectio is closed. You can reconnect it again. Could you provide the elf file to us to debug the crash issue? BTW, did you use PSRAM?

perll, 2018-04-09:
New issue:

GATT_CLIENTE with HTTP_REQUEST. 12 bytes bytes of request.
I think Heap, memory leak... are come back

I (241143) WIFI...: ... socket send success
I (241143) WIFI...: ... set socket receiving timeout success
HTTP/1.1 200 OK
Date: Sat, 07 Apr 2018 21:59:14 GMT
Server: Apache
abort() was called at PC 0x40082d4b on core 0
0x40082d4b: lock_init_generic at /home/raimundo/esp/esp-idf/
components/newlib/./locks.c:81

Backtrace: 0x40090050:0x3ffd56b0 0x40090227:0x3ffd56d0
0x40082d4b:0x3ffd56f0 0x40082d6c:0x3ffd5710 0x40082eb1:0x3ffd5740
0x400d971f:0x3ffd5760 0x400d54a9:0x3ffd5a20 0x400d5438:0x3ffd5a70
0x4008dc97:0x3ffd5aa0 0x4008f96a:0x3ffd5ac0 0x400827c4:0x3ffd5ae0
0x400827f5:0x3ffd5b00 0x40082ca1:0x3ffd5b20 0x4000beaf:0x3ffd5b40
0x4008ccfe:0x3ffd5b60 0x4008cf30:0x3ffd5b80 0x40082d46:0x3ffd5ba0
0x40082d6c:0x3ffd5bc0 0x40082eb1:0x3ffd5bf0 0x400d971f:0x3ffd5c10
0x400d54a9:0x3ffd5ed0 0x400d5438:0x3ffd5f20 0x4008dc97:0x3ffd5f50
0x4008f96a:0x3ffd5f70 0x400827c4:0x3ffd5f90 0x400827f5:0x3ffd5fb0
0x40082ca1:0x3ffd5fd0 0x4000beaf:0x3ffd5ff0 0x4008ccfe:0x3ffd6010
0x4008cf30:0x3ffd6030 0x40082d46:0x3ffd6050 0x40082d6c:0x3ffd6070
0x40082eb1:0x3ffd60a0 0x400d971f:0x3ffd60c0 0x400d54a9:0x3ffd6380
0x400d5438:0x3ffd63d0 0x4008dc97:0x3ffd6400 0x4008f96a:0x3ffd6420
0x400827c4:0x3ffd6440 0x400827f5:0x3ffd6460 0x40082ca1:0x3ffd6480
0x4000beaf:0x3ffd64a0 0x4008ccfe:0x3ffd64c0 0x4008cf30:0x3ffd64e0
0x40082d46:0x3ffd6500 0x40082d6c:0x3ffd6520 0x40082eb1:0x3ffd6550
0x400d971f:0x3ffd6570 0x400d54a9:0x3ffd6830 0x400d5438:0x3ffd6880
0x4008dc97:0x3ffd68b0
0x40090050: invoke_abort at /home/raimundo/esp/esp-idf/
components/esp32/./panic.c:648

0x40090227: abort at /home/raimundo/esp/esp-idf/
components/esp32/./panic.c:648

0x40082d4b: lock_init_generic at /home/raimundo/esp/esp-idf/
components/newlib/./locks.c:81

0x40082d6c: lock_acquire_generic at /home/raimundo/esp/esp-idf/
components/newlib/./locks.c:132

0x40082eb1: _lock_acquire_recursive at /home/raimundo/esp/esp-idf/
components/newlib/./locks.c:169

0x400d971f: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.
2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/
stdio/../../../.././newlib/libc/stdio/vfprintf.c:859 (discriminator 9)

0x400d54a9: fiprintf at /Users/ivan/e/newlib_xtensa-2.
2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/
stdio/../../../.././newlib/libc/stdio/fiprintf.c:49

0x400d5438: __assert_func at /Users/ivan/e/newlib_xtensa-2.
2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/
stdlib/../../../.././newlib/libc/stdlib/assert.c:58

0x4008dc97: vPortCPUAcquireMutexIntsDisabledInternal at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:3529
 (inlined by) vPortCPUAcquireMutexIntsDisabled at
/home/raimundo/esp/esp-idf/components/freertos/./portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/raimundo/esp/esp-idf/
components/freertos/./tasks.c:4216

0x4008f96a: multi_heap_internal_lock at /home/raimundo/esp/esp-idf/
components/heap/./multi_heap.c:352
 (inlined by) multi_heap_malloc_impl at /home/raimundo/esp/esp-idf/
components/heap/./multi_heap.c:388
[... 117 lines cut ...]

0x40082eb1: _lock_acquire_recursive at /home/raimundo/esp/esp-idf/
components/newlib/./locks.c:169

0x400d971f: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.
2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/
stdio/../../../.././newlib/libc/stdio/vfprintf.c:859 (discriminator 9)

0x400d54a9: fiprintf at /Users/ivan/e/newlib_xtensa-2.
2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/
stdio/../../../.././newlib/libc/stdio/fiprintf.c:49

0x400d5438: __assert_func at /Users/ivan/e/newlib_xtensa-2.
2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/
stdlib/../../../.././newlib/libc/stdlib/assert.c:58

0x4008dc97: vPortCPUAcquireMutexIntsDisabledInternal at
/home/raimundo/esp/esp-idf/components/freertos/./tasks.c:3529
 (inlined by) vPortCPUAcquireMutexIntsDisabled at
/home/raimundo/esp/esp-idf/components/freertos/./portmux_impl.h:98
 (inlined by) vTaskEnterCritical at /home/raimundo/esp/esp-idf/
components/freertos/./tasks.c:4216

Rebooting...
ets Jun  8 2016 00:22:57

perll, 2018-04-09:
Sure!

I got some problems attaching the code. So, ELF below. Yes. I'm using
PSRAM. Please, Special atemption tofor ESP_GAP_BLE_SCAN_RESULT_EVT.
There I'm getting the BLE SCAN RESULT and setting array that will be
the REQUEST.

Regards!

/*
   This example code is in the Public Domain (or CC0 licensed, at your
option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

/****************************************************************************
*
* This file is for gatt client. It can scan ble device, connect one device.
* Run the gatt_server demo, the client demo will automatically connect to
the gatt_server demo.
* Client demo will enable gatt_server's notify after connection. Then the
two devices will exchange
* data.
*
****************************************************************************/
#include <inttypes.h>
#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"
#include "controller.h"

#include "esp_bt.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"
#include "esp_gatt_common_api.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include "lwip/netdb.h"
#include "lwip/dns.h"

#define GATTC_TAG "GATTC_DEMO"
#define REMOTE_SERVICE_UUID        0x00FF
#define REMOTE_NOTIFY_CHAR_UUID    0xFF01
#define PROFILE_NUM      1
#define PROFILE_A_APP_ID 0
#define INVALID_HANDLE   0
#define DURATION 10

/* Constants that aren't configurable in menuconfig */
#define WEB_SERVER "ipolvo.com.br"
#define WEB_PORT 80
#define WEB_URL "http://www.ipolvo.com.br/ipolvoTeste/agua/index2.php?"
[... 758 lines cut ...]
    if(ret){
        ESP_LOGE(GATTC_TAG, "%s gattc register failed, error code = %x\n",
__func__, ret);
        return;
    }

    ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTC_TAG, "%s gattc app register failed, error code =
%x\n", __func__, ret);
    }
    esp_err_t local_mtu_ret = esp_ble_gatt_set_local_mtu(517);
    if (local_mtu_ret){
        ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x",
local_mtu_ret);
    }

    initialise_wifi();
    xTaskCreate(&http_get_task, "http_get_task", (1024 * 64), NULL, 7,
NULL);
    //http_get_task(NULL);

    printf("xPortGetFreeHeapSize :%d\n", xPortGetFreeHeapSize() );

}

TimXia, 2018-04-10:
@perll Can you show me your sdkconfig file?

@liuzfesp Is the last crash log similar to the bug that you are fixing? (allocate queue in psram)

perll, 2018-04-10:
Sure! Attached.

Yes. I've getting same errors again. Today, I got "IP LOST" (I (2140007)
event: station ip lost).

Just remembering, There are 4 esp wroom32 working at same time in the same
net wifi.

In fact, after the last update, I was able to increase the working time of
the ESP Wroom32. The amount of reboots has decreased significantly. But,
there is still a reboot at least 10 times in an hour and today, 2 of 4
espwroom32 have stopped and the reason was LOST IP.

Regards!

perll, 2018-04-10:
I (1254743) WIFI...: ... done reading from socket. Last read return=0
errno=128

I (1254753) WIFI...: 10...
I (1255763) WIFI...: 9...
I (1256763) WIFI...: 8...
I (1257763) WIFI...: 7...
I (1258763) WIFI...: 6...
I (1259763) WIFI...: 5...
I (1260763) WIFI...: 4...
I (1261763) WIFI...: 3...
I (1262763) WIFI...: 2...
I (1263763) WIFI...: 1...
I (1264763) WIFI...: 0...
I (1265763) WIFI...: Starting again!
I (1265763) WIFI...: Connected to AP
I (1265763) WIFI...: DNS lookup succeeded. IP=50.30.38.132
I (1265763) WIFI...: ... allocated socket
I (1271763) wifi: bcn_timout,ap_probe_send_start
I (1274263) wifi: ap_probe_send over, resett wifi status to disassoc
I (1274263) wifi: state: run -> init (1)
I (1274263) wifi: pm stop, total sleep time: 0/68137917

I (1274263) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
*I (1303373) event: station ip lost*

perll, 2018-04-11:
Sir,

some news about the fix for my issues?

Regards!

perll, 2018-04-12:
> I updated with last master git  version but, unfortunally now, the wifi
> connection stopped after 10'. And there isn't log message. Is not possible
> known what is wrong.
>
> On Wed, Apr 11, 2018 at 6:41 AM, Raimundo Miranda Neto <

Sir,

could you help me:

E (109) boot: ota data partition invalid, falling back to factory
I (116) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020
size=0x36bd0 (224208) map
I (203) esp_image: segment 1: paddr=0x00046bf8 vaddr=0x3ffc0000
size=0x03ae8 ( 15080) load
I (209) esp_image: segment 2: paddr=0x0004a6e8 vaddr=0x40080000
size=0x00400 (  1024) load
0x40080000: _iram_start at
/home/raimundo/esp/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (210) esp_image: segment 3: paddr=0x0004aaf0 vaddr=0x40080400
size=0x05520 ( 21792) load
I (228) esp_image: segment 4: paddr=0x00050018 vaddr=0x400d0018
size=0xb2fd0 (733136) map
0x400d0018: _stext at ??:?

I (485) esp_image: segment 5: paddr=0x00102ff0 vaddr=0x40085920
size=0x0e314 ( 58132) load
0x40085920: r_lld_evt_restart at ??:?

I (509) esp_image: segment 6: paddr=0x0011130c vaddr=0x400c0000
size=0x00000 (     0) load
E (510) esp_image: Image length 1053488 doesn't fit in partition length
1048576
E (515) boot: Factory app partition is not bootable
E (521) esp_image: image at 0x110000 has invalid magic byte
E (527) boot: OTA app partition slot 0 is not bootable
E (533) esp_image: image at 0x210000 has invalid magic byte
W (539) esp_image: image at 0x210000 has invalid SPI mode 255
W (545) esp_image: image at 0x210000 has invalid SPI size 15
E (552) boot: OTA app partition slot 1 is not bootable
E (557) boot: No bootable app partitions in the partition table
user code done

perll, 2018-04-14:
Some news about ip lost issue?

Regards

TimXia, 2018-05-22:
@perll We have improved the coexistence of WiFi and BT. Please use the latest IDF to test. BTW, could you please provide your code by an attached file?

perll, 2018-05-22:
Tku, sir!
ASAP I will proceed the tests and return you