espressif/esp-idf #1775: Losing WiFi connection on certain networks

korstiaanS, 2018-03-26:
Hi,

What can be the reason that always after reboot of the ESP32 pinging to the module stops after 10-60sec. I have used the "simple_wifi" example with nothing in it. I get connected, get an ip and if I ping from my PC to the ESP32 it always stops after 10-60sec. on certain networks? This is the case on Meraki and Cisco with a WLC4400 controller.
I also have the problem with an ESP32 and Arduino framework and PlatformIO.
On my wifi AP at home it keeps on working. But the ESP8266 works on ALL networks, home and work.
What is different and what causes this problem? Is it like a wifi preamble parameter or something else that causes the ESP32 to stop working?
Also tried v2.1.1 and v3.0rc1.
Please help.
Korstiaan

liuzfesp, 2018-04-03:
@korstiaanS  do you means the WiFi is disconnected or ping is stop, after 10~60 seconds?  I just can't reproduce this issue, could you provide more detailed log?

korstiaanS, 2018-04-03:
After further investigation we found the reason:

By further investigations and packet captures we have encountered the root cause of the connectivity issue.

In normal condition we see by packet capture that the ESP32 module is using the correct IP and correct MAC address of it’s default gateway.
So IP communication with devices outside it’s subnet is correctly send to the default gateway mac (example: 00:08:e3:aa:aa:aa)

Every 60 seconds the ESP32 module receives an IGMP general membership query from the IGMP querier in it’s subnet. (in our case this IGMP querier is our Cisco core switch).
Upon receival of the first IGMP query the ESP module changes the default gateway mac address (example: 00:08:e3:aa:aa:aa) with the source mac address of the igmp querier (example: 00:09:e3:bb:bb:bb)
As of now all IP packets towards it’s default gateway are send towards the incorrect mac address (00:09:e3:bb:bb:bb) and the ESP32 is not reachable anymore from outside it’s subnet.

In our specific setup we have disabled the IGMP querier functionality for the specific VLAN/IP subnet on the Cisco core switch to solve this encountered issue.

Why the ESP32 have problems and the **ESP8266 not** is for me not clear.

negativekelvin, 2018-04-03:
http://lwip.100.n7.nabble.com/ETHARP-TRUST-IP-MAC-and-IGMP-querier-td16082.html

liuzfesp, 2018-04-10:
HI @korstiaanS just as @negativekelvin pointed out, you can configure ETHARP_TRUST_IP_MAC to 0 by  LWIP menuconfig. Maybe we can do some optimization when ETHARP_TRUST_IP_MAC is enabled: 
1. Only trust unicast packets
2. If the MAC address for a specific IP is different than the existing mac in the table, print some warning message
We will discuss how to improve it...

korstiaanS, 2018-04-10:
Hi,
Ok.
The problem is also that I use ESP32 on Arduino and there it is not so simple to change the default from 1 to 0.

hrishikesh-thakre, 2019-05-11:
I use ESP32 on Arduino, is there a way to set ETHARP_TRUST_IP_MAC to 0 and solve this problem temporarily? I am having the same issue currently.

zhangyanjiaoesp, 2019-06-17:
@hrishikesh-thakre The definition for the "ETHARP_TRUST_IP_MAC" is[ here](https://github.com/espressif/arduino-esp32/blob/7d7824701fe5e22f08555d3e1ce3180a922b2151/tools/sdk/include/lwip/lwipopts.h#L714) . You can set it in the menuconfig.

closed 2020-07-08 by Alvin1Zhang