espressif/esp-idf #1776: UART Reset

thiker5, 2018-03-26:
I did some testing with UART-1 , with the changes applied from commit  4052803e161ba06d1cae8d36bc66dde15b3fc8c7 . The changes fixes the UART-2 overflow reset hardware issue, however after some time of using UART-1 (> 512MB) , and having few other tasks running (good chance of overflow!) the UART-1 stops working at all , and soft restart doesn't fix it. I tried the same test again with adding Hard reset for rxfifo and txfifo and it works fine. The commit there fixes the silicon issue, but the digital team added the reset fifo for a reason, and I believe in some scenarios It has to be utilized.  
I recommend to add fifo reset after soft restart regardless of the silicon version, and allow the developers to do fifo reset on their own risk if they use UART-2.

negativekelvin, 2018-03-26:
Did you also apply https://github.com/espressif/esp-idf/commit/492db0591d0c5643cb863b76600ab0bf5a204b64

thiker5, 2018-03-26:
I will be running the test over the next couple of days, and will get back to you with the results.

thiker5, 2018-03-27:
after applying https://github.com/espressif/esp-idf/commit/492db0591d0c5643cb863b76600ab0bf5a204b64 , it got fixed. Thanks.

closed 2018-03-29 by FayeY