espressif/esp-idf #1779: wifi power save: Unnecessary null frames sent

mikaelkanstrup, 2018-03-27:
When wifi power save mode is activated (WIFI_PS_MODEM) there are unnecessary NULL data frames sent by esp32.

Background:
When in legacy power save mode STA inform its power state by sending NULL data frames to AP with PWR MGT flag either set or cleared. Depending on STA power state AP will either:

- STA awake: Send data right away to STA
- STA sleeping: Buffer data. At next (DTIM) Beacon indicate whether there's data buffered or not for STA

A sleeping STA will have to wake up and listen to beacons at certain intervals to check whether there's downlink data buffered at AP side.

Pattern seen with ESP32 (with WIFI_PS_MODEM) activated:
STA sends NULL frame (go to sleep)
AP ACK
STA sends NULL frame (wake)
AP ACK
AP sends beacon (TIM indicate no traffic for STA)
STA sends NULL frame (go to sleep)
AP ACK
STA sends NULL frame (wake)
AP ACK
AP sends beacon (TIM indicate no traffic for STA)
STA sends NULL frame (go to sleep)
AP ACK
..

This pattern is repeated over an over when there's no traffic (up or downlink)

For power save reasons sending NULL frames should be avoided when not needed. A suggested change to implement for ESP-IDF is to avoid sending NULL frames indicating power state changes that would anyway not result in receiving downlink data. The above scenario would then look like this:

Optimized WIFI_PS_MODEM mode (STA enters power save state):
STA sends NULL frame (go to sleep)
AP ACK
STA silently wakes up to listen to beacon (no NULL frame sent)
AP sends beacon (TIM indicates no traffic for STA)
STA silently goes to sleep (no NULL frame sent)
STA silently wakes up to listen to beacon (no NULL frame sent)
AP sends beacon (TIM indicates no traffic for STA)
STA silently goes to sleep (no NULL frame sent)

When there's downlink traffic:
STA silently wakes up (no NULL frame sent)
AP sends beacon (TIM indicates traffic for STA)
STA sends NULL frame (awake)
AP ACK
AP sends data frame(s) to STA
STA ACK(s)
AP sends beacon (TIM now indicates no more traffic for STA)
STA sends NULL frame (go to sleep)
...

This suggested change should have a significant effect on power consumption in power save mode.

TimXia, 2018-04-01:
@mikaelkanstrup Thanks for your suggestion. But we did implement power save as you suggested. We did not send NULL frame (wake) before listening to beacon. How did get pattern like the following?

Pattern seen with ESP32 (with WIFI_PS_MODEM) activated:
STA sends NULL frame (go to sleep)
AP ACK
STA sends NULL frame (wake)
AP ACK
AP sends beacon (TIM indicate no traffic for STA)
STA sends NULL frame (go to sleep)
AP ACK
STA sends NULL frame (wake)
AP ACK
AP sends beacon (TIM indicate no traffic for STA)
STA sends NULL frame (go to sleep)
AP ACK
..

liuzfesp, 2018-04-03:
@mikaelkanstrup, if there is no data at all, the null frame is sent for keep alive reason. If no null frame, most of the AP will death the STA.

mikaelkanstrup, 2018-04-03:
@TimXia I was just connected to a simple wifi router, STA configured with WIFI_PS_MODEM. No uplink of downlink data. Then looked at traffic with a wifi sniffer. I used latest ESP-IDF from master at that time (9ea3856 or c97b875 I think).

Today I noticed the refactored wifi modem sleep API merged to master branch and ran some new tests. That change actually seems to solve this issue. Tests done so far seems good. Power consumption much better and power pattern looks good with clean spikes indicating "silent" wake-ups at configured listen_interval. When downlink traffic is indicated in beacon STA signal wake-up with NULL frame and receive data properly. With no downlink traffic there seems to be a few NULL frames sent each 20s. Maybe for keep alive reasons which I guess is good to avoid what @liuzfesp mention.

Closing issue

closed 2018-04-03 by mikaelkanstrup