espressif/esp-idf #1780: ADC2 not working even with WiFi radio turned off.

S-March, 2018-03-27:
I had made a PCB with an ESP32-PICO-D4 before finding out that ADC2 can't be used simultaneously with WiFi. But in other comments and posts I've seen people mentioning that as long as they had the radio off they could use the ADC2 pins and that they could just alternate having the radio on vs taking measurements. 

One of the biggest issues is that 8/10 touch lines are connected to ADC2 making them unusable. I don't mind having WiFi off for the majority of the time since I rarely use it, but I do still need it for OTA. 

Any thoughts on a work around until the release of the IDF 3.1 comes along with a supposed fix? Also, any thoughts on when the IDF 3.1 will be released?

negativekelvin, 2018-03-27:
I think touch sensors have their own fsm and don't need adc2

S-March, 2018-03-27:
Hmm... I will double check. Other than that, I have a battery monitoring pin tied to ADC2, would be very nice to be able to read that each wake cycle...

ginkgm, 2018-06-14:
Hi @S-March ,
There are unit test for the ADC2, which turns on Wi-Fi then close it before the ADC2 reading. Like
```
esp_wifi_init();
esp_wifi_start();
esp_wifi_stop();
adc2_get_raw();
```
And I've just run it on latest master and it passed.

Could you provide which version (commit) is your IDF, and the function calling sequence before you called ADC2 (in brief)?

BTW, Touch sensor does not using ADC2 as @negativekelvin said.

closed 2018-07-12 by FayeY