espressif/esp-idf #1782: If esp_wifi_80211_tx is used. Can esp32 enter modem sleep?

jixunx, 2018-03-28:
Hi !
 I want to use esp_wifi_80211_tx, and I hope that esp32 can enter in modem sleep.I wrote a sample project.
Base on esp-idf commit c97b8756f740297e23f3011603724005db22c557.The attatchment is my sample code ,sdkconfig file and experimental picture.

No matter whether I use "esp_wifi_set_ps (WIFI_PS_MODEM") or not, the minimum current of our board has not changed.

Please give me some suggestion ,thanks!
[esp_80211_tx.zip](https://github.com/espressif/esp-idf/files/1854449/esp_80211_tx.zip)

Best regards!

TimXia, 2018-04-01:
@jixunx Modem sleep only works when station connects with AP. So, please call esp_wifi_connect() after esp_wifi_start() to connect to AP.

jixunx, 2018-04-02:
Thanks for your reply,I make esp32 as station and connected with AP.And I still use esp_wifi_80211_tx to send data. The code is still based on commit c97b8756f740297e23f3011603724005db22c557. The current of esp32 still is >100ma. Why? And esp_wifi_80211_tx always report"wifi:out of memory".Why? The attachment is my code after added "wifi connection".
Please give me some suggestion.Thanks!
Best regards!

TimXia, 2018-04-02:
@jixunx esp_wifi_80211_tx() was called too frequently. WiFi is short of memory to transmit these data in time. What's more, WiFi has no chance to enter modem sleep if data is transmitted frequently. If the interval between transmitting data is more than 1.3s, WiFi will enter modem sleep.

jixunx, 2018-04-02:
I called vTaskDelay(5000) in my code.It will modem sleep 5s. But it didn't sleep.

TimXia, 2018-04-02:
You should call vTaskDelay(5000 / portTICK_RATE_MS) to delay 5s.

jixunx, 2018-04-02:
Thanks for your reply.
No difference.This is not the point.Even though  vTaskDelay(5000 / portTICK_RATE_MS) is called,esp32 can not enter in modem sleep,current >100ma. I make connection( station to AP),but once using esp_wifi_80211_tx,esp32 can not enter in modem sleep.
Best regards!

jixunx, 2018-04-03:
> If the interval between transmitting data is more than 1.3s, 
1.3s？It is too long.Is it 1.3ms?  I did the test,It does not need 1.3s interval.
Thanks!
Best regards!

TimXia, 2018-04-03:
@jixunx So, did it enter modem sleep if interval is 1.3s? 1.3s is a small feature in modem sleep in order to tx/rx data more successfully. In IDFv3.1, we remove this feature, because we have another better method.

TimXia, 2018-07-10:
@jixunx Does this issue still exist? Please try IDF v3.1. It may have solved your issue.

jixunx, 2018-07-11:
Yes,it is still exist.Any sample code?

jixunx, 2018-07-11:
Idf v3.1 is not officially released on espressif official website, when will idf v3.1 be officially released? I don't really want to use the idf version that is constantly updated on github, it will not be stable enough.It is bad for my application.
If esp_wifi_80211_tx is used, esp32  can enter modem sleep .If  you have verified.Please provide sample code.

I will really appreciate.

Alvin1Zhang, 2019-06-14:
@jixunx Sorry for getting back to you so late... Would you kindly check and share whether your issue still exists? Thanks.

Alvin1Zhang, 2019-09-29:
@jixunx Thanks for reporting, and sorry for getting back to you so late... Would you kindly check and share whether your issue still exists? Thanks.

Alvin1Zhang, 2019-11-01:
@jixunx Will close the ticket for now, feel free to reopen if the issue still happens. Thanks.

closed 2019-11-01 by Alvin1Zhang