espressif/esp-idf #1783: GPIO10

SirRibeiro, 2018-03-28:
Hello,
In the previous versions of esp-idf I was able to use the GPIO10 (FLASH D3) for some things, for example I was using it to read a keypad matrix.
I was only doing that because i dont have any more free pins left in my project.

Now i can't use that pin anymore so i fall back to the 2.1 version.

My question is, what can i do to use once again that pin? (i really can't use any other)
And what problems can i have for using that pin to read the keypad?

Thanks

costaud, 2018-04-09:
The GPIO10 has been used as QUAD mode of SPI flash by default.

1. If you are using an official module like WROOM-32, since GPIO10 is connected with the FLASH /WP pin, you should not use it in this case.

2. If you want to use GPIO10, we should meet the following conditions:
    a) Use DIO mode for SPI flash.
    b) Disconnect GPIO10 with SPI flash /WP pin.
    c) Add pull-up resistor on SPI flash /WP pin.

SirRibeiro, 2018-04-09:
Thank you for your response.
Do you know if there is a way of simply unbid the GPIO10 from de WP function by software?

panfeng-espressif, 2018-04-16:
GPIO10 is connected with FLASH/WP pin in hardware. Unless you disconnect it with the pin, there will be no suitable way in software only.

closed 2018-06-22 by FayeY

HFRibeiro, 2018-11-06:
Hello,
So once again i had this same problem in another project, as i was able to use it as input on the version 2.1, I was not convinced with your responses, so after some reading i came up with a simple solution.
If anyone stumble on this as well just use  gpio_reset_pin((gpio_num_t)10);
This will make you once again able to use the GPIO10.
Thanks anyway.

xiongyumail, 2018-11-07:
```
* @note This function also configures the IOMUX for this pin to the GPIO
 *       function, and disconnects any other peripheral output configured via GPIO
 *       Matrix.
```
This function can configure the corresponding io as gpio function.
You may also use the following method:
```
 PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[10], PIN_FUNC_GPIO);
```