espressif/esp-idf #1784: SYSTEM_EVENT_AP_STACONNECTED received even if handshake fails (idf 3.0rc)

lollizze, 2018-03-28:
I have set up one esp32 as a wpa2 Access point, then i try and connect an other esp32 station but with a faulty password on the AP i get a SYSTEM_EVENT_AP_STACONNECTED event but on the station i get a disconnect. 

The event is misleading and should only only be received if a station is connected but it isn't if the handshake isn't finished

negativekelvin, 2018-03-28:
It is useful to know about connection attempts, even failed ones.  So either it should also generate an immediate SYSTEM_EVENT_AP_STADISCONNECTED or there should be some other event for a failed connection.

lollizze, 2018-03-29:
It might be useful to know about failed connection attempts, but it would be more useful to get it in an other event that makes it clear it is just an attempt, it shouldn't be in STACONNECTED because it isn't a station that have connected

TimXia, 2018-04-01:
@lollizze It is connected though 4-way-handshake will fail. Another event SYSTEM_EVENT_AP_STADISCONNECTED will be generated when 4-way-handshake time out. When 4-way-handshake suceeds, IP address will be assigned to station. We will add another event in IDFv3.1 when softap assigns IP address to station.

negativekelvin, 2018-04-01:
@TimXia what if dhcp server is disabled and client is using static IP? Maybe there could be SYSTEM_EVENT_AP_STACONNECT before handshake and SYSTEM_EVENT_AP_STACONNECTED after handshake.

TimXia, 2018-04-02:
@negativekelvin Good suggestion. We will have a discussion about it. Thank you.

lollizze, 2018-04-03:
@TimXia we want to know if the station is added and the 4 way handshake succeeds, if it is possible to do as @negativekelvin suggested that would be good then it is possible know if the link is valid without relying on IP traffic and it would still be possible to catch connect attempts

TimXia, 2018-05-22:
@lollizze We can add another event SYSTEM_EVENT_AP_STAAUTHORIZED after handshanke. How do you think?

lollizze, 2018-05-22:
@TimXia yes i think that is a good idea! then it is clear that when that event arrives the link is completely up and i can wait for that instead

closed 2019-05-02 by lollizze