espressif/esp-idf #1785: Bluetooth Controller doesn't respond to LL_CONNECTION_PARAM_REQ when pairing

jakkra, 2018-03-28:
Hi

We are having problems pairing/bonding using Android phones. The pairing procedure times out eventually. So far we have seen the same issue on LG G6, OnePlus 2, Samsung Galaxy S8.

We are using [the BlueKitchen stack](https://github.com/bluekitchen/btstack/tree/develop) instead of Bluedroid.

This example [sm_pairing_peripherale](https://github.com/bluekitchen/btstack/blob/develop/example/sm_pairing_peripheral.c) with LE legacy pairing Just Works can be used to reproduce. 

I have attached the HCI log from the ESP32 running the `sm_pairing_peripheral` example, the HCI log from the LG G6. I also used a BLE sniffer while trying to pair, that log is also attached.

From the sniffer log, we can se that the phone sends a `LL_CONNECTION_PARAM_REQ` which is never answered. I suspect this is the issue, since either the Espressif Bluetooth Controller should respond, or an `Connection Parameters Request Event ` event should be emitted to the stack. In the esp32 HCI log, we can see that this event is never sent by the Bluetooth Controller.

To verify the potential cause I masked the `LE Remote Connection Parameter Request Event` in the `LE_Set_Event_Mask`, which forces the Bluetooth Controller to decide the Connection Parameter on its own without notifying the stack. Doing this, the Bluetooth Controller responds to the `LL_CONNECTION_PARAM_REQ` and the bond/pairing is successful.

We also tried a similar scenario using the Bluedroid stack and the gatt_security_server example and the pairing worked as it should. The `LL_CONNECTION_PARAM_REQ` was answered and pairing successful.

So our guess is that there is a bug in the Bluetooth Controller where for some configuration the `LL_CONNECTION_PARAM_REQ` event is never passed to the Bluetooth stack. 

[Instructions to run the example is found here](https://github.com/bluekitchen/btstack/tree/develop/port/esp32)

If you need further information or have questions, please ask. 

[Here are the attached HCI and sniffer logs](https://drive.google.com/file/d/10HZLadaewxXhkjbGU0gRHhcmzH6qDWB7/view?usp=sharing)

Thank you

chegewara, 2018-03-28:
Im not sure if i understand, but i have oposite case with gatt_secure_server example. In my case esp32 is sending connection parameters update request twice. Its no big deal with samsung S5 or S9+, because they are fast enough to respond both requests, but with my old samsung S4 im getting this error:
`E (31124) BT: l2cble_start_conn_update, the last connection update command still pending.`

mringwal, 2018-03-29:
@chegewara The issue here is about the Bluetooth Controller and the behaviour of the Link Layer. Your error message look like an L2CAP request. In general, you need to wait for one request to be completed before starting the next one - the error message is correct and the problem is in "your" code (whoever wrote it :)

chegewara, 2018-03-29:
@mringwal Thanks

btw its espressif exmple i am using (whoever wrote it)

mringwal, 2018-03-29:
@chegewara I had a quick look at the gatt_security_server and there's no mention of a connection parameter update, so it's in one of the lower layers of Bluedroid. I'm not affilated with Espressif, but if you need this fixed, I guess you could file another issue with your error message (after checking it doesn't exist already).

chegewara, 2018-03-29:
Sorry for offtopic here, @mringwal i know you are not affiliated, i posted at first because i tought it may be related to OP. I know there is no code that is trying to send connection parameter update requests in this example and thats why i thought its related. My ble knowledge is not as good as yours. You pointed me im wrong, thanks for that. 

This issue is not bugging me because it exists only with samsung S4; S5 and S9+ can handle both requests. There is no need to do more off topic post here. Thanks again.

Weijian-Espressif, 2018-04-02:
@jakkra you can test our gatt_server Demo in ESP-IDF with your phone, it works well.

jakkra, 2018-04-02:
@Weijian-Espressif 

Hi, yes I tired and that works, so likely there is some bug in the controller which doesn't happend with the Bludrioid stack, but it's still there. As you can see from the logs, there is a LL_CONNECTION_PARAM_REQ send from the Android phone, which never is answered by the Bluetooth controller, nor a `Connection Parameters Request Event` emitted. 

> We also tried a similar scenario using the Bluedroid stack and the gatt_security_server example and the pairing worked as it should. The LL_CONNECTION_PARAM_REQ was answered and pairing successful.

mringwal, 2018-04-02:
@Weijian-Espressif 

Following up. When the LE Remote Connection Parameter Request Event is disabled with the LE Set Event Mask Command by BTstack, the controller rejects the LL_CONNECTION_PARAM_REQ correctly. Looking at the sources, Bluedroid in the esp-idf does not mask the event.

So, it looks like Bluedroid can responds to the LE Remote Connection Parameter Request Event, but BTstack doesn't get it. The question now is why is the event not emitted. Could you check in what conditions the LE Remote Connection Parameter Request Event is not emitted? Thanks!

Weijian-Espressif, 2018-04-03:
@jakkra when the phone sends a LL_CONNECTION_PARAM_REQ , the Espressif Bluetooth Controller will post a hci event to host, and host decide whether to accept or reject. you can refer "l2cble_process_rc_param_request_evt" in bludroid.

mringwal, 2018-04-03:
@Weijian-Espressif Thanks. We agree here, the Espressif Bluetooth Controller should post an HCI event to the host. However, in the logs provided above, this event is missing. That's why this issue was created. 

It didn't happen with my Android 6 device, but it happened with @jakkra's devices. I can try later to test different Bluetooth controllers in master role.

mringwal, 2018-04-04:
Hi all. You may close this issue, as it is invalid.

The LE Remote Connection Parameter Request Event is emitted as it should be but I've missed it (as Apple's PacketLogger does not decode it and only showed LE Meta Event - Unknown LE Sub Event 0X6.

closed 2018-04-04 by jakkra