espressif/esp-idf #1786: Error (-28)

kmakhil90, 2018-03-29:
Hi, 
I am trying to connect the esp32 to aws core. I have genearted and activated all the certificates in aws console which mentioned in the procedure. But still it is not connecting with mqtt and giving error -28.
Please suggest a workaround for this.

Akhil.

negativekelvin, 2018-03-29:
https://forums.aws.amazon.com/thread.jspa?messageID=753782&tstart=0

kmakhil90, 2018-03-29:
Hi,

Thanks.
I tried the change in timeout value as well as my policy contains all publish,receive,subscribe and connect permission as explained in above thread. 
then what else would be the problem here?

Akhil.

kmakhil90, 2018-04-02:
Hi:

Means i need to give the policy name start with iot*?

makefitter, 2018-04-02:
iot* means all permissions connect, publish, subscribe, delete. It's an option not a name.

kmakhil90, 2018-04-02:
i added this one too (iot*) also into my policy and connected in certificates but still error it is same

mahavirj, 2019-04-18:
@kmakhil90 Is this issue still exists? If not can this be closed?

closed 2019-04-24 by mahavirj