espressif/esp-idf #1789: Zero i2s dma buffer after write, so we have silence after buffer underflow

cranphin, 2018-03-29:
See also: https://github.com/earlephilhower/ESP8266Audio/issues/48

If a ets_memset(finish_desc->buf, 0, finish_desc->buf_size);
is added before
https://github.com/espressif/esp-idf/blob/da2781631455b2c87c01704503f70443799a5187/components/driver/i2s.c#L504
It would give us silence when a i2s buffer underflow happens, instead of the looping noise we get now :)

jeanthom, 2018-05-11:
I guess that some code has been changed in esp-idf (`ets_memset()` doesn't exist anymore, `finish_desc`'s `buf_size` element has been renamed to `size`), therefore the correct fix would be this :

```c
memset(finish_desc->buf, 0, finish_desc->size);
```

h3ndrik, 2018-07-12:
this doesn't result in a warning:
`memset((void*)(finish_desc->buf), 0, finish_desc->size);`

but is that correct? does it zero all the dma buffers? i still have problems with https://github.com/earlephilhower/ESP8266Audio/issues/48 even with this patch.
but that is probably another bug on my side.

pad52, 2018-07-13:
@h3ndrik what kind of problems do you have?
With this line I got no problems in pause/break of a song but I have frequent glitches - tuning dma settings make it better but there are anyway. (If I play a Sine Wave for example I perfectly can listen to them)

closed 2018-11-28 by igrr