espressif/esp-idf #1790: ESP32 stuck in programming mode

valentinAlkan, 2018-03-29:
I have a bunch of dev boards (ESP-WROOM-32 DEVKIT V1, from DOIT), and some of them will boot up normally and others wait for the EN/FLASH button to be pressed before they will boot normally. What's going on and how can I fix it?

valentinAlkan, 2018-03-30:
For reference, the exact same program runs fine on some boards but not others. There are no obvious physical defects and on the boards that are having issues, no program will load without pressing the EN/FLASH button, even the example programs.

valentinAlkan, 2018-03-30:
I have discovered that quickly power cycling them (<1s) causes them to boot into the normal program mode. Does anyone have any insight into what could be causing this? I have checked the strapping pins, and they have no effect (ie I force GPIO0 and GPIO2 to 3.3V and it still wants to startup in flash mode).

Alvin1Zhang, 2019-06-14:
@valentinAlkan Sorry for getting back to you so late... Would you kindly check and share whether your issue still exists? Thanks.

valentinAlkan, 2019-06-14:
Sorry I forgot I had this issue opened. I think I discovered that it, along with some other issues I was having, was related to insufficient power. Operation was generally inconsistent until I bumped up the power and then everything seemed fine. 

https://github.com/espressif/arduino-esp32/issues/1274

closed 2019-06-14 by valentinAlkan