espressif/esp-idf #1793: macOS - Failed to find Xtensa toolchain Make error

joshuadcampbell, 2018-03-30:
Here is my PATH:

/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/local/bin:/opt/local/sbin:/Users/jcampbell/esp/xtensa-toolchain/release/xtensa-toolchain/xtensa-lx106-elf/bin/

Here is what I have entered into menuconfig:

"xtensa-lx106-elf-"

In terminal I can execute the binaries by just trying  (ex. xtensa-lx106-elf-gcc) and it works just fine. Why is make telling me it cant find the toolchain?

igrr, 2018-03-31:
xtensa-lx106-elf is the name of the toolchain for the ESP8266. If you are using the ESP32 (which is what this SDK supports), you need to use xtensa-esp32-elf, as described in the getting started guide.

joshuadcampbell, 2018-03-31:
Copy that, I was following this tutorial, https://esp-idf.readthedocs.io/en/v1.0/macos-setup.html, and it wasnt working. Even with the ESP32 toolchain.. Finally got it working by using a newer toolchain not on that tutorial. I still don't quite understand why that error was coming up if the toolchain was working (tested with both lx106 and ESP32 tool chains). Thanks for responding back though!

closed 2018-03-31 by joshuadcampbell

igrr, 2018-03-31:
'v1.0' in the link indicates that the version of the tutorial is the one corresponding to ESP-IDF 1.0, a fairly old release. Newer instructions can be found here: https://esp-idf.readthedocs.io/en/latest/get-started/macos-setup.html

joshuadcampbell, 2018-03-31:
Yes, thats the other one I used. Thanks! Wonder why they haven't updated the mac setup guide.. Thanks again!