espressif/esp-idf #1794: SD card / FAT - file not visible on computer

Reiner1210, 2018-03-30:
Hello,

I have connected a SD card via SPI. I used the example code.
Card is detected an write and stat work.

BUT when I connect the SD card on my computer I do not see the file the ESP32 has created.
The SD card was formatted under Linux with mkfs.vfat /dev/sdf1. Partition type is 0xc.

Any ideas what is going on ..... ?

Reiner1210, 2018-03-31:
An other interesting fact: open / write does not work - only fopen / fwrite.
I assume that open /write will be in conflict with lwip which defines open / write & Co.

igrr, 2018-03-31:
Could you please post the non-working code with open/write? Conflicts with LwIP have been resolved a few months ago, so that shouldn't be a problem. Suggest checking the return values of these functions and `errno` variable.

Reiner1210, 2018-03-31:
It's tricky. No errors, but now file on the SDCard. I can remake an example.

Mit AquaMail Android
https://www.mobisystems.com/aqua-mail

Reiner1210, 2018-03-31:
I tried it again - and it works !
I think I tried it on a FAT32 created by Linux - neither fopen nor open was able to create a file in this case (no error codes but no file on card).  Now I tried this again - and it works. So I will close this issue because I can not reproduce it

closed 2018-03-31 by Reiner1210