espressif/esp-idf #1795: mdns doesn't advertise continuously

formus14, 2018-03-31:
So I have the read the description of  [mdns](http://esp-idf.readthedocs.io/en/latest/api-reference/protocols/mdns.html) and decide to use it to advertise some data while running the esp on APSTA mode.

The code implementation looks pretty much like this 

```
void start_mdns_service()
{
    //initialize mDNS service
    esp_err_t err = mdns_init();
    if (err) {
        printf("MDNS Init failed: %d\n", err);
        return;
    }

    //set hostname
    mdns_hostname_set("my-esp32");
    //set default instance
    mdns_instance_name_set("Jhon's ESP32 Thing");
}
void add_mdns_services()
{
    //add our services
    mdns_service_add(NULL, "_http", "_tcp", 80, NULL, 0);
    mdns_service_add(NULL, "_arduino", "_tcp", 3232, NULL, 0);
    mdns_service_add(NULL, "_myservice", "_udp", 1234, NULL, 0);

    //NOTE: services must be added before their properties can be set
    //use custom instance for the web server
    mdns_service_instance_name_set("_http", "_tcp", "Jhon's ESP32 Web Server");

    mdns_txt_item_t serviceTxtData[3] = {
        {"board","esp32"},
        {"u","user"},
        {"p","password"}
    };
    //set txt data for service (will free and replace current data)
    mdns_service_txt_set("_http", "_tcp", serviceTxtData, 3);
}

extern "C" void app_main(void)
{
	switch(SetupState)
	{
	   case A:
		{
			.
			.
			SetupState	=	B;
			break;

		}

		case B :
		{
			.
			.
			SetupState	=	MDNS;
			break;

		}

		case MDNS :
		{
			start_mdns_service();
			add_mdns_services();
			vTaskDelay(200/portTICK_PERIOD_MS); // check again the delay
			SetupState	=	C;
			break;
		}

		case C :
		{
			while(true){
				.
				// Infinite loop / different functionalities
				.
			}
		}

	} 
}

```

Code compiles and mdns start advertising only for few seconds then it disappears !! 

any help is appreciated

me-no-dev, 2018-03-31:
I do not see you to relay the system events to the mdns server. Do you have that in your code?
https://github.com/espressif/esp-idf/blob/master/examples/protocols/mdns/main/mdns_example_main.c#L74

formus14, 2018-03-31:
@me-no-dev Yes, I have an event handler function that I call it also in the main method

```
esp_err_t event_handler(void *ctx, system_event_t *event)
{
	printf("event: %d\n", event->event_id);

	switch (event->event_id)
        {
           case .. :
                 .
                 .
        }
       mdns_handle_system_event(ctx, event);

       return lws_esp32_event_passthru(ctx, event);
}

```

me-no-dev, 2018-04-03:
@formus14 I ran a test for 2 days and observe no such issues.
Could your main loop be blocking the CPU in a way? WDT enabled? Any other info in the console?

FayeY, 2018-08-28:
Closing as can't reproduce. Please reopen if the issue persists and you can supply more details. Thanks.

closed 2018-08-28 by FayeY

antonio-fiol, 2018-09-15:
I see the same issue (it starts working ok, but after some time the ESP32 is not discoverable anymore).
Source code for my project is a bit of a mess because I started from quite a few different examples, my project does quite a few things (I2C master + slave, PWM using ledc, mDNS + httpserver, esp_http_client, OTA) and my "mixing" skills certainly need to improve.
https://github.com/antonio-fiol/vias
All the wifi+httpserver+mdns stuff is inside my variation on the [simple_ota_example.c](https://github.com/antonio-fiol/vias/blob/master/main/simple_ota_example.c) -- I guess it is not that simple anymore.
I checked that I'm relaying the system events to mdns.
The HTTP server continues running well.
The [I2C slave task](https://github.com/antonio-fiol/vias/blob/master/main/i2c_slave.c#L343) and [ADS task](https://github.com/antonio-fiol/vias/blob/master/main/Adafruit_ADS1015.cpp#L375) continue to run normally.
I have not seen anything weird in the console so far, but I'll try to find out more and get back.
WDT? How do I check?

me-no-dev, 2018-09-15:
Hi @antonio-fiol ! How long is "after some time" around? Which version of IDF are you running?

outermanle, 2018-09-17:
still not work while test the mdns demo - https://github.com/espressif/esp-idf/blob/master/examples/protocols/mdns/main/mdns_example_main.c , it will stuck and no response. @me-no-dev 

```
TX[0][0]: To: 224.0.0.251:5353, Packet[2770]: 
    Q: *U* ESP32-WebServer._http._tcp.local. ANY IN
    Q: *U* esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86

TX[0][0]: To: 224.0.0.251:5353, Packet[3070]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[3270]: 
    Q: *U* ESP32-WebServer._http._tcp.local. ANY IN
    Q: *U* esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86
    S: esp32mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][0]: To: 224.0.0.251:5353, Packet[3370]: 
    Q: *U* ESP32-WebServer._http._tcp.local. ANY IN
    Q: *U* esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86
    S: esp32mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[3570]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86
    S: esp32mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][0]: To: 224.0.0.251:5353, Packet[3670]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86
    S: esp32mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[3870]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86
    S: esp32mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][0]: To: 224.0.0.251:5353, Packet[3970]: 
    Q: ESP32-WebServer._http._tcp.local. ANY IN
    Q: esp32mdns...local. ANY IN
    S: ESP32-WebServer._http._tcp.local. SRV IN 120 [8] 0 0 80 esp32mdns.local.
    S: esp32mdns.local. A IN 120 [4] 172.26.58.86
    S: esp32mdns.local. AAAA IN 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[4170]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [13] ._http._tcp.local.
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.
    A: ESP32-WebServer._http._tcp.local. SRV IN FLUSH 120 [18] 0 0 80 esp32mdns.local.
    A: ESP32-WebServer._http._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32mdns.local. A IN FLUSH 120 [4] 172.26.58.86
    A: esp32mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][0]: To: 224.0.0.251:5353, Packet[4270]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [13] ._http._tcp.local.
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.
    A: ESP32-WebServer._http._tcp.local. SRV IN FLUSH 120 [18] 0 0 80 esp32mdns.local.
    A: ESP32-WebServer._http._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
[... 10 lines cut ...]

TX[0][0]: To: 224.0.0.251:5353, Packet[5370]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [13] ._http._tcp.local.
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.
    A: ESP32-WebServer._http._tcp.local. SRV IN FLUSH 120 [18] 0 0 80 esp32mdns.local.
    A: ESP32-WebServer._http._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32mdns.local. A IN FLUSH 120 [4] 172.26.58.86
    A: esp32mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][1]: To: ff02:0000:0000:0000:0000:0000:0000:00fb:5353, Packet[6370]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [13] ._http._tcp.local.
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.
    A: ESP32-WebServer._http._tcp.local. SRV IN FLUSH 120 [18] 0 0 80 esp32mdns.local.
    A: ESP32-WebServer._http._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32mdns.local. A IN FLUSH 120 [4] 172.26.58.86
    A: esp32mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc

TX[0][0]: To: 224.0.0.251:5353, Packet[6470]: AUTHORITATIVE
    A: _services._dns-sd._udp.local. PTR IN 4500 [13] ._http._tcp.local.
    A: _http._tcp.local. PTR IN 4500 [18] ESP32-WebServer._http._tcp.local.
    A: ESP32-WebServer._http._tcp.local. SRV IN FLUSH 120 [18] 0 0 80 esp32mdns.local.
    A: ESP32-WebServer._http._tcp.local. TXT IN FLUSH 4500 [44] path=/foobar; p=password; u=admin; board=esp32
    A: esp32mdns.local. A IN FLUSH 120 [4] 172.26.58.86
    A: esp32mdns.local. AAAA IN FLUSH 120 [16] fe80:0000:0000:0000:260a:c4ff:fe23:c8dc
```

me-no-dev, 2018-09-17:
do you have this commit: https://github.com/espressif/esp-idf/commit/49125a87bb7e03ce53254f09001df5dc48fb6617 ? Seems you are only sending and not getting packets

formus14, 2019-01-29:
@me-no-dev 

Well, I have been off this topic for a while... yesterday I picked it up again.

I have updated my IDF to **ESP-IDF Release v3.1.2** with Toolchain **xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0**

Running this simple example below with AP mode... still gives me the same result as before, bonjour information appears on any station connected to the ESP AP for few seconds then it totally disappears! 

WDT is enabled with 300ms and 5 seconds timeout for task WDT.
```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "mdns.h"
#include "driver/gpio.h"
#include <sys/socket.h>
#include <netdb.h>

#define EXAMPLE_ESP_WIFI_SSID      "ESP_TEST"
#define EXAMPLE_ESP_WIFI_PASS      "testtest"

#define EXAMPLE_MDNS_HOSTNAME "_test_host"
#define EXAMPLE_MDNS_INSTANCE "_test_inst"

static EventGroupHandle_t wifi_event_group;

const int IP4_CONNECTED_BIT = BIT0;
const int IP6_CONNECTED_BIT = BIT1;

static const char *TAG = "mdns-test";
static bool auto_reconnect = true;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        ESP_LOGI(TAG, "got ip:%s",
                 ip4addr_ntoa(&event->event_info.got_ip.ip_info.ip));
        xEventGroupSetBits(wifi_event_group, IP4_CONNECTED_BIT);

        break;
    case SYSTEM_EVENT_AP_STACONNECTED:
        ESP_LOGI(TAG, "station:"MACSTR" join, AID=%d",
                 MAC2STR(event->event_info.sta_connected.mac),
                 event->event_info.sta_connected.aid);
        break;
    case SYSTEM_EVENT_AP_STADISCONNECTED:
        ESP_LOGI(TAG, "station:"MACSTR"leave, AID=%d",
                 MAC2STR(event->event_info.sta_disconnected.mac),
                 event->event_info.sta_disconnected.aid);
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        esp_wifi_connect();
		xEventGroupClearBits(wifi_event_group, IP4_CONNECTED_BIT);

        break;
    default:
        break;
    }
    mdns_handle_system_event(ctx, event);
    return ESP_OK;
}

static void initialise_wifi(void)
{
    wifi_event_group = xEventGroupCreate();

    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(event_handler, NULL));

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = EXAMPLE_ESP_WIFI_SSID,
            .ssid_len = strlen(EXAMPLE_ESP_WIFI_SSID),
            .password = EXAMPLE_ESP_WIFI_PASS,
            .max_connection = 3,
            .authmode = WIFI_AUTH_WPA_WPA2_PSK
        },
    };
    if (strlen(EXAMPLE_ESP_WIFI_PASS) == 0) {
        wifi_config.ap.authmode = WIFI_AUTH_OPEN;
    }

    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_softap finished.SSID:%s password:%s",
             EXAMPLE_ESP_WIFI_SSID, EXAMPLE_ESP_WIFI_PASS);
}

static void initialise_mdns(void)
{
	//Initialize NVS
	    esp_err_t ret = nvs_flash_init();

    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
	    //structure with TXT records
    mdns_txt_item_t serviceTxtData[3] = {
        {"board","esp32"},
        {"u","user"},
        {"p","password"}
    };
    //initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );
	
    //set mDNS hostname (required if you want to advertise services)
    ESP_ERROR_CHECK( mdns_hostname_set(EXAMPLE_MDNS_HOSTNAME) );
	
    //set default mDNS instance name
    ESP_ERROR_CHECK( mdns_instance_name_set(EXAMPLE_MDNS_INSTANCE) );

    //initialize service
    ESP_ERROR_CHECK( mdns_service_add("ESP32-WebServer", "_http", "_tcp", 1, serviceTxtData, 3) );
}

void app_main()
{	
    initialise_mdns();
    initialise_wifi();
}
```

xpete, 2019-01-29:
I have the same problem. It disappears after a few minutes. Tested with the mdns example from branch release/v3.2.

formus14, 2019-02-01:
> Closing as can't reproduce. Please feel free to reopen if the issue persists and you can supply more details. Thanks.

@FayeY could you re-open the issue please, i can't do it from my side :)

formus14, 2019-02-08:
Ok after using **ESP-IDF Release v3.2** with Toolchain **xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0**

I realized that the whole problem was in the android app where I was browsing the services broadcasted by the ESP.

a good alternative to avoid my mistake is to browse services over the windows terminal, this link will help a lot!

https://www.ibm.com/support/knowledgecenter/en/SSB2MG_4.6.0/com.ibm.ips.doc/concepts/gx_gv_bonjour_service_discovery.htm