espressif/esp-idf #1797: esp_wifi_connect() returns ESP_OK, but no system event posted (ap_loss)

vonnieda, 2018-03-31:
I have found a situation where esp_wifi_connect() returns ESP_OK but no system event is emitted. This makes it impossible to know if the connection attempt is complete and has succeeded or failed.

I have documented the problem at https://esp32.com/viewtopic.php?f=2&t=5183

Here is a sample project that runs on any dev board that demonstrates the issue. It includes a README.md that explains how to set up the test.

[ap_loss.zip](https://github.com/espressif/esp-idf/files/1865753/ap_loss.zip)

In short, if you attempt to connect to an known network with a bad password, get a SYSTEM_EVENT_STA_DISCONNECTED, then try to connect to an unknown network you will never get a system event and on the console you will see `wifi: ap_loss`.

TimXia, 2018-04-02:
@vonnieda Before trying connect to an unknown network, did you call esp_wifi_disconnect() to disconnect from the known network?

vonnieda, 2018-04-02:
@TimXia I've tested with and without esp_wifi_disconnect() - no difference. Also, please note that I specified that the known network uses a bad / incorrect password - so it never connects.

Here is the log from the example application I posted above:

```
I (0) cpu_start: App cpu up.
I (308) heap_init: Initializing. RAM available for dynamic allocation:
I (315) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (321) heap_init: At 3FFB8E80 len 00027180 (156 KiB): DRAM
I (327) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (334) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (340) heap_init: At 4008E8DC len 00011724 (69 KiB): IRAM
I (346) cpu_start: Pro cpu start user code
I (29) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (161) wifi: wifi firmware version: c202b34
I (161) wifi: config NVS flash: enabled
I (161) wifi: config nano formating: disabled
I (161) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (171) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (201) wifi: Init dynamic tx buffer num: 32
I (201) wifi: Init data frame dynamic rx buffer num: 32
I (201) wifi: Init management frame dynamic rx buffer num: 32
I (201) wifi: wifi driver task: 3ffc0360, prio:23, stack:4096
I (211) wifi: Init static rx buffer num: 10
I (211) wifi: Init dynamic rx buffer num: 32
I (221) wifi: wifi power manager task: 0x3ffc4fa0 prio: 21 stack: 2560
I (291) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (291) wifi: mode : sta (24:0a:c4:09:8c:d0)
I (291) simple wifi: SYSTEM_EVENT_STA_START
I (301) simple wifi: Wait for disconnect.
I (301) simple wifi: Connect to KnownNetwork.
I (1311) simple wifi: Wait for disconnect.
I (1641) simple wifi: SYSTEM_EVENT_STA_DISCONNECTED: "KnownNetwork"
I (1641) simple wifi: Connect to test.
I (2641) simple wifi: Wait for disconnect.
I (2851) wifi: ap_loss
```

TimXia, 2018-04-05:
@vonnieda Do `KonwnNetwork` and `UnknownNetwork` exist? I tries your code. First connect to `KnownNetwork` which exists with an incorrect password and then connect to `UnknownNetwork` which does not exist with an password. It did not print `ap_loss`. The following is my log (`mx` is the `KnownNetwork` and 'test' is the `UnknowNetwork`:

I (0) cpu_start: App cpu up.
I (315) heap_init: Initializing. RAM available for dynamic allocation:
I (321) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (328) heap_init: At 3FFB9210 len 00026DF0 (155 KiB): DRAM
I (334) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (340) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (347) heap_init: At 4008F7FC len 00010804 (66 KiB): IRAM
I (353) cpu_start: Pro cpu start user code
I (35) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (100) wifi: wifi firmware version: 2fc15c0
I (100) wifi: config NVS flash: enabled
I (100) wifi: config nano formating: disabled
I (100) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (110) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (140) wifi: Init dynamic tx buffer num: 32
I (140) wifi: Init data frame dynamic rx buffer num: 32
I (140) wifi: Init management frame dynamic rx buffer num: 32
I (140) wifi: wifi driver task: 3ffc0804, prio:23, stack:4096
I (150) wifi: Init static rx buffer num: 10
I (150) wifi: Init dynamic rx buffer num: 32
I (220) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (220) wifi: mode : sta (24:0a:c4:01:4e:04)
I (220) simple wifi: SYSTEM_EVENT_STA_START
I (220) simple wifi: Wait for disconnect.
I (230) simple wifi: Connect to mx.
I (360) wifi: n:9 2, o:1 0, ap:255 255, sta:9 2, prof:6
I (1340) wifi: state: init -> auth (b0)
I (1340) wifi: state: auth -> assoc (0)
I (1340) simple wifi: Wait for disconnect.
I (1340) wifi: state: assoc -> run (10)
I (7790) wifi: state: run -> auth (fc0)
I (7790) wifi: n:9 0, o:9 2, ap:255 255, sta:9 2, prof:6
I (7790) simple wifi: SYSTEM_EVENT_STA_DISCONNECTED: "mx"
I (7790) simple wifi: Connect to test.
I (8800) simple wifi: Wait for disconnect.
I (10210) simple wifi: SYSTEM_EVENT_STA_DISCONNECTED: "test"
I (10210) simple wifi: Connect to mx.
I (10330) wifi: n:9 2, o:9 0, ap:255 255, sta:9 2, prof:6
I (11310) wifi: state: auth -> auth (b0)
I (11320) simple wifi: Wait for disconnect.
I (11320) wifi: state: auth -> assoc (0)
I (11320) wifi: state: assoc -> run (10)
I (17960) wifi: state: run -> auth (fc0)
I (17960) wifi: n:9 0, o:9 2, ap:255 255, sta:9 2, prof:6
I (17960) simple wifi: SYSTEM_EVENT_STA_DISCONNECTED: "mx"
I (17960) simple wifi: Connect to test.
I (18970) simple wifi: Wait for disconnect.
I (20380) simple wifi: SYSTEM_EVENT_STA_DISCONNECTED: "test"

TimXia, 2018-04-05:
@vonnieda Does the bug happen every time?

TimXia, 2018-04-05:
@vonnieda What is the commit ID of IDF which you use? Does the issue happen every time?

mahavirj, 2018-04-05:
@TimXia WiFi firmware version looks different in both logs `wifi firmware version: 2fc15c0`. Could that be the reason? (because I can reproduce this issue in every occurrence)

vonnieda, 2018-04-05:
@TimXia Yes, it happens every time. It's happening on current commit 595ddfd8250426f2e76f7c47ffe9697eb13c455a. Here is that log using that commit:

```
I (0) cpu_start: App cpu up.
I (309) heap_init: Initializing. RAM available for dynamic allocation:
I (316) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (322) heap_init: At 3FFB9408 len 00026BF8 (154 KiB): DRAM
I (328) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (334) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (341) heap_init: At 4008F694 len 0001096C (66 KiB): IRAM
I (347) cpu_start: Pro cpu start user code
I (29) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (95) wifi: wifi firmware version: 00e96c4
I (95) wifi: config NVS flash: enabled
I (95) wifi: config nano formating: disabled
I (95) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (105) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (135) wifi: Init dynamic tx buffer num: 32
I (135) wifi: Init data frame dynamic rx buffer num: 32
I (135) wifi: Init management frame dynamic rx buffer num: 32
I (145) wifi: wifi driver task: 3ffc0a54, prio:23, stack:4096
I (145) wifi: Init static rx buffer num: 10
I (155) wifi: Init dynamic rx buffer num: 32
I (225) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (235) wifi: mode : sta (24:0a:c4:09:8e:60)
I (235) simple wifi: SYSTEM_EVENT_STA_START
I (235) simple wifi: Wait for disconnect.
I (245) simple wifi: Connect to Snoutercakes.
I (365) simple wifi: SYSTEM_EVENT_STA_DISCONNECTED: "Snoutercakes"
I (1245) simple wifi: Wait for disconnect.
I (1245) simple wifi: Connect to UnknownNetwork.
I (1265) wifi: ap_loss
I (2245) simple wifi: Wait for disconnect.
```

TimXia, 2018-04-06:
@vonnieda @mahavirj Yes, it happens every time if I set a empty password for both `KnownNetwork` and `UnknownNetwork`. We will find the root cause ASAP.

mahavirj, 2018-04-06:
@TimXia Thanks, that would be helpful.

vonnieda, 2018-04-06:
@TimXia thank you!

TimXia, 2018-04-12:
@vonnieda We have found the root cause and will fix it ASAP.

vonnieda, 2018-04-12:
@TimXia Excellent, thank you!

vonnieda, 2018-04-12:
Sorry, for the accidental close. Clicked the wrong button :)

FayeY, 2018-05-14:
Hi, please have a try with the latest IDF, and feel free to reopen it if the issue is still unsolved.

closed 2018-05-14 by FayeY

vonnieda, 2018-05-16:
Thanks @FayeY, will try it soon!

spli, 2018-09-03:
Hi

I'm using the master of the IDF and I'm still having this problem. Even testing with the simple_wifi example, the SYSTEM_EVENT_STA_START event is fired, but then nothing. With correct and incorrect wifi credentials, there's no SYSTEM_EVENT_STA_DISCONNECTED or SYSTEM_EVENT_STA_CONNECTED. 

Thanks

spli, 2018-09-04:
Ok. I think what was happening was make clean wasn't cleaning properly, possibly due to some change to the underlying mbedtls project makefile? I deleted the build directory manually and built is again and it worked fine.

rtheil-growlink, 2022-01-21:
Apologies for posting on a closed item, but I found this while searching for a solution to why esp_wifi_connect wouldn't connect properly despite wifi_config_t.ssid and password being correct. I found that while setting the wifi_config_t using esp_wifi_set_config, that the wifi_config_t var went out of scope shortly after setting config. This caused the config in nvs to be incorrect, and would not connect to wifi. This may help someone in the future.