espressif/esp-idf #1801: phy_init: failed to allocate memory for RF calibration data

my-abousamra, 2018-04-01:
Hi,

I enabled WiFi and classic bluetooth and I got that error while running the code on esp32-wroom-32 module

`E (342) phy_init: failed to allocate memory for RF calibration data`

I have a customized partition table

# Name,   Type, SubType, Offset,  Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,

any ideas?

negativekelvin, 2018-04-01:
https://github.com/espressif/esp-idf/blob/5d4c4b19d49f5a61aa3189af93e55140a841eb42/components/bt/include/esp_bt.h#L210-L236

my-abousamra, 2018-04-01:
@negativekelvin 
esp_bt_controller_mem_release helped me to overcome this problem by releasing the unused BLE but in future I will use BLE beside BT classic hence this solution won't be suitable

Weijian-Espressif, 2018-04-08:
Sorry, the memory is not enough, please release  BLE part if you do not use

my-abousamra, 2018-04-10:
@Weijian-Espressif Yes, I already released it plus I disabled unnecessary bluetooth profiles to store some memory

my-abousamra, 2018-04-10:
any way I pulled last update which eliminated this issue. Thanks!

closed 2018-04-10 by my-abousamra