espressif/esp-idf #1802: Shadow Register Delta Error

Ken77107, 2018-04-01:
Hi
I test the example subscribe_publish was ok .
When I test example thing_shadow and I got that error 

shadow: Shadow Register Delta Error
shadow: An error occurred in the loop -28

any ideas?

mahavirj, 2018-04-02:
@Ken77107 Did you check troubleshooting guidelines provided [here](https://github.com/espressif/esp-idf/tree/master/examples/protocols/aws_iot#tls-connection-closes-immediately)?

Ken77107, 2018-04-02:
Hi , @mahavirj 
Thank you for your reply.
I edit policy document and I add about update topics resolve my problem.
Now I can see update data  in aws shadow console , But have new problems ...

1. In the ESP32's serial output, everytime update data will see "Update timed out" logged  not "Update rejected" logged. Why is it so?
2.When I changed the desired structure on cloud console.
"desired": {
    "windowOpen": true
  }
I can't get callback function  "windowActuate_Callback",why?

Thanks

anujdeshpande, 2018-04-06:
@Ken77107 Can you provide more information about what changes you have made to the standard thing shadow example that can be found in `esp-idf/examples/protocols/aws_iot/thing_shadow`?

Ken77107, 2018-04-09:
Hi,@anujdeshpande
I update my edit policy document
{
        "Version": "2012-10-17",
        "Statement": [
        {
              "Action": [
                    "iot:Publish",
                    "iot:Subscribe",
                    "iot:Connect",
                    "iot:Receive"
               ],
              "Effect": "Allow",
              "Resource": [
                              "*"
               ]
         }
   ]
}
everything work fine now , thanks.

mahavirj, 2018-04-12:
@Ken77107 Could you please help to close this?

closed 2018-04-13 by Ken77107