espressif/esp-idf #1804: Connecting to WEP AP causes infinite number of SYSTEM_EVENT_STA_AUTHMODE_CHANGE, pegging wifi task

vonnieda, 2018-04-02:
Connecting to an AP with 40 bit WEP causes the wifi event loop to receive SYSTEM_EVENT_STA_AUTHMODE_CHANGE over and over. If I ignore them, as the documentation suggests, they come so fast that the wifi task pegs an entire core, keeping the idle task from running. I have confirmed this on two different brands of access point: Linksys WRT54G and Belkin N600. 

An example project is attached. Simply enter a 40 bit WEP SSID and password in the example configuration. Here is a log from the example application:

```
 (0) cpu_start: App cpu up.
I (308) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (320) heap_init: At 3FFB8E80 len 00027180 (156 KiB): DRAM
I (327) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (333) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (339) heap_init: At 4008E698 len 00011968 (70 KiB): IRAM
I (346) cpu_start: Pro cpu start user code
I (28) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (169) simple wifi: ESP_WIFI_MODE_STA
I (169) wifi: wifi firmware version: c202b34
I (169) wifi: config NVS flash: enabled
I (169) wifi: config nano formating: disabled
I (169) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (179) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (209) wifi: Init dynamic tx buffer num: 32
I (209) wifi: Init data frame dynamic rx buffer num: 32
I (209) wifi: Init management frame dynamic rx buffer num: 32
I (219) wifi: wifi driver task: 3ffc0360, prio:23, stack:4096
I (219) wifi: Init static rx buffer num: 10
I (229) wifi: Init dynamic rx buffer num: 32
I (229) wifi: wifi power manager task: 0x3ffc4fa0 prio: 21 stack: 2560
I (319) phy: phy_version: 383.0, 79a622c, Jan 30 2018, 15:38:06, 0, 0
I (319) wifi: mode : sta (24:0a:c4:09:8c:d0)
I (319) simple wifi: wifi_init_sta finished.
I (319) simple wifi: connect to ap SSID:clock password:efb402f677
I (439) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
I (439) wifi: state: init -> auth (b0)
I (439) wifi: state: auth -> assoc (0)
I (449) wifi: state: assoc -> run (10)
I (449) wifi: connected with clock, channel 1
I (489) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (609) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (689) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (789) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (899) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (999) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (1099) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (1199) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3249) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3379) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3449) wifi: pm start, type:0

I (3459) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3559) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3659) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3769) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3869) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (3989) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (4069) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (4169) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (4279) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (5909) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6019) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6119) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6219) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6319) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6439) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6529) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6599) event: sta ip: 192.168.2.4, mask: 255.255.255.0, gw: 192.168.2.1
I (6599) simple wifi: got ip:192.168.2.4
I (6629) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6729) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6839) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (6939) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (14929) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (15029) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
I (15129) simple wifi: SYSTEM_EVENT_STA_AUTHMODE_CHANGE
... continues forever...
```

[wep_bug.zip](https://github.com/espressif/esp-idf/files/1868569/wep_bug.zip)

vonnieda, 2018-04-02:
One more note, I am sometimes seeing this message:

`E (5086) wifi: pp.c 3194`

XinDeng11, 2018-04-11:
Hi,@vonnieda, thank you for reminding us, the issue is happened in WEP mode due to old auth mode record wrong, we have fixed it, you can check the commit 56a798e2 after it merged.

vonnieda, 2018-04-11:
@XinDeng11 That's great news, thank you!

XinDeng11, 2018-04-13:
Hi, @vonnieda. Due to some operation mistake, we recheck in the changes and close the commit 56a798e2, please check the commit fb24ff41, sorry.

vonnieda, 2018-04-13:
Thank you @XinDeng11, but I don't see this commit. Has it been pushed?

XinDeng11, 2018-04-14:
@vonnieda Yeah, it has already pushed, put not merge yet

XinDeng11, 2018-04-17:
Hi@vonnieda, there may be still need some times to merge, if it urge, maybe I can offer you a template lib before it merge

vonnieda, 2018-04-17:
Thank you @XinDeng11, I can wait until it's merged.

closed 2018-04-24 by FayeY

vonnieda, 2018-07-23:
Hi @XinDeng11 and @FayeY: Can you tell me if this fix has been merged into the 3.0.2 release branch? I am trying to follow the history to figure it out, but since it's a binary it's hard for me to tell if it's included.

Thanks,
Jason