espressif/esp-idf #1805: What does this error means? core: E () WIFI: TIMQ_NUL

ali-slim-sst, 2018-04-02:
Sometimes I get this error when the ESP32 is trying to connect to a WiFi network, I'm using IDF v2.1.1. After this error code is received the internal WiFi task stops calling event_handler which results in it stops trying to reconnect.

liuzfesp, 2018-04-10:
Hi @ali-slim-sst it means there timer Queue is full. In IDF v3.0, the timer is re-designed. Is it possible for you to switch to idfv3.0?

vonnieda, 2018-04-12:
@ali-slim-sst This sounds like the same problem I am experiencing in this thread: https://esp32.com/viewtopic.php?f=2&t=5183 and in this issue: https://github.com/espressif/esp-idf/issues/1797

In short, sometimes SYSTEM_EVENT_STA_DISCONNECTED does not appear, even though esp_wifi_connect() returns ESP_OK.

liuzfesp, 2018-04-12:
@XinDeng11 please help to check it.

XinDeng11, 2018-04-12:
Hi @ali-slim-sst @vonnieda， there are one case may cause the issue, you are in an environment where have more than two same SSID APs and one of them is you try to connect. However, I am not sure whether it is the scenario I guess, so can you give me more information like the config parameter you set and your environment when you meet the issue, thanks.

XinDeng11, 2018-04-16:
Hi,@ali-slim-sst, I mean I need the scan type you set when set station config, because when in all channel scan type, we will scan all match SSID APs information in the list, and try with the best one first, if it fail to connect and we found there are still others in the list, we wouldn't send disconnect while try to connect with the others, and put the one which fail to connect to the blacklist until we connect success or all the APs in the list fail to connect, so I hope to know whether you set to scan all the channel before connect

ali-slim-sst, 2018-04-17:
Hi @XinDeng11 , I'm not setting scan_method as in the WiFi connection example, I'm just setting SSID and password. 

So you recommend using WIFI_ALL_CHANNEL_SCAN to overcome the multiple SSID's issue?

Thanks!

XinDeng11, 2018-04-17:
@ali-slim-sst No, All channel scan is used to scan the best AP in a multiple SSID's environment. However, only when you in this config may cause don't send disconnect when not connect with the AP, may be you can offer the logs for us when you meet the issue again

ali-slim-sst, 2018-04-17:
@XinDeng11 unfortunately I cannot offer any logs, the test device is on a truck so I cannot log through serial. Is there a way monitor and recover the WiFi task? 
Currently I have a separate task to monitor the WiFi while its disconnected from the AP and forcing a software reset if SYSTEM_EVENT_STA_DISCONNECTED not called within a set time frame. However this is bad for my application I prefer more elegant error recovery that doesn't involve reset.

XinDeng11, 2018-06-18:
Hi@ali-slim-sst, I found there might be one case like yours, but I need make sure whether you invoked scan API when the STA disconnect and was trying to reconnect?

FayeY, 2018-06-27:
Hi @ali-slim-sst , 

Is there any update? Could you provide more information to help debug this issue?

Thanks.

ali-slim-sst, 2018-07-04:
Hi @FayeY 
I will do more testing in the next few weeks and update this issue.

ali-slim-sst, 2018-08-29:
@FayeY sorry I didn't have time to work on that yet.

FayeY, 2018-08-29:
Closing this, please feel free to reopen if this is still an issue after you testing it and you can supply more details. Thanks.

closed 2018-08-29 by FayeY