espressif/esp-idf #1807: Code Example: Reading from an I2C device with addressable registers

mws-rmain, 2018-04-02:
The existing I2C example does not illustrate how to communicate (especially read from) a device with addressible registers (the BH1750 has no registers).
[This gist](https://gist.github.com/mws-rmain/2ba434cd2a3f32d6d343c1c60fbd65c8) illustrates communication with a MMA8451Q accelerometer, which has multiple registers. This illustrates use of the ['I2C repeated start'](https://www.i2c-bus.org/repeated-start-condition/) between sending (writing) the register address, and reading the value(s).

Please feel free to add this to the examples in ESP-IDF.

mainr, 2018-04-08:
Thanks for your code.
However, if you had bothered to look at the GIST I posted, I provided code that performs the same task.

koobest, 2018-04-09:
Hi, @mainr, I can't open the  gist link you provided.

FayeY, 2018-04-24:
Closing due to lack of response, please feel free to reopen if this is still an issue and you can supply more details.
Thanks.

closed 2018-04-24 by FayeY

sherief, 2019-09-18:
This would be super helpful as an example in esp-idf. Most non-trivial devices require this and bundling it with esp-idf would save a lot of man-hours of googling around at least.

mws-rmain, 2019-09-18:
Since @koobest had problems accessing the gist, I've pasted the code below:

This example is based on the ESP-IDF i2c example. However, the example provided does not illustrate how to read from an I2C device with registers.

The simple device in the original example (BH1750) is read with the following sequence:

```
send: <START> condition
send: (DEVICE_ADDR << 1) | READ_BIT
read: ACK
read: requested bytes, ACKing all but last byte
send: <STOP> condition
```
This example (MMA8451 accelerometer) is read with the following sequence:

```
send: <START> condition
send: (DEVICE_ADDR << 1) | WRITE_BIT
read: ACK
send: register address to read
read: ACK
send: <START> condition
send: (DEVICE_ADDR << 1) | READ_BIT
read: ACK
read: requested bytes, ACKing all but last byte
send: <STOP> condition
```
Here is the example code, which periodically reads the X, Y & Z acceration components, and prints them to  the console.

```
/* i2c_restart - Example
   For other examples please check:
   https://github.com/espressif/esp-idf/tree/master/examples
   This example code is in the Public Domain (or CC0 licensed, at your option.)
   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "esp_log.h"
#include "driver/i2c.h"

/**
 * TEST CODE BRIEF
 *
 * This example will show you how to use I2C module by configuring & reading a MMA8451
 * accelerometer. This device has a registered I2C interface, which requires the I2C
 * bus to issue a 'repeated start' between sending the register to be accessed, and
 * the subsequent read command.
 * (see: https://www.i2c-bus.org/repeated-start-condition/ )
 *
 * - read external i2c sensor, here we use a MMA8451 (accelerometer) for instance.
 *
 * Pin assignment:
 *
 * - master:
 *    GPIO18 is assigned as the data signal of i2c master port
 *    GPIO19 is assigned as the clock signal of i2c master port
 *
 * Connection:
 *
 * - connect sda/scl of sensor (MMA8451) with GPIO18/GPIO19
 * - no need to add external pull-up resistors, driver will enable internal pull-up resistors.
 *
 * Test items:
 *
 * - i2c master(ESP32) will read data from i2c slave (MMA8451 accelerometer)
 */

#define SAMPLE_PERIOD_MS		200
[... 218 lines cut ...]
    	acc.X = byte_swap( acc.X );
    	acc.Y = byte_swap( acc.Y );
    	acc.Z = byte_swap( acc.Z );

    	// shift each value to align 14-bits in 16-bit ints
    	acc.X /= 4;
    	acc.Y /= 4;
    	acc.Z /= 4;

        ESP_LOGI( TAG, "Accelerometer err:%d  x:%5d  y:%5d  z:%5d", err, acc.X, acc.Y, acc.Z );

        vTaskDelay( pdMS_TO_TICKS( SAMPLE_PERIOD_MS ) );
    }
}

void app_main()
{
    i2c_master_init();

    mma8451_init();

    xTaskCreate(i2c_test_task, "i2c_test_task_0", 1024 * 2, (void* ) 0, 10, NULL);
}

```