espressif/esp-idf #1809: WiFi sends data related problems with UDP and uses light sleep.

Gaoqianjin, 2018-04-03:
Requirements:
     WiFi sends a packet of data with UDP, light sleep for a second, then sends the next packet data, which keeps circulating. 
Problem description ：
     No package has been sent successfully, it has always been Not enough space.
Thanks:
Thanks to everyone who browsed the question .
Code ：
#define TEST_BUFFER_ADDRESS 0x3f400000
#define EXAMPLE_WIFI_SSID "iperftest"
#define EXAMPLE_WIFI_PASS "12345678"
#define REMOTE_PORT 1112

/* FreeRTOS event group to signal when we are connected & ready to make a request */
static EventGroupHandle_t wifi_event_group;

/* The event group allows multiple bits for each event,
   but we only care about one event - are we connected
   to the AP with an IP? */
const int CONNECTED_BIT = BIT0;

static const char *TAG = "Log_Test";
#define SEND_BUF_LEN 480
#define SEND_START_LEN 5
#define SEND_END_LEN 3
const unsigned char start[5]="start";
const unsigned char end[3]="end";
unsigned char test_buffer[1440];

static void timer_wake_init(void)
{
    const int time_wakeup_sec = 1;
    esp_sleep_enable_timer_wakeup(time_wakeup_sec * 1000000);
}

static void udp_thread(void *p)
{
	int i = 0;
	int ret=0;
	int sock;
	struct sockaddr_in toAddr;
  
	
	LWIP_UNUSED_ARG(p);

	sock = socket(AF_INET,SOCK_DGRAM,IPPROTO_UDP);
	if(sock < 0)
	{
		ESP_LOGI(TAG, "socket err");
	}
        
        fcntl(sock,F_SETFL,0);

	memset(&toAddr,0,sizeof(toAddr));
	toAddr.sin_family=AF_INET;
	toAddr.sin_addr.s_addr=inet_addr("192.168.43.1");
	toAddr.sin_port = htons(REMOTE_PORT);
	printf("test_buffer is %p",test_buffer);
	int fail_count=0;
	while(1) {
	xEventGroupWaitBits(wifi_event_group, CONNECTED_BIT, false, true, portMAX_DELAY);
		sendto(sock,start,SEND_START_LEN,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
		
		for(i=0; i<80; i++)
		{
			vTaskDelay(1/portTICK_RATE_MS);
		
			ret=sendto(sock,test_buffer,1440,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
			
			if(ret<0)
			{
				
					perror("fail reason is:\n");
                                        fail_count++;
			}
			
		}
		
	  	vTaskDelay(1000/portTICK_RATE_MS); 
                timer_wake_init();
               esp_light_sleep_start();
		
		printf("fail_count=%d\n",fail_count);
		fail_count=0;
		sendto(sock,end,SEND_END_LEN,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
	}
	close(sock);
	vTaskDelete(NULL);
}

static void udp_client_init(void)
{
	int ret;
	ret = xTaskCreate(&udp_thread, "udp_client_task", 2048, NULL, 5, NULL);
	if(ret != pdPASS) 
        {
		ESP_LOGI(TAG, "create thread udp_client_task failed");
	}
        	
}

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
        xEventGroupSetBits(wifi_event_group, CONNECTED_BIT);
		udp_client_init();
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
        esp_wifi_connect();
        xEventGroupClearBits(wifi_event_group, CONNECTED_BIT);
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void initialise_wifi(void)
{
    
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_WIFI_SSID,
            .password = EXAMPLE_WIFI_PASS,
        },
    };
    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(WIFI_IF_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start());
    esp_wifi_set_ps(WIFI_PS_MODEM);
}   

void app_main()
{

    	nvs_flash_init();
	initialise_wifi();
}

Gaoqianjin, 2018-04-20:
Hello, in the above code, I call the vTaskDelay (1000/portTICK_RATE_MS) function should be an idle task, and then I then use light sleep why it is still unsuccessful?            
Thanks very much

Gaoqianjin, 2018-04-20:
Hello, I saw in ESP-IDF Programming Guide: 
   WiFi/BT and sleep modes
In deep sleep mode, wireless peripherals are powered down. Before entering sleep mode, applications must disable WiFi and BT using appropriate calls ( esp_bluedroid_disable(), esp_bt_controller_disable(), esp_wifi_stop()).

WiFi can coexist with light sleep mode, allowing the chip to go into light sleep mode when there is no network activity, and waking up the chip from light sleep mode when required. However APIs described in this section can not be used for that purpose. esp_light_sleep_start() forces the chip to enter light sleep mode, regardless of whether WiFi is active or not. Automatic entry into light sleep mode, coordinated with WiFi driver, will be supported using a separate set of APIs.

I want to do this: 
 As the code above does, send a frame and go to sleep. Wake up and send the next frame. 
WiFi can coexist with light sleep mode, allowing the chip to go into light sleep mode when there is no network activity, and waking up the chip from light sleep mode when required.
I need your help            
Thanks very much

TimXia, 2018-06-07:
@Gaoqianjin Light sleep is ready in the latest IDF and IDF v3.1 rc1. Please have a try.

closed 2018-06-13 by FayeY