espressif/esp-idf #1810: Query on aws example

hvishu, 2018-04-03:
From the examples/aws_iot/thing_shadow/main/thing_shadow_sample.c

in aws_iot_task(), I see below line
    while(NETWORK_ATTEMPTING_RECONNECT == rc || NETWORK_RECONNECTED == rc || SUCCESS == rc) {
...
...
}
Seems if the error returned is "NETWORK_ATTEMPTING_RECONNECT" or "NETWORK_RECONNECTED", attempt will be made to connect and be in while loop.  

The list of errors in IoT_Error_t, is quite a few.  Are there any of these errors possible?  If so, what is the sequence to recover from tem?

mahavirj, 2018-04-04:
@hvishu This looks to be more relevant to AWS IoT SDK, could you please ask [here](https://github.com/aws/aws-iot-device-sdk-embedded-C)

hvishu, 2018-04-06:
Raised this query in AWS IoT SDK, closing this issue here...

closed 2018-04-06 by hvishu