espressif/esp-idf #1811: BLE reconnect fails with iPhone 7

thomasschalch, 2018-04-03:
Hi everyone

I use a GATT Server very similar to the [GATT Security Server Demo](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/gatt_security_server). The first pairing works quite reliable using iOS (LightBlue Explorer) and Android. Now I've noticed that using an iPhone 7 (iOS 11.2), I cannot reconnect to the GATT Server, unless I remove the device from the bonded devices in the phone's settings.
Interestingly, using an iPhone X or my Android device, reconnection works perfectly.

Has anyone a clue if this is a general issue in esp-idf, or a wrong configuration in my application?

Thanks in advance for your help!

Here are some outputs in the log, for reconnecting with different devices:

iPhone 7 (iOS 11.2) --> Reconnection fails
```
W (13:39:11) BLE GATT event = 14
I (13:39:11) BLE GATT Connection event, conn_id 0, remote 6f:0a:fe:0f:ed:9b:
W (13:39:11) BLE GAP Event 8
I (13:39:12) Remote BD_ADDR: 6F0AFE0FED9B
I (13:39:12) Address type = 1
I (13:39:12) Pair status = SUCCESS
I (13:39:12) SCS: send command: hidePasskey
W (13:39:12) BLE GAP Event 20
I (13:39:12) BLE GAP Update Connection Params
I (13:39:12) Min connection interval = 6
I (13:39:12) Max connection interval = 6
I (13:39:12) Slave latency = 0
I (13:39:12) Connection timeout = 72
W (13:39:12) BLE GATT event = 15
**I (13:39:12) BLE GATT Disconnect event, reason: 8**
W (13:39:12) BLE GAP Event 6
I (13:39:12) Advertising start success
```

iPhone X (iOS 11.2.2) --> Recoonection successful
```
W (13:42:01) BLE GATT event = 14
I (13:42:01) BLE GATT Connection event, conn_id 0, remote 77:c3:25:36:55:34:
W (13:42:01) BLE GAP Event 8
I (13:42:01) Remote BD_ADDR: 77C325365534
I (13:42:01) Address type = 1
I (13:42:01) Pair status = SUCCESS
I (13:42:01) SCS: send command: hidePasskey
W (13:42:02) BLE GAP Event 20
I (13:42:02) BLE GAP Update Connection Params
I (13:42:02) Min connection interval = 6
I (13:42:02) Max connection interval = 6
I (13:42:02) Slave latency = 0
I (13:42:02) Connection timeout = 600
W (13:42:02) BLE GATT event = 4
I (13:42:02) BLE GATT MTU changed to 517
W (13:42:02) BLE GATT event = 1
I (13:42:02) BLE GATT Read handle: 45
W (13:42:02) BLE GATT event = 1
I (13:42:02) BLE GATT Read handle: 48
W (13:42:02) BLE GATT event = 1
I (13:42:02) BLE GATT Read handle: 53
W (13:42:02) BLE GAP Event 20
I (13:42:02) BLE GAP Update Connection Params
I (13:42:02) Min connection interval = 24
I (13:42:02) Max connection interval = 24
I (13:42:02) Slave latency = 0
I (13:42:02) Connection timeout = 72
```

Android (Blackberry) --> Reconnection successful
```
W (13:44:35) BLE GATT event = 14
I (13:44:35) BLE GATT Connection event, conn_id 0, remote 5e:a0:3b:58:37:a6:
W (13:44:36) BLE GAP Event 8
I (13:44:36) Remote BD_ADDR: 5EA03B5837A6
I (13:44:36) Address type = 1
I (13:44:36) Pair status = SUCCESS
I (13:44:36) SCS: send command: hidePasskey
W (13:44:36) BLE GATT event = 1
I (13:44:36) BLE GATT Read handle: 52
I (13:44:36) Read last sent notification for connection state.
W (13:44:36) BLE GATT event = 21
W (13:44:36) Uncaught GATT event: 21
W (13:44:36) BLE GATT event = 1
I (13:44:36) BLE GATT Read handle: 50
I (13:44:36) Read current WLAN configuration.
W (13:44:36) BLE GATT event = 21
W (13:44:36) Uncaught GATT event: 21
W (13:44:37) BLE GAP Event 20
I (13:44:37) BLE GAP Update Connection Params
I (13:44:37) Min connection interval = 6
I (13:44:37) Max connection interval = 6
I (13:44:37) Slave latency = 0
I (13:44:37) Connection timeout = 2000
W (13:44:37) BLE GAP Event 20
I (13:44:37) BLE GAP Update Connection Params
I (13:44:37) Min connection interval = 39
I (13:44:37) Max connection interval = 39
I (13:44:37) Slave latency = 0
I (13:44:37) Connection timeout = 600
W (13:44:37) BLE GAP Event 20
I (13:44:37) BLE GAP Update Connection Params
I (13:44:37) Min connection interval = 0
I (13:44:37) Max connection interval = 0
I (13:44:37) Slave latency = 0
I (13:44:37) Connection timeout = 2000
```

tssh2005, 2018-04-10:
I found out that if I call `esp_ble_set_encryption(..., ESP_BLE_SEC_ENCRYPT_MITM);` **not** in the `ESP_GATTS_CONNECT_EVT`, but in the read/write-events, everything works fine.

Is this usage of the function `esp_ble_set_encryption` allowed? It is a bit strange, that no GAP-event is triggered if the function is called in the GATT read/write-events. Does anyone know whether the link then is still secure?

chegewara, 2018-04-10:
There is/was some bug related to iPhone 8 and iOS 11.2, it may or may not be your case:
https://github.com/espressif/esp-idf/issues/1407#issuecomment-372895714

As far as i know `esp_ble_set_encryption ` should be set before connection, but again i may be wrong.

semireg, 2018-04-12:
For what it's worth, [Apple design guidelines](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf) page 62 notes:

> The accessory should not request pairing until an ATT request is rejected using the Insufficient Authentication error code. See the Bluetooth 4.0 specification, Volume 3, Part F, Section 4 for details.
If, for security reasons, the accessory requires a bonded relationship with the Central, the Peripheral should reject the ATT request using the Insufficient Authentication error code, as appropriate. As a result, the Apple product may proceed with the necessary security procedures.

tssh2005, 2018-04-12:
Thanks for your replies!

The system is now working with all tested phones, if I completely remove the `esp_ble_set_encryption` request. For setting up the connection, I use `ESP_LE_AUTH_REQ_SC_MITM_BOND` as requested authentication mode. So the ESP32 should refuse to connect with anyone who does not support the requested authentication mode. Therefore the connection does not need to be updated using `esp_ble_set_encryption`, because the connection will already fulfill the requested condition in every case.

I suppose that the ESP32 requested more security before the connection was fully set up, causing those issues with some phones. However, this is only what I learned by trying out. If anyone could confirm this, I would appreciate very much.

closed 2018-04-16 by tssh2005

dconyers, 2018-05-30:
@tssh2005 - We're hitting this exact issue; thanks so much for going to the effort to post your experience. We have made the changes as you prescribed (removed esp_ble_set_encryption() completely from our code and just changed the ESP_BLE_SM_AUTHEN_REQ_MODE to ESP_LE_AUTH_REQ_SC_MITM_BOND in the esp_ble_gap_set_security_param function call. This allows the iPhone 7 (and all other devices) to connect, but does not "Force" the connections to be bonded. I'm not overly savvy in BLE, so I'm uncertain how to confirm if they are even encrypted/paired. 

Can you please confirm that your setup (without the esp_ble_set_encryption() function call) still sets up the connections to be encrypted/bonded)?

Thanks in advance for any guidance you might be able to provide!

Thansk, doug.

tssh2005, 2018-06-01:
Hi @dconyers 

Yes, we checked our application with a BLE sniffer and could see that the connection is still encrypted as expected. Bonding also works. However, we did not check what happens, if the initiator of the connection does not want to create an encrypted connection.
In our application, the ESP32 always communicates with our app on a smartphone or tablet and possibly sensitive data is only transmitted from the app to the ESP32. As the smartphone will always try to create a connection that is as secure as possible, the mentioned case is not really interesting for us.

I'm also not a BLE expert, but in my understanding, both parties must be willing to create a bond. If one is not, secure communication can take place after pairing, but no bond is created. The connection is not less secure if not bonded. So it is neither possible nor necessary for security reasons to force bonding. Please correct me if I'm wrong!

In any case, if security is important in your application, I would recommend to spend an hour in investigating the connection using a BLE sniffer.