espressif/esp-idf #1812: Wifi mode NONE

Elektrik1, 2018-04-04:
After latest commit, on WiFi initialization it errors, and WiFi is not enabled:
`W (1477) wifi: Do not support NONE mode any more!`

Would be nice to have more information about what mode this error is.

jack0c, 2018-04-04:
None mode equals do not open wifi, so this mode was deleted.

TimXia, 2018-04-05:
@Elektrik1 WiFi supports `Station`, `Softap` and `Station+Softap` mode. It also supported `None` mode in previous IDF version. Now, `None` mode is not supported any more. `esp_wifi_set_mode()` should be called after `esp_wifi_init()` and before `esp_wifi_start()`. If WiFi is not used any more, please call `esp_wifi_stop()` to disable WiFi and `esp_wifi_init()` to de-initialize WiFi.

negativekelvin, 2018-04-05:
> Generally, this mode is used for Sniffer

So now if you want to use sniffer only you have to pick a mode which will allocate some unneeded memory?

TimXia, 2018-04-05:
@negativekelvin Sorry, I forgot sniffer mode. I will fix it ASAP.

heyinling, 2018-04-05:
@negativekelvin BTW, sniffer can be used with STA / SoftAP / STA+SoftAP mode (although force switch to some channel will cause current mode not work properly). In this case you can use STA mode without connected with AP, then enable sniffer.

negativekelvin, 2018-04-05:
Ok but it will allocate small amount of unneeded memory correct?

Elektrik1, 2018-04-05:
So basically wifi_mode_t cannot be assigned as WIFI_MODE_NULL any more ?
http://esp-idf.readthedocs.io/en/latest/api-reference/wifi/esp_wifi.html#_CPPv211wifi_mode_t

pantaluna, 2018-04-12:
FYI The official example in ~/examples/wifi/iperf/main/ still uses WIFI_MODE_NULL

pantaluna, 2018-04-12:
This code sequence causes a crash with the latest git commit (of Apr11th). So it does not show the warning "W (259) wifi: Do not support NONE mode any more!" but crashes instead.

CODE:
```
    ESP_ERROR_CHECK(esp_event_loop_init(wifi_event_handler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_NULL)); // crash in this call
```

LOG:
```
...
I (1057) wifi: Init dynamic tx buffer num: 32
I (1057) wifi: Init data frame dynamic rx buffer num: 32
I (1067) wifi: Init management frame dynamic rx buffer num: 32
I (1067) wifi: wifi driver task: 3ffbb5a8, prio:23, stack:4096
I (1077) wifi: Init static rx buffer num: 10
I (1077) wifi: Init dynamic rx buffer num: 32
C:/myiot/esp/esp-idf/components/newlib/locks.c:134 (lock_acquire_generic)- assert failed!
abort() was called at PC 0x40082c96 on core 0
0x40082c96: lock_acquire_generic at C:/myiot/esp/esp-idf/components/newlib/locks.c:134 (discriminator 1)
Backtrace: 0x4008d3e0:0x3ffc3d40 0x4008d583:0x3ffc3d60 0x40082c96:0x3ffc3d80 0x40082dc1:0x3ffc3db0 0x400d9bda:0x3ffc3dd0 0x400dce15:0x3ffc40e0 0x40082bd1:0x3ffc4110 0x400ef6c9:0x3ffc4160 0x400eff47:0x3ffc41a0 0x40086c74:0x3ffc41c0
0x4008d3e0: invoke_abort at C:/myiot/esp/esp-idf/components/esp32/panic.c:648
0x4008d583: abort at C:/myiot/esp/esp-idf/components/esp32/panic.c:648
0x40082c96: lock_acquire_generic at C:/myiot/esp/esp-idf/components/newlib/locks.c:134 (discriminator 1)
0x40082dc1: _lock_acquire_recursive at C:/myiot/esp/esp-idf/components/newlib/locks.c:169
0x400d9bda: _vfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
0x400dce15: vprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vprintf.c:39
0x40082bd1: esp_log_write at C:/myiot/esp/esp-idf/components/log/log.c:215
0x400ef6c9: wifi_set_mode_process at ??:?
0x400eff47: ieee80211_ioctl_process at ??:?
0x40086c74: ppTask at ??:?

CPU halted.
```

dpgeorge, 2018-05-14:
If this mode is no longer supported it would be good to remove it from the docs, eg here: http://esp-idf.readthedocs.io/en/latest/api-guides/wifi.html#wi-fi-mode

TimXia, 2018-06-21:
@Elektrik1 @pantaluna @dpgeorge The latest IDF master branch and IDFv3.1 will support WiFi NONE mode again. Please check it.

bschofield, 2018-06-24:
@TimXia I still seem to get this error with both _master_ and _release/v3.1_.

If I grep for the string, I see that it is still in one of the binary libraries:
```
~/tools/esp/esp-idf$ fgrep -R 'Do not support NONE mode any more'
Binary file components/esp32/lib/libnet80211.a matches
```
Does the _espressif/esp32-wifi-lib_ repo need updating, perhaps? Or is it possible for me to build a local copy?

Thank you for your help!

TimXia, 2018-06-26:
@bschofield espressif/esp32-wifi-lib repo has been updated. Did you update the wifi submodule which is in components/esp32/lib? I have tested te master and release/v3.1. The log 'Do not support NONE mode any more' is removed.

closed 2018-07-11 by FayeY