espressif/esp-idf #1815: RMT ISR null pointer dereference when initialising driver with multiple RMT channels in use after an unexpected reset

DavidAntliff, 2018-04-04:
Refer https://esp32.com/viewtopic.php?f=13&t=5252&p=22750#p22750

Applies to release/v3.0 f8515688cbc9f7227f4c5ef1706eb3f60c696b13

When two or more RMT channels are in use, and an unexpected reset occurs (unrelated to RMT), the RMT peripheral will continue to run and generate interrupts. There is a race condition between the interrupt handler installed by the first call to rmt_driver_install responding to an interrupt for the other (uninitialised) channel, and the second call to rmt_driver_install which would actually allocate the data structure for the second channel.

The following before a call to `rmt_config()` will avoid the issue:
```
    periph_module_disable(PERIPH_RMT_MODULE);
    periph_module_enable(PERIPH_RMT_MODULE);
```

However this might affect programs that sleep, since it will reset the peripheral. So it seems to me that the better approach is for the ISR to _not_ assume that `p_rmt_obj[channel]` is always non-NULL, and ignore interrupts for uninitialised channels.

I'll submit a PR shortly that ignores interrupts when the channel data structure has not been allocated yet. One side effect of this is that a "case 2" error occurs shortly after the second channel is installed, however the peripheral does appear to function correctly.

koobest, 2018-06-03:
Hi, @DavidAntliff 
    We reset some peripherals at system startup, including rmt, https://github.com/espressif/esp-idf/blob/64b56beff5a4d6a165bd88027f585bfe51fbb990/components/esp32/clk.c#L240
 please try the latest IDF.
thanks !!

DavidAntliff, 2018-06-04:
Hi @koobest, thanks for letting me know, I'll try it out and report back if I see any issues.

closed 2018-06-05 by FayeY

guirespi, 2023-07-14:
Hi, I was in ESP IDF v4.4.2 and in my code, i made a loop that constantly turn on and turn off rgb leds. At certain moment the ISR sets the NULL pointer. Can you tell me how to fix this?

guirespi, 2023-07-14:
I have to add that I only use RMT channel 0 with 4 blocks of the rmt buffer.