espressif/esp-idf #1818: with ADV_TYPE_NONCONN_IND esp32 still can be connected to

chegewara, 2018-04-04:
Maybe i dont understand something, but i thought that `ADV_TYPE_NONCONN_IND` suppose to mean that esp32 server is not connectable, but i can connect to esp32 with this flag set.

chegewara, 2018-04-05:
Looks like every device with set flags in advertising packet is connectable.

chegewara, 2018-04-21:
This is my advertising params setup:
```
static esp_ble_adv_params_t adv_params = {
    .adv_int_min        = 0x20,
    .adv_int_max        = 0x40,
    .adv_type           = ADV_TYPE_NONCONN_IND,
    .own_addr_type      = BLE_ADDR_TYPE_PUBLIC,
    //.peer_addr            =
    //.peer_addr_type       =
    .channel_map        = ADV_CHNL_ALL,
//    .adv_filter_policy = ADV_FILTER_ALLOW_SCAN_WLST_CON_WLST
};
```
My advertising data is raw data without set flags, and with nRF connect esp32 still is recognized as connectable device. For testing i have another android device and if i setup advertising parameters in nRF connect with flags, android device is connectable, without flags its no longer connectable.

Only difference i can find is that esp32 is setup with public address and android is with private.

Yulong-espressif, 2018-04-22:
@chegewara  I will check it and find what the problem it is, thanks.

chegewara, 2018-04-22:
@Yulong-espressif For some reasons the same code with the same android device works good, its not connectable anymore. I will update if some change. Thanks.

chegewara, 2018-04-22:
[image]
[image]

crespum, 2018-07-23:
I can reproduce the same error using the [iBeacon](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/ble_ibeacon) example (it uses `ADV_TYPE_NONCONN_IND` parameter too) and IDF v3.0.2. I've checked with 3 different tools that the device is still connectable. This seems to have been fixed in v3.1-beta1.