espressif/esp-idf #1819: noise of a2dp_sink

mingshima, 2018-04-04:
Hello all,
I use A2DP source and sink on esp32 github. I just simply did a test, I use my smartphone to connect to the A2DP sink by bluetooth and play music, I chose internal dac. When I don't play music, there is no noise, when I play music, there is a lot of noise. I found that the sample frewuency is 2777Hz
```
I (12121) I2S: PLL_D2: Req RATE: 44100, real rate: 2777.000, BITS: 16, CLKM: 30,
 BCK: 60, MCLK: 30.234, SCLK: 88864.000000, diva: 64, divb: 14
```
Why will this happen and why there are so much noise? I have tried to add a low pass filter between ESP32 board and speaker, but can't get improve. Can someone explain to me and help me to decrease the noise? Many thanks.

koobest, 2018-04-08:
@mingshima The internal DAC has only 8 bit width, so there is noise.

closed 2018-04-10 by FayeY

macaba, 2018-10-24:
For anyone else who comes across this, please see my comment:
https://github.com/espressif/esp-idf/issues/1614#issuecomment-432858740

MrityunjaiKumar, 2019-01-12:
> For anyone else who comes across this, please see my comment:
> [#1614 (comment)](https://github.com/espressif/esp-idf/issues/1614#issuecomment-432858740)

@macaba this step do help, but there are still noises here and there, on oscilloscope it shows sudden peaks in waveform. Tried low pass filter but that doesn't help. Any clue?

ronyeapen, 2019-01-16:
Hello all, I am new to esp32 and I started off with A2DP_sink as my first code to experiment. @macaba Thank you very much for that comment. It helps a LOT! It improved the sound quality by 90%. But as @MrityunjaiKumar  mentioned, there is still noise and 'crack' like noise every so often when playing music. Any ideas why that is the case? any ideas on how it can be fixed?