espressif/esp-idf #1820: Wifi Scan does not return all AP for channel = 0

Patrik-Berglund, 2018-04-04:
Hi,

Shouldn't channel = 0 cause the SDK to automatically scan all channels? Looking at the log I was expecting to see the AP present at channel 11 also in the case when channel is  set to 0? 

Using SDK: ESP-IDF v3.0-rc1

```
    tcpip_adapter_init();
    ESP_ERROR_CHECK(esp_event_loop_init(wifiEventHandler, NULL));
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));
    ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_RAM));
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_start() );
    
    for(int x = 0; x < 14; x++)
    {
        wifi_scan_config_t scan_config = {
            .ssid = NULL,
            .bssid = NULL,
            .channel = x,
            .show_hidden = true,
            .scan_type = WIFI_SCAN_TYPE_ACTIVE,
            .scan_time.active.min = 10,
            .scan_time.active.max = 100        
        };
        ESP_LOGI(TAG, "Start Scan");
        ESP_ERROR_CHECK(esp_wifi_scan_start(&scan_config, true));

        uint16_t apCount = 0;
        esp_wifi_scan_get_ap_num(&apCount);
        ESP_LOGI(TAG, "Number of access points found: %d", apCount);
        if (apCount > 0) 
        {
            wifi_ap_record_t *list = (wifi_ap_record_t *)malloc(sizeof(wifi_ap_record_t) * apCount);
            ESP_ERROR_CHECK(esp_wifi_scan_get_ap_records(&apCount, list));
            int i;
            ESP_LOGI(TAG, "======================================================================");
            ESP_LOGI(TAG, "             SSID             |    RSSI    |           AUTH           ");
            ESP_LOGI(TAG, "======================================================================");
            for (i=0; i<apCount; i++) {
                char *authmode;
                switch(list[i].authmode) {
                case WIFI_AUTH_OPEN:
                    authmode = "WIFI_AUTH_OPEN";
                    break;
                case WIFI_AUTH_WEP:
                    authmode = "WIFI_AUTH_WEP";
                    break;           
                case WIFI_AUTH_WPA_PSK:
                    authmode = "WIFI_AUTH_WPA_PSK";
                    break;           
                case WIFI_AUTH_WPA2_PSK:
                    authmode = "WIFI_AUTH_WPA2_PSK";
                    break;           
                case WIFI_AUTH_WPA_WPA2_PSK:
                    authmode = "WIFI_AUTH_WPA_WPA2_PSK";
                    break;
                default:
                    authmode = "Unknown";
                    break;
                }
                ESP_LOGI(TAG, "%26.26s    |    % 4d    |    %22.22s",list[i].ssid, list[i].rssi, authmode);
            }
            free(list);   
        }
    }
```

Log
[... 99 lines cut ...]
I (1849) WIFI: Number of access points found: 0 for channel: 8
I (1849) WIFI: Start Scan
I (1859) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (1949) WIFI: Number of access points found: 0 for channel: 9
I (1949) WIFI: Start Scan
I (1959) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (2049) WIFI: Number of access points found: 0 for channel: 10
I (2049) WIFI: Start Scan
I (2149) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (2249) WIFI: Number of access points found: 1 for channel: 11
I (2249) WIFI: ======================================================================
I (2249) WIFI:              SSID             |    RSSI    |           AUTH           
I (2259) WIFI: ======================================================================
I (2269) WIFI:             TestAp            |     -64    |        WIFI_AUTH_WPA2_PSK
I (2269) WIFI: Start Scan
I (2639) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (2679) WIFI: Number of access points found: 0 for channel: 12
I (2679) WIFI: Start Scan
I (3039) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3079) WIFI: Number of access points found: 0 for channel: 13
I (3079) wifi: flush txq
I (3079) wifi: stop sw txq
I (3079) wifi: lmac stop hw txq
I (3079) wifi: Deinit lldesc rx mblock:10
```

TimXia, 2018-04-05:
@Patrik-Berglund Yes, it should scan all of the channel. I have tried your code. I did see all the APs when scanning with channel 0. The following is my log:

I (69) boot: Detected ESP32
I (33) boot: ESP-IDF v3.0-rc1 2nd stage bootloader
I (33) boot: compile time 14:37:51
I (33) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0f32c ( 62252) map
I (118) esp_image: segment 1: paddr=0x0001f354 vaddr=0x3ffb0000 size=0x00cbc (  3260) load
I (119) esp_image: segment 2: paddr=0x00020018 vaddr=0x400d0018 size=0x50ba8 (330664) map
0x400d0018: _stext at ??:?

I (239) esp_image: segment 3: paddr=0x00070bc8 vaddr=0x3ffb0cbc size=0x0249c (  9372) load
I (243) esp_image: segment 4: paddr=0x0007306c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/sky/work/code/esp32/ssc/idf/components/freertos/./xtensa_vectors.S:1685

I (246) esp_image: segment 5: paddr=0x00073474 vaddr=0x40080400 size=0x0fd98 ( 64920) load
I (281) esp_image: segment 6: paddr=0x00083214 vaddr=0x400c0000 size=0x00000 (     0) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (293) cpu_start: Pro cpu up.
I (297) cpu_start: Starting app cpu, entry point is 0x400810ec
0x400810ec: call_start_cpu1 at /home/sky/work/code/esp32/ssc/idf/components/esp32/./cpu_start.c:222

I (0) cpu_start: App cpu up.
I (307) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (320) heap_init: At 3FFB8E18 len 000271E8 (156 KiB): DRAM
I (326) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (333) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (339) heap_init: At 40090198 len 0000FE68 (63 KiB): IRAM
I (345) cpu_start: Pro cpu start user code
I (28) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (96) wifi: wifi firmware version: f204566
I (96) wifi: config NVS flash: enabled
I (96) wifi: config nano formating: disabled
I (96) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (106) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (136) wifi: Init dynamic tx buffer num: 32
I (136) wifi: Init data frame dynamic rx buffer num: 32
I (136) wifi: Init management frame dynamic rx buffer num: 32
I (146) wifi: wifi driver task: 3ffbc3ac, prio:23, stack:4096
I (146) wifi: Init static rx buffer num: 10
I (156) wifi: Init dynamic rx buffer num: 32
I (156) wifi: wifi power manager task: 0x3ffc5090 prio: 21 stack: 2560
I (186) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (186) wifi: mode : sta (24:0a:c4:01:4e:04)
I (186) WIFI: Start Scan
I (196) WIFI: SYSTEM_EVENT_STA_START
I (1666) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (1696) WIFI: Number of access points found: 5 for channel 0
I (1696) WIFI: ======================================================================
I (1696) WIFI:              SSID             |    RSSI    |           AUTH           
I (1706) WIFI: ======================================================================
I (1716) WIFI:                         mx    |     -44    |    WIFI_AUTH_WPA_WPA2_PSK
I (1716) WIFI:              midea_ac_1134    |     -63    |        WIFI_AUTH_WPA2_PSK
I (1726) WIFI:               ECOVACS_1068    |     -70    |            WIFI_AUTH_OPEN
I (1736) WIFI:                 201-family    |     -88    |    WIFI_AUTH_WPA_WPA2_PSK
I (1746) WIFI:                               |     -90    |        WIFI_AUTH_WPA2_PSK
I (1756) WIFI: Start Scan
I (1856) WIFI: SYSTEM_EVENT_SCAN_DONE.
[... 27 lines cut ...]
I (2726) WIFI: Number of access points found: 0 for channel 7
I (2726) WIFI: Start Scan
I (2736) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (2826) WIFI: Number of access points found: 0 for channel 8
I (2826) WIFI: Start Scan
I (2926) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3026) WIFI: Number of access points found: 3 for channel 9
I (3026) WIFI: ======================================================================
I (3026) WIFI:              SSID             |    RSSI    |           AUTH           
I (3036) WIFI: ======================================================================
I (3046) WIFI:                         mx    |     -44    |    WIFI_AUTH_WPA_WPA2_PSK
I (3046) WIFI:                               |     -89    |        WIFI_AUTH_WPA2_PSK
I (3056) WIFI:                     wx614D    |     -94    |    WIFI_AUTH_WPA_WPA2_PSK
I (3066) WIFI: Start Scan
I (3086) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3176) WIFI: Number of access points found: 0 for channel 10
I (3176) WIFI: Start Scan
I (3186) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3276) WIFI: Number of access points found: 0 for channel 11
I (3276) WIFI: Start Scan
I (3636) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3676) WIFI: Number of access points found: 0 for channel 12
I (3676) WIFI: Start Scan
I (4036) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (4076) WIFI: Number of access points found: 0 for channel 13

TimXia, 2018-04-05:
@Patrik-Berglund  Yes, it should scan all of the channel. I have tried your code. I did see all the APs when scanning with channel 0. The following is my log:

`I (69) boot: Detected ESP32
I (33) boot: ESP-IDF v3.0-rc1 2nd stage bootloader
I (33) boot: compile time 14:37:51
I (33) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (68) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (83) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0f32c ( 62252) map
I (118) esp_image: segment 1: paddr=0x0001f354 vaddr=0x3ffb0000 size=0x00cbc (  3260) load
I (119) esp_image: segment 2: paddr=0x00020018 vaddr=0x400d0018 size=0x50ba8 (330664) map
0x400d0018: _stext at ??:?

I (239) esp_image: segment 3: paddr=0x00070bc8 vaddr=0x3ffb0cbc size=0x0249c (  9372) load
I (243) esp_image: segment 4: paddr=0x0007306c vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/sky/work/code/esp32/ssc/idf/components/freertos/./xtensa_vectors.S:1685

I (246) esp_image: segment 5: paddr=0x00073474 vaddr=0x40080400 size=0x0fd98 ( 64920) load
I (281) esp_image: segment 6: paddr=0x00083214 vaddr=0x400c0000 size=0x00000 (     0) load
I (291) boot: Loaded app from partition at offset 0x10000
I (291) boot: Disabling RNG early entropy source...
I (293) cpu_start: Pro cpu up.
I (297) cpu_start: Starting app cpu, entry point is 0x400810ec
0x400810ec: call_start_cpu1 at /home/sky/work/code/esp32/ssc/idf/components/esp32/./cpu_start.c:222

I (0) cpu_start: App cpu up.
I (307) heap_init: Initializing. RAM available for dynamic allocation:
I (314) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (320) heap_init: At 3FFB8E18 len 000271E8 (156 KiB): DRAM
I (326) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (333) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (339) heap_init: At 40090198 len 0000FE68 (63 KiB): IRAM
I (345) cpu_start: Pro cpu start user code
I (28) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (96) wifi: wifi firmware version: f204566
I (96) wifi: config NVS flash: enabled
I (96) wifi: config nano formating: disabled
I (96) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (106) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (136) wifi: Init dynamic tx buffer num: 32
I (136) wifi: Init data frame dynamic rx buffer num: 32
I (136) wifi: Init management frame dynamic rx buffer num: 32
I (146) wifi: wifi driver task: 3ffbc3ac, prio:23, stack:4096
I (146) wifi: Init static rx buffer num: 10
I (156) wifi: Init dynamic rx buffer num: 32
I (156) wifi: wifi power manager task: 0x3ffc5090 prio: 21 stack: 2560
I (186) phy: phy_version: 366.0, ba9923d, Oct 31 2017, 18:06:17, 0, 0
I (186) wifi: mode : sta (24:0a:c4:01:4e:04)
I (186) WIFI: Start Scan
I (196) WIFI: SYSTEM_EVENT_STA_START
I (1666) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (1696) WIFI: Number of access points found: 5 for channel 0
I (1696) WIFI: ======================================================================
I (1696) WIFI:              SSID             |    RSSI    |           AUTH           
I (1706) WIFI: ======================================================================
I (1716) WIFI:                         mx    |     -44    |    WIFI_AUTH_WPA_WPA2_PSK
I (1716) WIFI:              midea_ac_1134    |     -63    |        WIFI_AUTH_WPA2_PSK
I (1726) WIFI:               ECOVACS_1068    |     -70    |            WIFI_AUTH_OPEN
I (1736) WIFI:                 201-family    |     -88    |    WIFI_AUTH_WPA_WPA2_PSK
I (1746) WIFI:                               |     -90    |        WIFI_AUTH_WPA2_PSK
I (1756) WIFI: Start Scan
I (1856) WIFI: SYSTEM_EVENT_SCAN_DONE.
[... 28 lines cut ...]
I (2726) WIFI: Start Scan
I (2736) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (2826) WIFI: Number of access points found: 0 for channel 8
I (2826) WIFI: Start Scan
I (2926) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3026) WIFI: Number of access points found: 3 for channel 9
I (3026) WIFI: ======================================================================
I (3026) WIFI:              SSID             |    RSSI    |           AUTH           
I (3036) WIFI: ======================================================================
I (3046) WIFI:                         mx    |     -44    |    WIFI_AUTH_WPA_WPA2_PSK
I (3046) WIFI:                               |     -89    |        WIFI_AUTH_WPA2_PSK
I (3056) WIFI:                     wx614D    |     -94    |    WIFI_AUTH_WPA_WPA2_PSK
I (3066) WIFI: Start Scan
I (3086) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3176) WIFI: Number of access points found: 0 for channel 10
I (3176) WIFI: Start Scan
I (3186) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3276) WIFI: Number of access points found: 0 for channel 11
I (3276) WIFI: Start Scan
I (3636) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (3676) WIFI: Number of access points found: 0 for channel 12
I (3676) WIFI: Start Scan
I (4036) WIFI: SYSTEM_EVENT_SCAN_DONE.
I (4076) WIFI: Number of access points found: 0 for channel 13
`

Patrik-Berglund, 2018-04-05:
HI @TimXia 

Thanks for your quick reply which led me to the error!

Looking at my log you can see that i get "I (949) WIFI: SYSTEM_EVENT_STA_START.". My wifiEventHandler (code included below)  does a esp_wifi_connect() in that state, which seems to mess up the scan. Commenting out esp_wifi_connect() gives me the expected scan result.

```
static esp_err_t wifiEventHandler(void *ctx, system_event_t *event)
{
    switch(event->event_id) 
    {
        case SYSTEM_EVENT_STA_START:
            ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START.");
            //esp_wifi_connect();
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
            ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP.");
            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED.");
            //esp_wifi_connect();        
            break;
        case SYSTEM_EVENT_AP_STACONNECTED:
            ESP_LOGI(TAG, "Station:"MACSTR" join, AID=%d", MAC2STR(event->event_info.sta_connected.mac), event->event_info.sta_connected.aid);
            break;
        case SYSTEM_EVENT_AP_STADISCONNECTED: 
            ESP_LOGI(TAG, "Station:"MACSTR" leave, AID=%d", MAC2STR(event->event_info.sta_disconnected.mac), event->event_info.sta_disconnected.aid);
            break;   
        case SYSTEM_EVENT_SCAN_DONE:
            ESP_LOGI(TAG, "SYSTEM_EVENT_SCAN_DONE.");
            break;
        default:
            ESP_LOGI(TAG, "UNKNOWN EVENT");
            break;
    }
    return ESP_OK;
}
```

TimXia, 2018-04-06:
@Patrik-Berglund Yes, it is not supported to do `connect` and `scan` at the same time.

closed 2018-04-06 by Patrik-Berglund

rmc135, 2018-12-16:
I'm also seeing inconsistent results between scanning of channel 0, and individual scans of the remaining channels.

@TimXia If you look at your log carefully you can actually see a discrepancy: channel 0 shows 5 results, but the sum total of the individual scan is 6 results. It's reasonable to expect that a weak signal may not be picked up consistently - and there's probably not much point trying to connect to that AP anyway - but I found that over several runs, sometimes the access point which is only a couple of metres away was not picked up by the channel 0 scan. It did show every time on its own channel 7 scan. I've attached a screen cap of one such instance.

[image]