espressif/esp-idf #1827: E (26795) BT: No pending command

chocolatebrown, 2018-04-05:
I want esp32 as client to receive data but I keep getting errors like this:

`I (25335) GATTC_DEMO: connect to the remote device.
I (25345) GATTC_DEMO: stop scan successfully
I (25765) GATTC_DEMO: ESP_GATTC_CONNECT_EVT conn_id 0, if 3
I (25765) GATTC_DEMO: REMOTE BDA:
I (25765) GATTC_DEMO: f8 ce 32 8f 7d 21 
I (25775) GATTC_DEMO: open success
I (26275) GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 23, conn_id 0
I (26275) GATTC_DEMO: ESP_GATTC_SEARCH_CMPL_EVT
E (26795) BT: No pending command
I (30945) GATTC_DEMO: update connection params status = 0, min_int = 80, max_int = 160,conn_int = 160,latency = 0, timeout = 400
`

Weijian-Espressif, 2018-04-26:
sorry for being late, I will try to reprodue the issue and fix it.

Weijian-Espressif, 2018-05-31:
@chocolatebrown please update to the latest IDF and try, thanks.

closed 2018-06-01 by FayeY