espressif/esp-idf #1828: SD card high speed mode in via SPI

Reiner1210, 2018-04-05:
Hello,

in sdmmc_cmd.c there is the following

    if (!freq_switched &&
        config->max_freq_khz >= SDMMC_FREQ_DEFAULT) {
        ESP_LOGD(TAG, "switching to DS bus mode");
        err = (*config->set_card_clk)(config->slot, SDMMC_FREQ_DEFAULT);
        if (err != ESP_OK) {
            ESP_LOGE(TAG, "failed to switch peripheral to HS bus mode");
            return err;
        }
        freq_switched = true;

I think it would be a good idea to allow high speed if the SPI is connected via native pins
I tried the following
        err = (*config->set_card_clk)(config->slot, SDMMC_FREQ_HIGHSPEED);
which worked.

I assumed that the config->max_freq_khz is used as clock frequency - we should have a parameter to configure the speed.

igrr, 2018-04-10:
Hi @Reiner1210, in SD card example the `config` is initialized here:
https://github.com/espressif/esp-idf/blob/9a985359d63f32dba11815d1d31158e4244bcbd2/examples/storage/sd_card/main/sd_card_example_main.c#L69

If you add one line after that to set `config.max_freq_khz = SDMMC_FREQ_HIGHSPEED`, then it should work with native pins.

You can see this being done in this test, for example:

https://github.com/espressif/esp-idf/blob/9a985359d63f32dba11815d1d31158e4244bcbd2/components/sdmmc/test/test_sd.c#L181

Reiner1210, 2018-04-17:
> I had previously experimented with this. The driver code checks for SPI and
> will not enable HS mode. I had to remove the SPI (is_spi) check here:
> https://github.com/espressif/esp-idf/blob/c1c49b635a6c59b4ca0c4a285e490545241e7a73/components/sdmmc/sdmmc_cmd.c#L351-L352
> Not all SD cards work with high speed SPI mode. In my tests, Sandisk cards
> did not work.
>
> Mit AquaMail Android
> https://www.mobisystems.com/aqua-mail

I tried it with a Scan disc card and this adapter

https://www.amazon.de/your-orders/pop/ref=oh_aui_i_d_old_o10?_encoding=UTF8&gen=canonical&lineItemId=phjmqqinnkonqw&orderId=028-8225947-3665112&packageId=1&returnSummaryId=&returnUnitIndices=&shipmentId=DzDhfbMxH

pacucha42, 2022-03-08:
In case this issue is still interesting for anybody:

SD High-Speed mode over SPI is available for
* ESP32 - SDSPI using IOMUX pins
* other chips - SDSPI using any pins

with ESP-IDF v3.3.4, v4.0.2, v4.1.1, v4.2 and later releases

If you have any questions or comments to the topic, feel free to add them to this report or open a new one

closed 2022-03-08 by pacucha42