espressif/esp-idf #1830: Recommendation for usage of S/W

hvishu, 2018-04-06:
Hi from the below git repo, link
https://github.com/espressif/esp-idf.git, I see below branches

  remotes/origin/feature/psram_malloc
  remotes/origin/feature/rmt_translator
  remotes/origin/master
  remotes/origin/release/v2.0
  remotes/origin/release/v2.1
  remotes/origin/release/v3.0

And seems from the master, v3.1 is in progress.  We are planning to go into production with ESP32 WROOM module in about 2 months time.  
What is your suggestion on what branch to be used?  Should we go with release branches?  or the master branch?

Also, it seems there are different toolchains available.  I don't find any documentation, which mentions details of
-What is recommendation software version to be used?
-What toolchain to be used for which version?
-Where to find the toolchain for windows and Linux platforms?
Could you help me these?

hvishu, 2018-04-06:
Thank you.  I took the tar file and the tool chain.
Now get error like below, indicating itsn't a 

~/esp/esp-idf-3.0-rc1/examples/get-started/blink# make
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
DEFCONFIG
#
# configuration written to /home/user/esp/esp-idf-3.0-rc1/examples/get-started/blink/sdkconfig
#
MENUCONFIG

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

GENCONFIG
^[[Cfatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
fatal: Not a git repository (or any of the parent directories): .git
/home/user/esp/esp-idf-3.0-rc1/make/project.mk:498: warning: undefined variable 'which git'
WARNING: Missing submodule components/micro-ecc/micro-ecc...
ERROR: esp-idf must be cloned from git to work.
/home/user/esp/esp-idf-3.0-rc1/make/project.mk:498: recipe for target '/home/user/esp/esp-idf-3.0-rc1/components/micro-ecc/micro-ecc/.git' failed
make[1]: *** [/home/user/esp/esp-idf-3.0-rc1/components/micro-ecc/micro-ecc/.git] Error 1
/home/user/esp/esp-idf-3.0-rc1/components/bootloader/Makefile.projbuild:40: recipe for target '/home/user/esp/esp-idf-3.0-rc1/examples/get-started/blink/build/bootloader/bootloader.bin' failed
make: *** [/home/user/esp/esp-idf-3.0-rc1/examples/get-started/blink/build/bootloader/bootloader.bin] Error 2

mahavirj, 2018-04-06:
@hvishu Please confirm if you have downloaded this [artifact](https://github.com/espressif/esp-idf/releases/download/v3.0-rc1/esp-idf-v3.0-rc1.zip)
Please see section `Obtaining v3.0-rc1` for more details on link shared earlier.

hvishu, 2018-04-06:
OK, Got it. 
From the earlier link, at the top there was .tar file.  I downloaded that and faced the problem.  
After taking the commit id from the log, things are fine.

hvishu, 2018-04-09:
Reopened this issue to understand what is the plan for the next release?

Right now, I am using the patch for below issue as it is not part of v3.0-rc1.
https://github.com/espressif/esp-idf/issues/1619

Knowing the release plan will help us planning to migrate our application to that release.

mahavirj, 2018-04-09:
@hvishu Quoting from https://www.esp32.com/viewtopic.php?t=5321

> 3.0 is in final regression testing, there shouldn't be any more significant changes before the final release. I'd suggest using the release/v3.0 branch now (which is a long way ahead of the v3.0-rc1 tag, as you've noted). This branch will become the final release soon, and you can fast-forward it to that tag once it's available.

hvishu, 2018-04-10:
Thanks, what will be toolchain version to be used for this?  Could you point the link?
Regards
Vishwa